Koenkk/zigbee2mqtt #21254: 1.35.2-1 Z2M fails to start on ezsp controller, Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,12]}

tomazurro, 2024-02-06:
### What happened?

After updating to version 1.35.2-1 z2m addon container fails to start (randomly,  but 4 times out of 5) with
[...]
error 2024-02-05 15:04:23: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,12]}

### What did you expect to happen?

the addon should start without problems, as it does with 1.35.1-1

### How to reproduce it (minimal and precise)

Upgrade to 1.35.2-1 from previous 1.35.1-1

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812193316-if00

### Setup

Home Assistant OS on Pi 3b+

### Debug log

[log.z2m.failing.txt](https://github.com/Koenkk/zigbee2mqtt/files/14176576/log.z2m.failing.txt)

jvuotila, 2024-02-11:
I have excatly same error. @tomazurro did you get it fixed? My adapter is SkyConnect.

tomazurro, 2024-02-12:
@jvuotila, yes rollback fixes the problem but is no solution.

I mistakenly posted this issue at the wrong place, the error is likely related to zigbee-herdsman adapter driver update with z2m 1.35.2

https://github.com/Koenkk/zigbee-herdsman/issues/910

newlund, 2024-02-16:
Add:
rtscts: false
to adapter config.
Works for me

tomazurro, 2024-02-17:
This error is fixed in zigbee-herdsmann dev branch with applied commit 
https://github.com/Koenkk/zigbee-herdsman/pull/916

Until these reach the published version, its a little tricky, but possible, to patch the docker container in haos:

1. recompile node module from src with applied commit (on a linux machine or container)
2. patch zigbee-herdsmann module in zigbee2mqtt container  (actually i copied just the adapter/ezsp directory) with newer files 
3. export patched container to docker image 
4. rename docker images (original to something else, patched image to original name), since watchdog is starting new containers off the original image name, disable automatic updates for addon.

closed 2024-02-17 by tomazurro

mbrucher, 2024-05-05:
Is this something to fix on the HA side, or on zigbee2MQTT?