Koenkk/zigbee2mqtt #21255: [New device support]: Tuya 214C Ultrasonic water meter with valve

michaeleverist, 2024-02-06:
### Link

https://www.alibaba.com/product-detail/Smart-Ultrasonic-Water-Flow-Meter-With_1600722839075.html

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c1384c55aa7d9b","nwkAddr":23333,"manufId":4417,"manufName":"_TZE200_vuwtqx0t","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":1,"65534":0,"appVersion":72,"modelId":"TS0601","manufacturerName":"_TZE200_vuwtqx0t","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1707216891922}

### Comments

I've tried following the guide and i think I am close.  I got the Tuya DP's from the dev portal as well as i think the datatypes.  I'm getting let down in my lack of knowledge of JS/TS and Zigbee2MQTT in general.

Current WIP external converter:  https://gist.github.com/michaeleverist/f8399bb79ac08a7667883bc2bef1b11f

Output from the script in tuya_dev:
`{"1":"Water consumption","2":"Month consumption","3":"Daily consumption","4":"Report period set","5":"Warning","6":"Month and daily frozen time set","13":"Switch","14":"Enable auto clean","16":"Meter ID","18":"Reverse Water Consumption","21":"Instantaneous Flow Rate","22":"Working temperature","26":"Power supply voltage"}
`

Tuya Dev device debugging info:
[image]
[image]
[image]
[image]
[image]

Not sure if I've missed anything else?  Happy to keep working on it myself but am getting caught on the relationship between exposes :[] and mete tuyaDatapoints and how exposes knows which datapoint to use.

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_vuwtqx0t',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

michaeleverist, 2024-03-19:
I have made a little progress with this but still my lack of understanding has hampered.   The basic functions (total water consumed, valve open/close) work but the other values aren't being translated correctly.

I'd appreciate anyone who can offer help!

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',         
            manufacturerName: '_TZE200_vuwtqx0t',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Tuya water meter with valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [       
        e.numeric('water_consumed', ea.STATE).withUnit('L').withDescription('Total water consumed'),
        
        // Need to fix this month and daily consumption - currently shows as JSON
        e.numeric('month_consumption', ea.STATE).withDescription('month consumption'),
        e.numeric('daily_consumption', ea.STATE).withDescription('daily consumption'),        
        
        // Not sure what this does or how to change it
        e.numeric('month_and_daily_frozen_set', ea.STATE).withDescription('Month Daily Frozen Set'),
        
        //  I can't seem to get this to work - when i try to change the value i get an error in zigbee2mqtt, stuck on 1 Hour
        e.enum('report_period', ea.STATE_SET, ['1', '2', '3', '4', '6', '8', '12', '24']).withDescription('Temperature Report interval'),       
        e.text('warning', ea.STATE),  // FIX THIS - BITMAP -> fault types        
              
        // These are attempts at getting the valve open/closed switch and the autoclean switch working together.        
        // Valve Open/Close -  (Name isnt correct in Zigbee2MQTT)
        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Valve Open / Close'),
        // tuya.exposes.switch().withDescription('Valve Open / Close'),
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Valve position'),
        
        // Autoclean switch - deactivated because I dont know how to get it working independently to the valve open/close switch
        // e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Auto Clean'),
        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Auto Clean'),
        // tuya.exposes.switch().withDescription('Auto Clean'),                

        // Note sure what this does - it wasnt in the tuya DP's extracted from the IOT portal?
        e.text('something', ea.STATE).withDescription('Something'),
        e.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),        
        // Reverse consumption and instantaneous flow are appearing as JSON, in tuya they are a Base 64 but I dont know how to get it to show correctly
        e.numeric('reverse_water_consumption', ea.STATE).withUnit('L').withDescription('reverse water flow'),
        e.numeric('instantaneous_flow_rate', ea.STATE).withUnit('L/hr').withDescription('Inst flow rate'),        
        
        e.temperature().withDescription('Water Temperature'),
        //Voltage works but doesnt show as the battery voltage in Z2M - no doubt something i havent understood
        e.battery_voltage('battery_voltage', ea.STATE).withDescription('battery voltage'),
        ],

    meta: {        
        tuyaDatapoints: [
            [1,'water_consumed', tuya.valueConverter.raw],  // dtype number
            [2,'month_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [3,'daily_consumption', tuya.valueConverter.raw],   // dtype 0 - raw            
            // [4,'report_period', tuya.valueConverterBasic.lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3h': tuya.enum(2),
            //                                                         '4h': tuya.enum(3), '6h': tuya.enum(4), '8h': tuya.enum(5),
            //                                                         '12h': tuya.enum(6), '24h': tuya.enum(7)
            //                                                     })],   // dtype 4 - enum            
                                                
            [4,'report_period', tuya.valueConverterBasic.lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2),
                                                                '4': tuya.enum(3), '6': tuya.enum(4), '8': tuya.enum(5),
                                                                '12': tuya.enum(6), '24': tuya.enum(7)
                                                            })],   // dtype 4 - enum                                                                            
            [5,'warning', tuya.dataTypes.enum], // bitmap,
            [6,'month_and_daily_frozen_set', tuya.valueConverter.raw],  // dtype 0 - raw
            [13,'state', tuya.valueConverter.onOff],   // Bool   VALVE OPEN CLOSE
[... cut ...]
            [15,'something', tuya.valueConverter.raw],   // dtype 0 - raw  - No idea what this does?
            [16,'meter_id', tuya.valueConverter.raw],  //dtype 3 - raw   METER ID
            [18,'reverse_water_consumption', tuya.valueConverterBasic.numeric],   // dtype 0 - raw
            [21,'instantaneous_flow_rate', tuya.valueConverter.numeric], // dtype 0 - raw
            [22,'temperature', tuya.valueConverter.divideBy100],
            [26,'battery_voltage', tuya.valueConverter.raw],
        ],
    },
};
module.exports = definition;
```

Koenkk, 2024-03-19:
What values are not translated correctly?

michaeleverist, 2024-03-19:
Hi Koenkk,  When i say not translated correctly I mean I am not using the correct exposes and/or meta (due to lack of understanding :) )

Below is screenshot of zigbee2mqtt.

- I still have the autoclean switch disabled as if i enable it (enen with the different DP's in the meta) pressing one switch operates both.  
- The instantaneous and reverse should be saying 0 and 1 i think.
- the report period set gives an error whenever i choose something from the dropdown.
- I'm not too sure how to best approach the daily and monthly consumption values.

[image]

Koenkk, 2024-03-19:
> I still have the autoclean switch disabled as if i enable it (enen with the different DP's in the meta) pressing one switch operates both.

It also operates the main switch (valve)?

> The instantaneous and reverse should be saying 0 and 1 i think.

I think you need to use `tuya.valueConverter.raw` for these

> the report period set gives an error whenever i choose something from the dropdown.

What's the error?

> I'm not too sure how to best approach the daily and monthly consumption values.

I'm suprised a string is reported here, would expect it to be a number

michaeleverist, 2024-03-20:
Thanks for looking at this.

> I still have the autoclean switch disabled as if i enable it (enen with the different DP's in the meta) pressing one switch operates both.
> 
> It also operates the main switch (valve)?
> 
 Yes - I added 

` e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Auto Clean'),`

and 
` [14,'auto_clean', tuya.valueConverter.onOff], `

And i get another switch that changes state with the first one:
[image]

  
> The instantaneous and reverse should be saying 0 and 1 i think.
> I think you need to use tuya.valueConverter.raw for these.  

I originally had them as raw but there was no difference.  I've changed back to raw now.

> the report period set gives an error whenever i choose something from the dropdown.
> 
> What's the error?

see below:
[image]

To be honest, the one of value is the total consumed, as the daily/monthly can be derived in Home Assistant.  The instantaneous flow seems pointless since there is a 1hr min reporting period.

The 2 things that would be good to obtain are the battery voltage and the error codes.  

- Battery voltage shows in the state tab but as N/A in 'exposes'
- Error shows 6144 and is type bitmap with what i think are the translations in my original post

Current State and Exposes:

[image]

[image]

Koenkk, 2024-03-20:
- Use `e.binary('auto_clean', ea.STATE_SET, 'ON', 'OFF').withDescription('Auto Clean'),`
- For the report period, change `toZigbee: [tuya.tz.datapoints],` to `toZigbee: [tzdatapoints],` and above `const definition = {` add:

```js
const tzdatapoints = {
    key: [...tuya.tz.datapoints, 'report_period'],
    ...tuya.tz.datapoints,
}
```

michaeleverist, 2024-03-20:
for the report_period:  
When I add the above the device isnt recognised.  If i comment out the tzdatapoints added it is recognised again.  I added a semicolon at the end too but no change.

The auto_clean switch now appears to switch independently of the other switch but gives a no converter available for auto_clean "ON" when pressing it
[image]

Koenkk, 2024-03-21:
You also need to add that, e.g. `key: [...tuya.tz.datapoints, 'report_period', 'auto_clean`],`

michaeleverist, 2024-03-21:
Thanks, - I really appreciate you taking the time to help with this!

Whenever I have the `const tzdatapoints = { ....`    The device isnt recognised.

full config now:
```ts
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tzdatapoints = {
    key: [...tuya.tz.datapoints, 'report_period', 'auto_clean'],
    ...tuya.tz.datapoints,
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',         
            manufacturerName: '_TZE200_vuwtqx0t',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Tuya water meter with valve',
    fromZigbee: [tuya.fz.datapoints],
    // toZigbee: [tuya.tz.datapoints],
    toZigbee: [tzdatapoints],    
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [       
        e.numeric('water_consumed', ea.STATE).withUnit('L').withDescription('Total water consumed'),
        
        // Need to fix this month and daily consumption - currently shows as JSON
        e.numeric('month_consumption', ea.STATE).withDescription('month consumption'),
        e.numeric('daily_consumption', ea.STATE).withDescription('daily consumption'),        
        
        // Not sure what this does or how to change it
        e.numeric('month_and_daily_frozen_set', ea.STATE).withDescription('Month Daily Frozen Set'),
        
        //  I can't seem to get this to work - when i try to change the value i get an error in zigbee2mqtt, stuck on 1 Hour
        e.enum('report_period', ea.STATE_SET, ['1h', '2h', '3h', '4h', '6h', '8h', '12h', '24h']).withDescription('Temperature Report interval'),       
        e.text('warning', ea.STATE),  // FIX THIS - BITMAP -> fault types        
              
        // These are attempts at getting the valve open/closed switch and the autoclean switch working together.        
        // Valve Open/Close -  (Name isnt correct in Zigbee2MQTT)
        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Valve Open / Close'),
        // tuya.exposes.switch().withDescription('Valve Open / Close'),
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Valve position'),
        
        // Autoclean switch - deactivated because I dont know how to get it working independently to the valve open/close switch
        e.binary('auto_clean', ea.STATE_SET, 'ON', 'OFF').withDescription('Auto Clean'),
        
        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Auto Clean'),
        // tuya.exposes.switch().withDescription('Auto Clean'),                

        // Note sure what this does - it wasnt in the tuya DP's extracted from the IOT portal?
        e.text('something', ea.STATE).withDescription('Something'),
        e.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),        
        // Reverse consumption and instantaneous flow are appearing as JSON, in tuya they are a Base 64 but I dont know how to get it to show correctly
        e.numeric('reverse_water_consumption', ea.STATE).withDescription('reverse water flow'),
        e.numeric('instantaneous_flow_rate', ea.STATE).withDescription('Inst flow rate'),        
        
        e.temperature().withDescription('Water Temperature'),
        //Voltage works but doesnt show as the battery voltage in Z2M - no doubt something i havent understood
        e.battery_voltage('battery_voltage', ea.STATE).withDescription('battery voltage'),
        ],

    meta: {        
        tuyaDatapoints: [
            [1,'water_consumed', tuya.valueConverter.raw],  // dtype number
            [2,'month_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [3,'daily_consumption', tuya.valueConverter.raw],   // dtype 0 - raw                                                                    
            [4,'report_period', tuya.valueConverterBasic.lookup({'1h': tuya.enum(0), '2h': tuya.enum(1), '3h': tuya.enum(2),
                                                                '4h': tuya.enum(3), '6h': tuya.enum(4), '8h': tuya.enum(5),
                                                                '12h': tuya.enum(6), '24h': tuya.enum(7)
                                                            })],   // dtype 4 - enum                                                                            
            [5,'warning', tuya.dataTypes.enum], // bitmap,
            [6,'month_and_daily_frozen_set', tuya.valueConverter.raw],  // dtype 0 - raw
            [13,'state', tuya.valueConverter.onOff],   // Bool   VALVE OPEN CLOSE
            [14,'auto_clean', tuya.valueConverter.onOff],   // Bool   AUTOCLEAN SW -  DISABLED  UNTILL I CAN FIGURE IT OUT
            [15,'something', tuya.valueConverter.raw],   // dtype 0 - raw  - No idea what this does?
            [16,'meter_id', tuya.valueConverter.raw],  //dtype 3 - raw   METER ID
            [18,'reverse_water_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [21,'instantaneous_flow_rate', tuya.valueConverter.raw], // dtype 0 - raw
            [22,'temperature', tuya.valueConverter.divideBy100],
            [26,'battery_voltage', tuya.valueConverter.raw],
        ],
    },
};
module.exports = definition;
```

Koenkk, 2024-03-24:
Made a mistake, use:

```
const tzdatapoints = {
    key: [...tuya.tz.datapoints.key, 'report_period', 'auto_clean'],
    ...tuya.tz.datapoints,
};
```

Perfect-Web, 2024-04-07:
hi guys, any news about this ?

michaeleverist, 2024-04-08:
I didnt have any further success - i can open and close the valve as well as view total consumption updated every hour - I'm not using the other consumption values.  I will go back to it at some stage but have settled for the basics.

I still cant see battery voltage although i can see the value, and cant set the reporting period or the autoclean switch.

bkujszczyk, 2024-04-22:
Good morning

I have the same counter using this external converter. I have a situation where the valve closes automatically from time to time and remains closed until it is manually opened.

Do you also have this situation? how to solve this problem?

cosmico89, 2024-05-22:
HI,
after purchasing the valve I uploaded your file to check what really worked and what didn't, let's say that I only needed two main functions VALVE STATUS (open/closed) and FLOW RATE.
unfortunately I know that only the VALVE STATUS works
Is there any way to help "find" the missing parameters?

Thank you!

bkujszczyk, 2024-05-24:
Very strange

My file (attached) shows the flow and valve status (open/closed)

Unfortunately, it closes the valve every now and then for an indefinite and irregular period of time. most often when I'm not at home. The manufacturer doesn't know anything.

Connected to tuya it does not cause such surprises.

I don't know what is the cause
[_TZE200_vuwtqx0t.txt](https://github.com/Koenkk/zigbee2mqtt/files/15434936/_TZE200_vuwtqx0t.txt)

cosmico89, 2024-05-24:
Now I'll try your file, but if you turn on the water after how long do you see the instantflow move? always if it works for you

bkujszczyk, 2024-06-03:
That's how it works
the counter measures the amount of liquid. but unfortunately it comes out at unexpected moments. I don't know why and there's nothing in the logs that could help me

osonatech2021, 2024-06-17:
No puedo visualizar ningun valor, me sale que no esta soportado. Como hizieron para poder visualizar el dispositivo des de zigbee2mqtt?

Gracias gente

bkujszczyk, 2024-06-18:
I received an instruction video from the manufacturer

It's in the link

https://drive.google.com/file/d/1QDs58Q_XxV7qWa0PvZjHPiM6tdpggiuy/view?usp=sharing

osonatech2021, 2024-06-18:
El dispositivo, ya esta vinculado a z2m, pero no me lee ninguna entrada
[image]
[image]

osonatech2021, 2024-06-21:
[image]

Donde consigo los ficheros que faltan?

bkujszczyk, 2024-06-21:
https://community.home-assistant.io/t/solved-z2mqtt-problem-error-cannot-find-module-zigbee-herdsman-converters-lib-extend/347835

osonatech2021, 2024-06-22:
Gracias por los aportes familia, al final lo consegui, pero... solo recibo los datos cada 12h, al modificar a 1h o la que sea, sale: no converter avalible, o no sale nada, pero no modifica el valor

osonatech2021, 2024-06-22:
por cierto, en el apartado de icono se puede poner: https://th.bing.com/th/id/OIP.XLrfQ0N_eVoD-tkLs2wpDAHaGd?w=223&h=195&c=7&r=0&o=5&pid=1.7
asi sale el dibujo del contador

gabrielpowerc, 2024-10-08:
Hello,
I have this water meter installed in z2mqtt and worked fine until approximative one or two month ago. Since then, the device is unsupported.
I'm using the external converter posted [by bkujszczyk](https://github.com/Koenkk/zigbee2mqtt/issues/21255#issuecomment-2129529803)

In z2m I found the following errors:
[2024-10-08 12:21:23] error: 	z2m: Failed to load external converter file 'TS0601.js' (tuya.tz.datapoints.key is not iterable)
[2024-10-08 12:21:23] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-10-08 12:21:23] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

Do you have any solution?

paissad, 2024-10-16:
Hi @gabrielpowerc & @osonatech2021 

I just purchased and added this water metering product in z2m and used the file content available through https://github.com/Koenkk/zigbee2mqtt/issues/21255#issuecomment-2129529803
After that, I restarted zigbee2mqtt, but I still don't see anything into the `Expose` tab of the device into the UI of z2m

Here is what I did to be precise:

1. Added the tool (pairing mode) through z2m
2. Created `water_meter.js` file next to the `zigbee2mqtt/configuration.yaml` file with the file content I said above
3. Updated `zigbee2mqtt/configuration.yaml` file with the following content:
```yaml
external_converters:
  - water_meter.js
```
4. Restarted zigbee2mqtt

Did I miss something? Or am I doing something wrong?

Thanks in advance for your help 🙏

gabrielpowerc, 2024-10-28:
Hello @paissad & @Koenkk 
The water meter worked for about 2 month and I suppose that after an update, stopped working - it is present in z2m but unsupported.

The error is in the external converter: 

[2024-10-28 19:22:14] error: 	z2m: Failed to load external converter file 'TS0601.js' (tuya.tz.datapoints.key is not iterable)
[2024-10-28 19:22:14] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-10-28 19:22:14] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

Thank you!

> Hi @gabrielpowerc & @osonatech2021
> 
> I just purchased and added this water metering product in z2m and used the file content available through [#21255 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21255#issuecomment-2129529803) After that, I restarted zigbee2mqtt, but I still don't see anything into the `Expose` tab of the device into the UI of z2m
> 
> Here is what I did to be precise:
> 
> 1. Added the tool (pairing mode) through z2m
> 2. Created `water_meter.js` file next to the `zigbee2mqtt/configuration.yaml` file with the file content I said above
> 3. Updated `zigbee2mqtt/configuration.yaml` file with the following content:
> 
> ```yaml
> external_converters:
>   - water_meter.js
> ```
> 
> 4. Restarted zigbee2mqtt
> 
> Did I miss something? Or am I doing something wrong?
> 
> Thanks in advance for your help 🙏

danielewyss, 2024-11-01:
Hi, thanks to all for the effort.
Is there something that I could take to make it working? Or at least partially working ?

MiAutomations, 2024-11-06:
> Hello @paissad & @Koenkk The water meter worked for about 2 month and I suppose that after an update, stopped working - it is present in z2m but unsupported.
>
> The error is in the external [...]

Hello To all

I created also the file as external_converters I restarted the Z2M I try to pairing the device using the last versions of Z2M 1.41.0, but I still facing the same issue described by the @gabrielpowerc the device still as unsupported and there aren't any control available od data about this device on the Z2M GUI

[image]

Is there any think to do in order to fix this? 
Thank in advance for your support
Best regards

MiAutomations, 2024-12-24:
Any news about this?

bootix, 2025-01-06:
Same problem here after upgrade from v1.x -> v2.0.0-2. 
Here is how I made it working again:

Remove this entry from zigbee2mqtt/configuration.yaml:
```
external_converters:
  - water_meter.js
```

Create a directory "external_converters" in the data directory, e.g. "mkdir zigbee2mqtt/external_converters".

Move the file "water_meter.js" into it, e.g. "mv water_meter.js external_converters".

Restart the addon and your device should be displayed as "_TZE200_ajlu4cud" instead of "unsupported".

Your vendor ID may vary from mine. Use "develper_console->generate_external_definition" to find yours.

I have attached my "water_meter.js" for reference. Please rename it from "water_meter.txt" to "water_meter.js".

[water_meter.txt](https://github.com/user-attachments/files/18320271/water_meter.txt)

Abubakar506, 2025-01-14:
Hello!
I wat to convert this JSON into single value:
Month consumption: {"0":25,"1":1,"2":25,"3":1,"4":0,"5":0,"6":6,"7":192} Daily consumption: {"0":1,"1":14,"2":1,"3":14,"4":0,"5":0,"6":6,"7":192} Month and daily frozen set: {"0":111,"1":110} Reverse water consumption: {"0":0,"1":0,"2":0,"3":0} Instant flow rate: {"0":0,"1":0,"2":0,"3":0}

I tried different functions but no success. I'm curious is it possible or not?
Thanks

paissad, 2025-01-15:
Hi @MiAutomations 

would you please share your content when you got to the the `dev console` and click on `generate external_definitions`?

MiAutomations, 2025-01-15:
> Hi @MiAutomations
> 
> would you please share your content when you go to the the `dev console` and click on `generate external_definitions`?

'''
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_vuwtqx0t',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
'''

MiAutomations, 2025-01-15:
> @MiAutomations
> 
> Before the upgrade to z2m 2.0 , was it functional? You were able to see the water consumption? If yes, would you please share the content of your previous `water_meter.js` file?

Yes I have this functionality 

Before used your file water_meter.js as a external converter, I used the one that shared by the @bkujszczyk 

[https://github.com/Koenkk/zigbee2mqtt/issues/21255#issuecomment-2129529803](url)

MiAutomations, 2025-01-16:
@paissad sorry I notice that the link shared is not working... The file used before is this one shared by the @bkujszczyk 
in that works on the old Z2M version, that works also with the on/off or open/close the valve 

[_TZE200_vuwtqx0t.txt](https://github.com/user-attachments/files/18440759/_TZE200_vuwtqx0t.txt)

paissad, 2025-01-16:
@MiAutomations 

Sorry, but I can't really help here :/ 
On my side, it does not work properly either. The water consumption always show 0L ...
BTW, with the integration with Tuya, it does not show me much ... so i don't know if it's related!

bkujszczyk, 2025-01-16:
Hello

My tuya shows the correct amount of water used.

After connecting to HA I had correct water consumption readings, but at random moments it would cut off my water supply. Which was annoying and annoying especially when I was not at home.

That was quite a while ago around April 2023.

I didn't test it again after that.

MiAutomations, 2025-01-16:
> Hello
>
> My tuya shows the correct amount of water used.
>
> After connecting to HA I had correct water consumption readings, but at random moments it would cut off my water supply. Which was annoying [...]

Hello @bkujszczyk , when you mention "After connecting to HA" are you using the Z2M? If yes, can you please confirm that you are using the last Z2M version? Do you have the control valve available on/off open/close?

bkujszczyk, 2025-01-16:
Yes I use z2m, after all this is a forum about this particular add-on
I don't remember what version I had then. At the moment I use the meter through the TUYA system, it only serves me to control the amount of septic tank.

mattnemoz, 2025-01-21:
Hello !
I created both directory and js file, but the device still remaine unsupported.

[image]

The vendor is good. It's the first time I paired the device.

[image]

I have restarted Z2M and the pi, what can I do for make the device working good please ?

Thx

wojtulab, 2025-03-02:
try to change tzdatapoints :
`const tzdatapoints = tuya?.datapoints ? {
    key: [...tuya.datapoints.key, 'report_period', 'auto_clean'],
    ...tuya.datapoints,
} : {};`

its working in my scenario

PixelSwan, 2025-03-02:
> try to change tzdatapoints : `const tzdatapoints = tuya?.datapoints ? { key: [...tuya.datapoints.key, 'report_period', 'auto_clean'], ...tuya.datapoints, } : {};`
> 
> or this one: `const tzdatapoints = { key: ['report_period', 'state', 'auto_clean'], ...tuya.tz.datapoints, }; `
> 
> its working in my scenario

Hello, can you show whole code?

wojtulab, 2025-03-03:
[image]

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
// const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

//const tzdatapoints = tuya?.datapoints ? {
//    key: [...tuya.datapoints.key, 'report_period', 'auto_clean'],
//    ...tuya.datapoints,
//} : {};

const tzdatapoints = {
    key: ['report_period', 'state', 'auto_clean'],  // `undefined`
    ...tuya.tz.datapoints,  // `tuya.tz.datapoints`
};

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ajlu4cud',
    description: 'Water meter',
    extend: [],
    fromZigbee: [tuya.fz.datapoints],
    // toZigbee: [tuya.tz.datapoints],
    toZigbee: [tzdatapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric('water_consumed', ea.STATE).withUnit('L').withDescription('Total water consumed'),

        // Need to fix this month and daily consumption - currently shows as JSON
        e.numeric('month_consumption', ea.STATE).withDescription('month consumption'),
        e.numeric('daily_consumption', ea.STATE).withDescription('daily consumption'),

        // Not sure what this does or how to change it
        e.numeric('month_and_daily_frozen_set', ea.STATE).withDescription('Month Daily Frozen Set'),

        //  I can't seem to get this to work - when i try to change the value i get an error in zigbee2mqtt, stuck on 1 Hour
        e.enum('report_period', ea.STATE_SET, ['1h', '2h', '3h', '4h', '6h', '8h', '12h', '24h']).withDescription('Temperature Report interval'),
        e.text('warning', ea.STATE),  // FIX THIS - BITMAP -> fault types

        // These are attempts at getting the valve open/closed switch and the autoclean switch working together.
        // Valve Open/Close -  (Name isnt correct in Zigbee2MQTT)
        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Valve Open / Close'),
        // tuya.exposes.switch().withDescription('Valve Open / Close'),
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Valve position'),

        // Autoclean switch - deactivated because I dont know how to get it working independently to the valve open/close switch
        e.binary('auto_clean', ea.STATE_SET, 'ON', 'OFF').withDescription('Auto Clean'),

        // e.switch().setAccess('state', ea.STATE_SET).withDescription('Auto Clean'),
        // tuya.exposes.switch().withDescription('Auto Clean'),

// Note sure what this does - it wasnt in the tuya DP's extracted from the IOT portal?
        e.text('something', ea.STATE).withDescription('Something'),
        e.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
        // Reverse consumption and instantaneous flow are appearing as JSON, in tuya they are a Base 64 but I dont know how to get it to show correctly
        e.numeric('reverse_water_consumption', ea.STATE).withDescription('reverse water flow'),
        e.numeric('instantaneous_flow_rate', ea.STATE).withDescription('Inst flow rate'),

        e.temperature().withDescription('Water Temperature'),
        //Voltage works but doesnt show as the battery voltage in Z2M - no doubt something i havent understood
        e.battery_voltage('battery_voltage', ea.STATE).withDescription('battery voltage'),
        ],

    meta: {
        tuyaDatapoints: [
            [1,'water_consumed', tuya.valueConverter.raw],  // dtype number
            [2,'month_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [3,'daily_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [4,'report_period', tuya.valueConverterBasic.lookup({'1h': tuya.enum(0), '2h': tuya.enum(1), '3h': tuya.enum(2),
                                                                '4h': tuya.enum(3), '6h': tuya.enum(4), '8h': tuya.enum(5),
                                                                '12h': tuya.enum(6), '24h': tuya.enum(7)
                                                            })],   // dtype 4 - enum
            [5,'warning', tuya.dataTypes.enum], // bitmap,
            [6,'month_and_daily_frozen_set', tuya.valueConverter.raw],  // dtype 0 - raw
            [13,'state', tuya.valueConverter.onOff],   // Bool   VALVE OPEN CLOSE
            [14,'auto_clean', tuya.valueConverter.onOff],   // Bool   AUTOCLEAN SW -  DISABLED  UNTILL I CAN FIGURE IT OUT
            [15,'something', tuya.valueConverter.raw],   // dtype 0 - raw  - No idea what this does?
            [16,'meter_id', tuya.valueConverter.raw],  //dtype 3 - raw   METER ID
            [18,'reverse_water_consumption', tuya.valueConverter.raw],   // dtype 0 - raw
            [21,'instantaneous_flow_rate', tuya.valueConverter.raw], // dtype 0 - raw
            [22,'temperature', tuya.valueConverter.divideBy100],
            [26,'battery_voltage', tuya.valueConverter.raw],
        ],
    },
};
module.exports = definition;

`

PixelSwan, 2025-03-03:
> [image]
>
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

thanks!

MiAutomations, 2025-03-03:
> [image]
>
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]
@wojtulab 

Is this change to made on the external converter? 

Can you please share your file? 

Thank you

gabrielpowerc, 2025-03-03:
> water_meter.js

working!

thanks a lot!

MiAutomations, 2025-03-10:
> > water_meter.js
> 
> working!
> 
> thanks a lot!

Hello @gabrielpowerc can you please share your file ? I try again to remove the device and added it again with the file in the folder 
external_converters/water_meter.js

And I still without ant entity exposed .... :( 

[image]

Thank you

MiAutomations, 2025-03-11:
Hello To all 

After removed the device and added again the file to the folder I have now again the device available, Thank you 

But I still without the ON/OFF entities as mentioned by the @michaeleverist , can you please let us know how can we add this entities ? 

[image]

Thank you

gabrielpowerc, 2025-03-11:
> Hello [@gabrielpowerc](https://github.com/gabrielpowerc) can you please share your file ? I try again to remove the device and added it again with the file in the folder [...]

Hello  @MiAutomations,
see the file in attach (change the extension to .js)

[water_meter.txt](https://github.com/user-attachments/files/19189869/water_meter.txt)

freeewilly, 2025-03-12:
Hi everyone! My apologies for being a complete noob when it comes to installing quirks for Z2MQTT. I looked everywhere but seem to be unable to find step to step instructionsas to where exactly do I save the conf file and how do I make Z2MQTT look in the correct folder/ If anyone could point me in the right direction, I would be so grateful, thanks!

gabrielpowerc, 2025-03-14:
@wojtulab 

Hi,
is your water meter still working after the last update? mine is again unsupported
(js file is in the right place)

wojtulab, 2025-03-14:
yes but can't synchronize it... I need to reconnect it every time to sync all data. It's a bug?

freeewilly, 2025-03-27:
Used to get this error:

`error: 	z2m: Failed to load external converter file 'devices/water_flow_meter.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)`

After I commented out "const extend", the error disappeared:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
// const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
`

Now the module loaded, but since the device only updates every 12h, I think I have to wait 12h for the exposes to show up?

MiAutomations, 2025-04-09:
Hello to all 

I now have the device work on the Z2M, but the data on this entities are Null:

Daily comsumption
Daily Water Comsumption

Monthly comsumption
Monthly Water Comsumption

Instantaneous flow rate

[image]

I'm using this file as a external converter ( in attach) 

Can some one please help me in order to fix this issue? 

Thank you in advance for the support and help 

[watervalve.txt](https://github.com/user-attachments/files/19669274/watervalve.txt)

MiAutomations, 2025-04-09:
> Used to get this error:
> 
> <error: z2m: Failed to load external converter file 'devices/water_flow_meter.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
> 
> Require stack:
> 
> * /app/dist/util/utils.js
> * /app/dist/util/settings.js
> * /app/index.js)>
> 
> After I commented out "const extend", the error disappeared:
> 
> <const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> 
> const exposes = require('zigbee-herdsman-converters/lib/exposes'); const reporting = require('zigbee-herdsman-converters/lib/reporting'); // const extend = require('zigbee-herdsman-converters/lib/extend'); const e = exposes.presets; const ea = exposes.access; const tuya = require('zigbee-herdsman-converters/lib/tuya');>
> 
> Now the module loaded, but since the device only updates every 12h, I think I have to wait 12h for the exposes to show up?
> 
> UPDATE: Yes, had to wait a couple of hours until the exposed datapoints showed up. The sensor only sends data every 12 hours. I currently see Water Consumed, Temperature and Linkquality

Hello @freeewilly, just to confirm, do you have the data from the Daily Water Comsumption, Monthly Water Comsumption, Instantaneous flow rate, populated in the entities? 

Tks

freeewilly, 2025-04-10:
I don't but it's not needed. Instantaneous Waterflow is useless as the device only reports back once per hour. Daily and Monthly can be achieved by setting up daily/monthly utility meters in Homeassistant, based on Water Consumed.

MiAutomations, 2025-04-10:
> I don't but it's not needed. Instantaneous Waterflow is useless as the device only reports back once per hour. Daily and Monthly can be achieved by setting up daily/monthly utility meters in Homeassistant, based on Water Consumed.

@freeewilly yes ok but I don't have either the data related to the Daily and Monthly water consumption... Can you please share your external converter file?

Thank you

freeewilly, 2025-04-11:
Like I said, you don't need daily/monthly consumption. All you ned is "water consumed" from which you can derive daily and monthly utility meters via helpers in HA

MiAutomations, 2025-04-11:
> Like I said, you don't need daily/monthly consumption. All you ned is "water consumed" from which you can derive daily and monthly utility meters via helpers in HA

Ok Tks. its not exactly the same... And if the device can give us these data I think it is better if possible. I will keep try to have all the data from the device 

Tks

[16 more replies]

closed 2025-08-01 by github-actions[bot]