Koenkk/zigbee2mqtt #21256: Frontend is without any css styling: "Could not find specified theme"

sdotter, 2024-02-06:
### What happened?

Running frontend from port 8099 port as configured in:

[image]

Results in frontend without any css styles etc:
[image]

[image]

### Zigbee2MQTT version

Latest

### Adapter firmware version

6.10.3.0

### Adapter

Sonoff dongle e

### Setup

Add-on homeassistant docker rpi x64

sjorge, 2024-02-06:
The frontend is in this repo https://github.com/nurikk/zigbee2mqtt-frontend

Best to file the issue there instead of here.

closed 2024-02-06 by sdotter

jehy, 2024-02-07:
Error description and solution here: https://github.com/nurikk/zigbee2mqtt-frontend/issues/1921