Koenkk/zigbee2mqtt #21261: tuya knob remote (ERS-10TZBVK-AA) not working

issy, 2024-02-06:
### What happened?

I paired my brand new tuya smart knob [(ERS-10TZBVK-AA)](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) to Zigbee2MQTT in Home Assistant. However, none of the attributes/actions are working. All show as unknown, and I get errors if I try to change the operation mode (the only value I can change for this device)
[image]
[image]

### What did you expect to happen?

I expected to be able to use the button presses and knob rotations in zigbee2mqtt and home assistant. I've seen people on the hass forums saying that this device worked for them with zigbee2mqtt, and there is a herdsman adapter for it, so I assume that either my device is faulty or shipped with a different/newer firmware than devices that the adapter was tested with.

### How to reproduce it (minimal and precise)

1. Buy this device on Aliexpress https://www.aliexpress.com/item/1005006022684200.html
2. Put it in pairing mode by holding the reset button for 10 seconds
3. Pair to Zigbee2MQTT
4. Twist knob, press button. Observe as nothing happens

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8

### Setup

Home Assistant on x86-64 baremetal

ivan-sirosh, 2024-02-12:
for me it works but partially.
I can not switch operation mode from command to event.

`info 2024-02-12 06:37:53MQTT publish: topic 'zigbee2mqtt/0xdc8e95fffe15d77b', payload '{"linkquality":128,"state":"OFF"}'
error 2024-02-12 06:38:09Publish 'set' 'operation_mode' to '0x6c5cb1fffe9a55c4' failed: 'Error: Write 0x6c5cb1fffe9a55c4/1 genOnOff({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (243))'
info 2024-02-12 06:38:09MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'operation_mode' to '0x6c5cb1fffe9a55c4' failed: 'Error: Write 0x6c5cb1fffe9a55c4/1 genOnOff({\"tuyaOperationMode\":1}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (243))'","meta":{"friendly_name":"0x6c5cb1fffe9a55c4"},"type":"zigbee_publish_error"}'`

gurmukhp, 2024-06-16:
I'm also experiencing the same issue. Registers fine in Z2M, but no events are emitted when button pressed or rotated.
[image]

wznazir, 2024-08-26:
Similar issue for me, I can use the single and double press buttons and also change operation mode using the 3 presses. 

But I dont see any feedback when turning the knob.

oriaj3, 2024-10-15:
Same here:
Version Zigbee2MQTT
= [1.40.2]
Coordinator type = EmberZNet
Coordinator firmware = 7.4.3 [GA]
Adapter = Sonoff ZBDongle-e

Error:
z2m: Publish 'set' 'operation_mode' to 'Boton_dormitorio_jairo' failed: 'Error: ZCL command 0xa4c1386a0b24206f/1 genOnOff.write({"tuyaOperationMode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '32166'.)'

ivdimitro, 2025-01-01:
Any new clues on the topic? I don't see events coming to mqtt and no updates in HA. The events are displayed in the ZB2MQTT device page.

closed 2025-03-11 by github-actions[bot]