Koenkk/zigbee2mqtt #21263: Aqara T2 LLKZMK12LM binding issue

mrbeckers, 2024-02-06:
### What happened?

Trying to use the relay module as a decoupled switch, binded to a z2m lightgroup. Binding doesn't work as expected. When I toggle a hardware switch connected to S1 nothing happens. Instead when I turn the light group on/off in home assistant that toggles the physical relay of the Aqara module. Seems as if the binding is made in the wrong direction.

### What did you expect to happen?

I expected that changing the physical input switch of the Aqara T2 would toggle the state of the light group between on and off.

### How to reproduce it (minimal and precise)

[image]
Bind one of the source endpoints to a light group destination.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

20220219

### Adapter

zStack3x0 slaesh

### Setup

Add on Home Assistant

HarvsG, 2024-04-25:
I don't think binding is possible for this device? If you had success or have information to suggest binding should possible then a lot of people on the forum would be interested to know.
https://community.home-assistant.io/t/zigbee-light-switch-with-decoupled-mode/403680/70

closed 2024-11-22 by github-actions[bot]

andrei-lazarov, 2025-03-25:
@mrbeckers Did you find a device that works for this usecase (bind dumb switch to smart lights)? Or did the Aqara get some useful updates?