Koenkk/zigbee2mqtt #21273: Sonoff E - boot loop after update above 1.35.1-1

kizio1000, 2024-02-07:
### What happened?

After update to 1.35.2-1 (or 3-1) z2m is in boot loop.

### What did you expect to happen?

z2m will start and work

### How to reproduce it (minimal and precise)

latest ha + sonoff E + z2m above 1.35.1-1

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

proxmox VM

### Debug log

error 2024-02-07 07:03:41: Adapter disconnected, stopping
debug 2024-02-07 07:03:41: Saving state to file /config/zigbee2mqtt/state.json
info  2024-02-07 07:03:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-02-07 07:03:41: Disconnecting from MQTT server
info  2024-02-07 07:03:41: Stopping zigbee-herdsman...
error 2024-02-07 07:03:41: Failed to stop Zigbee2MQTT

Balooforever, 2024-02-07:
Same here but only crash when updating or pairing..

eligao, 2024-02-09:
Mine works with the same firmware, in a x86 Linux box, not VM

klu16, 2024-02-11:
Same here. After updating Z2M to 1.35.3 with coordinator firmware 7.3.1.0 on Sonoff Dongle-E, it constantly crashes every ~2 minutes.

Setup:
HAOS (2024.1.5) on QNAP VM.

closed 2024-09-09 by github-actions[bot]