Koenkk/zigbee2mqtt #21274: LYWSD03MMC Tempearture Sensors not shown on Dashboard

Bloodpack, 2024-02-07:
### What happened?

LYWSD03MMC Tempearture Sensors not shown on Dashboard but shown on Devices

Dashboard screenshot
[image]

Devices screenshot
[image]

### What did you expect to happen?

to be shown on the Dashboard

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20231112

### Adapter

tubeszb-cc2652-poe-2023

### Setup

Addon on Homeassistant OS on Proxmox

moodyblue, 2024-02-07:
In my case the dashboard does not show devices connected through routers, but shows all devices connected directly.

sanyafifa, 2024-02-08:
the same problem

Bloodpack, 2024-05-08:
Problem solved for me with the latest update to [v 1.37.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.0)
[image]

closed 2024-05-08 by Bloodpack