Koenkk/zigbee2mqtt #21276: 1.35.3-1 Conbee II network - communication issues and timeouts

jacek-szymborski, 2024-02-07:
### What happened?

After updateng from 1.35.1-1 to 1.35.3-1 my network becomes unusable due to multiple communication errors. The network behaves the same way as on 1.35.2-1, issues are not resolved by 1.35.3-1.

### What did you expect to happen?

Stable network, as on 1.35.1-1

### How to reproduce it (minimal and precise)

Upgrade grom 1.35.1-1 to 1.35.3-1 (or 1.35.2-1)

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Setup

Supervised Addon on HA on Proxmox full VM

### Debug log

[addon_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14190974/addon_log.txt)
[herdsman_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14190975/herdsman_log.txt)

Damien2407, 2024-02-07:
Same issue in my network.
1.35.3-1 did not resolve communication issues with combee II

rchiileea, 2024-02-07:
same here and after todays update my webui looks like this 
[image]

Bitmasss, 2024-02-07:
Same issue for a while here, thought I was going crazy.
Is there any logs I can provide to help?

stonewayNL, 2024-02-07:
After install of 1.35.3.1 nothing worked. So i restarted z2mqtt and everything works fine. The frontend problems i solved by clearing the browser cache.
Till now no problems

donagrejja, 2024-02-07:
Same for me. Old issue for reference https://github.com/Koenkk/zigbee2mqtt/issues/21162

SilverPhoenix99, 2024-02-07:
It seems I've managed to simulate the issue on `latest-dev` (1.35.2-dev commit: [3d37892](https://github.com/Koenkk/zigbee2mqtt/commit/3d37892)).

This version should coincide with `1.35.3`. I'm not aware if there's any more recent fix for this.

Please find the logs here: https://gist.github.com/SilverPhoenix99/cc2de08d89e14df7688de7e917cea0c0

(`DEBUG=zigbee-herdsman*` is enabled)

GaupeLnx, 2024-02-07:
https://github.com/Koenkk/zigbee2mqtt/issues/21268#issuecomment-1931812465 Just commented on another issue that handles about some of the same that is mentioned here.  I'm putting a link to it here.

rchiileea, 2024-02-07:
my epc rating is based on a fact a we got that rating because there was a sheet in the loft stating the lost insulation was due to be done (are loft insulation is about 2cm thick...) the loft insulation was never completed.

BR
Richard B

josemaX, 2024-02-07:
Same here (also the UI issue in Home Assistant iPadOS (at least) application). Restoring  1.35.1-1 makes all work again.

DoubleZed, 2024-02-07:
Same here with Conbee II. Restoring back to 1.35.2 fixed it.

miguelpineiroo, 2024-02-08:
Same issue here. I have Conbee II, working well with 1.35.2.1, after upgrading to 1.35.3.1 all Zigbee down. Need to revert back to 1.35.2.1 and it works fine as before

closed 2024-02-08 by Koenkk