Koenkk/zigbee2mqtt #21277: Legrand Cable outlet  (064882) : failed to configure

MrJeromeV, 2024-02-07:
### What happened?

Had to restart my z2m installation.
Everything went well except for my two Legrand outlet : after pressing the reset button, I see the device join and left in Z2M. It never stays connected. 

### What did you expect to happen?

The device is connected in Z2M

### How to reproduce it (minimal and precise)

Try to pair a Legrand outlet on a Z2M fresh install.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

    7.3.1.0 build 176

### Adapter

Sonoff  ZBDongle-E

### Setup

Home assistant OS on Home Assistant blue (ODroid N2+)

### Debug log

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/14191098/z2m.log)

Koenkk, 2024-02-07:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

MrJeromeV, 2024-02-07:
[z2m_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/14202307/z2m_debug.log)

Koenkk, 2024-02-08:
Seems the device immediately leaves the network: ` Device '0x000474000025c675' left the network`, therefore configure fails. Does this work on z2m 1.35.1?

MrJeromeV, 2024-02-09:
Same problem with 1.35.1.
It worked with 1.33.2 when I paired them fort the first time.

Koenkk, 2024-02-10:
can you go back to 1.33.2 and provide the herdsman debug logging when pairing them?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Benji67, 2024-02-11:
Hi, i have got the same issue. After the update, i can't pair anymore with my legrand outlet.

Benji67, 2024-02-11:
I found this process (https://github.com/Koenkk/zigbee2mqtt/issues/20664#issuecomment-1893561624) works for me :)

MrJeromeV, 2024-02-12:
[zigbee_herdsman_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/14244726/zigbee_herdsman_debug.log)

Koenkk, 2024-02-13:
@MrJeromeV can you check if 1.34.0 and 1.35.0 also work?

Koenkk, 2024-02-15:
I think it has something to do with the zigbee-herdsman version.

Can you see if the following fixes the problem:
- Make sure you are running 1.35.0
- ssh into your system and execute:

```bash
docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/sh
apk add npm
npm install zigbee-herdsman@0.25.0
exit
docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
```
- See if the issue is solved now

Koenkk, 2024-02-18:
@MrJeromeV I see the wrong herdsman version is used: ` Starting zigbee-herdsman (0.30.0)`, make sure to not restart the addon via the HA UI as this will revert the changes.

MrJeromeV, 2024-02-23:
> I think it has something to do with the zigbee-herdsman version.
> 
> Can you see if the following fixes the problem:
> 
>     * Make sure you are running 1.35.0
> 
>     * ssh into your system and execute:
> 
> 
> ```shell
> docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/sh
> apk add npm
> npm install zigbee-herdsman@0.25.0
> exit
> docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
> ```
> 
>     * See if the issue is solved now
> 
>     * If that works, try increasing the zigbee-herdsman version until it breaks again, you can find all the version numbers here: https://github.com/Koenkk/zigbee-herdsman/releases

[zigbee_herdsman_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/14383472/zigbee_herdsman_debug.log)

I tried again.

In Z2M dashboard I see 
Zigbee-herdsman version    0.25.0
but log still prints 
Starting zigbee-herdsman (0.30.0)

Koenkk, 2024-02-24:
Can you try with the improved [ember driver](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)?

MrJeromeV, 2024-02-26:
Is it compatible with ezsp driver or will it break my network ?

Koenkk, 2024-02-26:
Yes it's for ezsp adapters.

MrJeromeV, 2024-02-28:
Very strange : it worked for one device, not for the other one.
Z2M edge 1.35-3-dev
zigbee_herdsman_version 0.34.8
ZBDongle-E 7.4.1 firmware
adapter ember

MrJeromeV, 2024-02-28:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14433810/log.txt)

closed 2024-09-26 by github-actions[bot]