Koenkk/zigbee2mqtt #21279: Aqara TRV (SRTS-A01) really new update 2590 ??

DaN66O, 2024-02-07:
### What happened?

Is there really a new update 2590 or is it a bug?

[image]
[image]

### Zigbee2MQTT version

1.35.3 commit: fe0742a6

### Adapter firmware version

20230507

### Adapter

zigstar CC2652

### Setup

plain

macintosh-HD, 2024-02-07:
I'm also seeing that update. I haven't updated yet as I was unable to confirm online that this update really exists and what the changelog is.

verticalsoft, 2024-02-08:
I've got these in my Aqara Hub there is no official update available...

chris400, 2024-04-07:
I am facing the same issue.

fireport, 2024-08-06:
Same problem here today

almirus, 2024-10-16:
still here

nesty1990, 2024-10-17:
I have updated my new trvs manufactured 09/24 which came with fw 1030 to this 2590 fw ... Now it reports fw 030 and z2m reports update to 2590 available again :/

fireport, 2024-12-31:
the problem is still here

wduda, 2025-02-12:
The problem is still here.

marcuscps, 2025-04-10:
Happening for me as well.

filewalker1, 2025-06-10:
still open

closed 2025-08-17 by github-actions[bot]