Koenkk/zigbee2mqtt #21282: ZY-M100-24G - Unable to change presence sensitivity setting.

DuncanRae, 2024-02-07:
### What happened?

When changing the Presence Sensitivity value for the ZY-M100-24G Presence Sensor from the default 2.5, it changes for a few seconds then goes back to 2.5.

There is no indication as to why in the logs. The value just changes back to 2.5.

This setting used to work, (in the last two or 2 months) but now doesn't. I don't often change these settings and have done a few Z2M updates so can't be sure which update broke it. It was definitely working in the mid-January time frame though. 

### What did you expect to happen?

The Presence Sensitivity value would be set to what I tried to set it to.

### How to reproduce it (minimal and precise)

Move the Presence Sensitivity value to anything other than 2.5

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB 

### Setup

Addon to Home Assistant

### Debug log

Debug 2024-02-07 13:41:46Received MQTT message on 'zigbee2mqtt/Main Bedroom Presence Sensor/set/presence_sensitivity' with data '4'
Debug 2024-02-07 13:41:46Publishing 'set' 'presence_sensitivity' to 'Main Bedroom Presence Sensor'

Debug 2024-02-07 13:41:53Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":2}],"seq":62464}' from endpoint 1 with groupID 0
Debug 2024-02-07 13:41:53Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":105}],"seq":62720}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:53MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":693,"linkquality":47,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":4,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:53Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":112}],"seq":62976}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:53MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":693,"linkquality":43,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":4,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:53Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":63232}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:53MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":693,"linkquality":47,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":4,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:54Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,178],"type":"Buffer"},"datatype":2,"dp":104}],"seq":63488}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:54MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":690,"linkquality":43,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":4,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:54Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,20],"type":"Buffer"},"datatype":2,"dp":106}],"seq":63744}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:54MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":690,"linkquality":47,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":4,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:54Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,25],"type":"Buffer"},"datatype":2,"dp":111}],"seq":64000}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:54MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":690,"linkquality":43,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":2.5,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:54Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,94],"type":"Buffer"},"datatype":2,"dp":107}],"seq":64256}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:54MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":690,"linkquality":40,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":2.5,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'
Debug 2024-02-07 13:41:54Received Zigbee message from 'Main Bedroom Presence Sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":110}],"seq":64512}' from endpoint 1 with groupID 0
Info 2024-02-07 13:41:54MQTT publish: topic 'zigbee2mqtt/Main Bedroom Presence Sensor', payload '{"detection_distance_max":3.5,"distance":0,"fading_time":10,"illuminance_lux":690,"linkquality":43,"motion_sensitivity":2,"move_sensitivity":2,"presence":false,"presence_sensitivity":2.5,"presence_state":"none","presence_timeout":10,"radar_range":3.5,"state":"none","target_distance":0}'

AlfAlex, 2024-02-07:
I have the same problem with device "[ZY-M100-24G]" (I have this device twice, the other one does not have this problem.

Skyhawker231, 2024-02-07:
+1. Also see #21230 and #19448
I also have this device twice one has these issues but the other works fine

DuncanRae, 2024-02-07:
I also have 2 devices. Both exhibit the same behavior. They sometimes take a few seconds to go back to 2.5, and I had it once on one device seem to stick for a few minutes, but then went back to 2.5 again.

DuncanRae, 2024-02-08:
I’ve done a bit more testing. All of the sliders are behaving in the same way. Presence Sensitivity, Move Sensitivity and Radar Range. They are all returning to default values a few seconds after changing them.

DuncanRae, 2024-02-08:
Not sure if I'm barking up the wrong tree here, and would appreciate input from someone more experienced, but when comparing the converter file (tuya.ts) in an older version that worked (pretty sure) to now, I see the following difference:

Worked (from version 18.10.0 of Herdsman)
`e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1).withDescription('Presence sensitivity'),`

Doesn't work (18.28.3 of Herdsman):
`            e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1)
                .withValueMax(10)
                .withValueStep(1)
                .withDescription('Presence sensitivity'),`

The not working one seems to be missing the withValueMin piece. I don't have a way to make this change and test so if this helps point someone who can in the right direction, that is great.

AlfAlex, 2024-02-08:
For me these lines do look identical... ;)

I did not recongize any problem with version 1.34.0 besides frequently crashes (the reason why I did the upgrade, also I had some prblems with another sensor).
I do have these problems now with version 1.35.2 and also with 1.35.1.
I will try 1.35.0 and 1.35.3 now

DuncanRae, 2024-02-08:
> For me these lines do look identical... ;)
>
> I did not recongize any problem with version 1.34.0 besides frequently crashes (the reason why I did the upgrade, also I had some prblems with another [...]

You're right! Sorry. They were laid out differently, and I missed it. Wrong tree. :(

Thanks for the help.

AlfAlex, 2024-02-08:
1.35.3: same bug
1.35.0: same bug
1.34.0: same bug

So it looks like I had this problem before but did not reconize it

gustjoha, 2024-02-09:
Same problem for me  - values revert to default after a few seconds.

SonTypda, 2024-02-09:
Almost the same problem for me. I also mentioned it here: https://github.com/Koenkk/zigbee2mqtt/discussions/21180

For me, the Presence sensitivity value always jumps back to the highest level (10)

As a result, my network is completely spammed because someone in the room is constantly and erroneously detected at this sensitivity level

Skyhawker231, 2024-02-09:
See #19448 
Once I switched to the external converter in Fehuz's comment all my issues were resolved regarding not being able to change any settings as well as the many issues running multiple presence sensors at the same time. Would definitely recommend giving it a try and hope that it can be added natively within Z2M

DuncanRae, 2024-02-09:
> See #19448 Once I switched to the external converter in Fehuz's comment all my issues were resolved regarding not being able to change any settings as well as the many issues running multiple [...]

Thanks. I have done this and is seems to be all good now. I change presence sensitivity from 2.5 to 4, and it went to 0.4, but then changed to 4 a few seconds later and it seems to have stuck!

AlfAlex, 2024-02-09:
Working for me too. 
For me it looks like a serious bug in this deviiveBy10 converter stuff.

Koenkk, 2024-02-10:
Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-10 by Koenkk

monqui-ulia, 2024-08-08:
It seems that in version z2m 1.39.1 (zigbee-herdsman-converters version
19.72.0) the error persists. If you set the presence sensitivity parameter to 8 it changes to 0.8. Same behavior with move sensitivity.

inertia666, 2024-09-10:
Move and presence sensitivity reset to 0 on the Tuya ZY-M100-24GV2 using z2m 1.40.1-dev

201505409, 2024-09-20:
I am having the same issues on my ZY-M100-L

zimgui, 2024-10-22:
Same issue here, I have 6 ZY-M100-24GV3 and randomly reset their values and every time there is a power lost.

dasEtwasmitPunkten, 2024-11-06:
@Koenkk I recognized the same behavior with my ZY-M100-24G. The sensitivity values are set to 20 and 40 (out of range) instead of 2 and 4 and I am unable to change it. Is it possible to check this issue again?

Servizio-R, 2024-11-24:
ZY-M100-24G, same behaviour, set sensitivity to X and few seconds later it jumps to 10 times X (2->20, 8->80 etc)
please check, tx

Machaeon, 2024-11-28:
> few seconds later it jumps to 10 times X
Same here

Mark-Muc, 2024-11-30:
i have the same problem and own 2 devices. I found out, that if the "linkquality" variable reported from ZY-M100-24GV3 jumps very often, so much "traffic" is on the MQTT link also the ZY-M100-24G hangs after a while and stay on "presence": true,.
Is there any way to reduce the amount of reporting, esp.  "linkquality": 132, to 1 -3 seconds .. this will reduce the reporting .. 
and the device get more stable.
from 1 sec. log:

```
info 2024-11-30 11:56:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"yok70-2"}'
info 2024-11-30 11:56:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1239,"linkquality":136,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":136,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.4,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":136,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.4,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":132,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.2,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":132,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.2,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":132,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.2,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":132,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300,"state":"none"}'
info 2024-11-30 11:56:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.2,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload '{"detection_distance_max":4.5,"detection_distance_min":1.5,"distance":0,"find_switch":"ON","illuminance_lux":1245,"linkquality":132,"move_sensitivity":2,"presence":false,"presence_sensitivity":2,"presence_timeout":300," [... cut ...] ut":300,"state":"none"}'
info 2024-11-30 11:56:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -1-', payload '{"detection_distance_max":3,"detection_distance_min":0.75,"distance":1.2,"find_switch":"ON","illuminance_lux":303,"last_seen":"2024-11-28T12:48:09+01:00","linkquality":255,"move_sensitivity":4,"presence":true,"presence_sensitivity":4,"presence_timeout":120,"state":"presence"}'
info 2024-11-30 11:56:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Praesenzmelder -2-', payload

```