Koenkk/zigbee2mqtt #21286: [Wrong device]:

bruvv, 2024-02-07:
### Link

asdsad

### Model

asdsad

### Description

asdasd

### Vendor

asdasd

### Picture (link)

sad

### Database entry

asdasd

bruvv, 2024-02-20:
@catduckgnaf thanks for checking, I do not believe so. There is also no firmware update. I do however have them paired in a group with philips hue lights.
[image]

catduckgnaf, 2024-02-20:
It might be related. see if it goes away without hue lights in the group?

catduckgnaf, 2024-03-02:
> @catduckgnaf I tested it and it does not make any difference

Interesting, thank you

fabianteunissen, 2024-05-21:
Hi, I've encountered the same issue with these light bulbs. Is there a fix or workaround available yet?

pomeloy, 2024-08-22:
Still an issue sadly. TS0505B-based [Lidl HG07834C](https://www.zigbee2mqtt.io/devices/HG07834C.html) light bulbs are also affected.

RubenKelevra, 2024-10-13:
Happens here too, but only on very low brightnesses. Are we sure that this is not a firmware bug?

pomeloy, 2024-10-17:
Probably not, as this behavior seems not to be observed using the Lidl/TuYa gateway or ZHA.

RubenKelevra, 2024-10-17:
Okay, I've checked one of my Lidl lights and found the following abnormal behaviors and found the following.

- Turn on with transition 0, brightness value: 1
Light will turn on normally
- Turn light off, with transition 0 or 1
Light will report back, that it's off, but stay on.

Sending a Turn_on command to this light in this state, will make it dim either immediately up (`transition: 0`) or slowly up (`transition: >0`), and then turn off.

The brightness value for this need to be 3 or higher.

So, looks like brightness 1 and 2 are broken, and you can't turn the light off with them.

Since brightness values 1 and 2 are below 2.55, I think it's safe to assume that the light can only handle percentage values, and as 2 is below 2.55, it “thinks” that it's at 0% brightness.

---

What I think is happening here, is that the lights are turned on with a brightness value of 1 or 2 and then get a command to dim up. This is causing the lights (sometimes) to flicker off again. I could not replicate this behavior here, but I'm pretty sure that's what's happening.

So as @Koenkk suggested [here](https://github.com/Koenkk/zigbee2mqtt/issues/23802#issuecomment-2325277397) this bug could be handled by putting an...

``` 
if user_requested_brightness > 0:
  user_requested_brightness = max(user_requested_brightness, 3)
``` 

... somewhere in the converter code.

RubenKelevra, 2024-10-17:
Btw, I can confirm this bug also for [HG08010](https://www.zigbee2mqtt.io/devices/HG08010.html) and [HG07834A/HG09155A](https://www.zigbee2mqtt.io/devices/HG07834A_HG09155A.html#lidl-hg07834a-hg09155a)

Koenkk, 2024-10-18:
What if not using any transition at all? Does it always work then?

RubenKelevra, 2024-10-18:
@Koenkk no, it doesn't.

So if you turn it on with brightness steps1-2 and turn it off, it will stay on.

This is also true if you turn it on with any brightness and just dim it down to 1-2 brightness steps and then turn it off.

Lamp will report back "I'm off" and stay on.

If you now send a turn on with a higher brightness, 3-255 and transition 0, it will transition nearly instantly and turn off again.

If you send a longer transition it will transition slower to this brightness and then turn off again.

pomeloy, 2024-10-18:
@RubenKelevra try sending the off command twice in a row. The next time turning on the light will have it flicker on and off again. See https://github.com/Koenkk/zigbee2mqtt/issues/14714#issuecomment-1374600038 The brightness 0-2 bug is separate from this one IMO.

RubenKelevra, 2024-10-18:
> try sending the off command twice in a row. The next time turning on the light will have it flicker on and off again.

Ah thanks!

But it's more complicated, as both seems to be related: If the brightness 1-2 bug occours the lights will turn off again (physically), but if instructed to turn on again with a higher brightness setting, they will turn off (physically) but stay on. If instructed to turn off in this state and turn on again, they _might_ flicker with a low chance.

So in commands:

``` 
turn_on brightness=1 or 2
turn_off
# light doesn't turn off physically, reports back: off
turn_on brightness >= 3
# light turns flickers / transitions on and turns back off (physically), reports back: on
turn_off # to clear the *on* status
turn_on brightness 1-255 
# light will flicker with lower than 100% chance
``` 

The bug you described is:

```
# light is on normally with brightness 3-255
turn_off
# light turns off and reports back: off
turn_off
# light stays off and reports back: off
turn_on brightness 3-255
# light flickers, reports back: on
``` 

Main issue here is, that the light will report back: on, while it has flickered. Meaning once in this mode, it will basically stay there because:

```
# light is on normally with brightness 3-255
turn_off
# light turns off and reports back: off
turn_off
# light stays off and reports back: off
turn_on brightness 3-255
# light flickers, reports back: on
user switches the light, which is reportedly "on" in HA, off and on again:
# light flickers, reports back: on
user switches the light, which is reportedly "on" in HA, off and on again:
# light flickers, reports back: on
user switches the light, which is reportedly "on" in HA, off and on again:
# light flickers, reports back: on
user switches the light, which is reportedly "on" in HA, off and on again:
# light flickers, reports back: on
```

RubenKelevra, 2024-10-18:
I tried contacting Lidl once again, this time via mail, asking for a contact for technical issues with their Zigbee firmware.

But I don't have high hopes. I think we should instead add workarounds here:

```
If light_status=off and command=turn_off:
   skip command
``` 

and 

```
if user_requested_brightness > 0:
  user_requested_brightness = max(user_requested_brightness, 3)
```

Should hopefully fix both issues.

RubenKelevra, 2024-10-19:
Update: Lidl responded today, and asked me if I want more broken lights as replacement 🥲

I've insisted that I want a technical contact to discuss the firmware issue instead.

RubenKelevra, 2024-10-19:
Update: I got a technical support email address by Lidl and contacted them regarding both issues in their firmware.

bruvv, 2024-10-19:
Thanks for the work @RubenKelevra ! I must say that when using ZHA there is no issue with the lights, so it is some what z2m related I guess? Looking forward to what lidl comesup with!

RubenKelevra, 2024-10-19:
No, I think ZHA may keep track of the "off" status and won't send another "off" command to the light, if it already reported itself as off.

This can avoid sending duplicate commands to the lights, say if the light is in a HA group and some are off and some are on, and you send a turn-off to the group. I guess ZHA would skip the off lights, while Z2M sends an off command anyway.

Normally turning an off light off, should not cause any harm and has the advantage that if we missed an update by the light, and it is actually on, we can still turn it off.

However, the `brightness=1` and `brightness=2` bug should exist too, on ZHA. But you need dev tools or automations for it, as the normal GUI would just let you dim lights down to `brightness=3`, which is 1 %.

Would quickly test this, but last time I had some annoying bugs switching back and forth and had to reconnect a couple of Zigbee devices to my network. I would like to avoid that ;)

bruvv, 2024-10-20:
Ah gotcha, i am runnnig zha atm so I can test if you want me to test anything

RubenKelevra, 2024-10-20:
@bruvv sure!

- Turn everything off which may control the light, like Adaptive Lighting or automations.
- Make sure to turn the Lidl light off normally in your GUI.
- Then go Development Tools -> "ACTIONS"
- Select light.turn_on, select a single Lidl light.
- Expand the view on the bottom, select "Brightness Value" and select `1`, also add "Transition" `0`.
- The light should turn on, but very faint.
- Then select light.turn_off from the action list, and run this against the light again. It should stay on.
- Go into your normal GUI, and the light should report, that it's "off" in the GUI.
- Go back to Development Tools -> "ACTIONS".
- Now select light.turn_on again from the action list, and run it against the same light, again with "Transition" `0` but with "Brightness Value" and select `3`.
- It should flicker on and then turn off again.

RubenKelevra, 2024-10-21:
The "technical support" answered and asked random questions about which "App" I'm using and which "Mobile Phone".

Well, I've answered as best as I could, and pointed again to the fact that the lamps are:

- Reacting to Zigbee commands itself wrong
- The coordinator is not responsible, as this issue occurs with different coordinators
- My mobile phone doesn't matter, as it's not involved
- I'm using no app for this, as the Zigbee commands itself are interpreted wrongly.

I still doubt that this will lead to anything, but I mean it's worth a shot I guess.

bruvv, 2024-10-22:
@RubenKelevra when using zha and to this step:
Then select light.turn_off from the action list, and run this against the light again. It should stay on.
It turns off and also shows this in home assistant and the other steps i did too but perform normally and I have no flicker.

RubenKelevra, 2024-10-22:
@bruvv which coordinator do you use?

Adastreia, 2024-12-31:
I have this issue with both [HG07834B/HG08131B](https://www.zigbee2mqtt.io/devices/HG07834B_HG08131B.html#lidl-hg07834b%252Fhg08131b) and [399629_2110](https://www.zigbee2mqtt.io/devices/399629_2110.html#lidl-399629_2110). I'm using z2m with skyconnect, while waiting for my SMLIGHT SLZB-06. 

Leaving this comment mostly to get an update on the question @Midifreakz posed. 

If there is anything I can do on my end to test or maybe send emails to Lidl I'd be happy to. @RubenKelevra did you write to them in german or english?

RubenKelevra, 2024-12-31:
@Adastreia 
> If there is anything I can do on my end to test or maybe send emails to Lidl I'd be happy to. @RubenKelevra did you write to them in german or english?

Hey!

I wrote them in German. However I doubt it would be an issue to write them in English.

I contacted Lidl, they gave me a contact mail address at Optimex, which is their partner for replacing lights if they are broken.

However, I eventually ran out of patients talking to Level 1 support. 🙄

In the meantime I found out, that neither Optimex nor Lidl actually imports these lights. So maybe there's an option to instead contact the company who does this instead?

That would be "OWIM GmbH & Co, Stiftbergstraße 1, 74167 Neckarsulm" according to the Datasheet provided for the lamps by Lidl.

Adastreia, 2025-01-23:
I found this website [Owim.com](https://www.owim.com ) and I will send them an email this weekend. 

Meanwhile, mostly for myself, I tested my lamps as instructed by @RubenKelevra 

I fold this excursion as it might be out of scope for this issue, but I thought it might shed light on how inconsistent behavior these bulbs show, at least it did so to me. 

<details><summary>Excursion</summary>
<p>

### I wanted to see if it had something to do with `color_mode` and `color_power_on_behavior`. 

If the bulb had previously been on in `color_power_on_behavior": "previous"` and had`"color_mode": "color_temp"` and I changed to a scene with `xy`-values  it would turn on with `color_temp` and then go _much_ slower than the others to `xy`. But more often than not, not responding at all and stay in `color_temp`. It also has a tendency to show in HA logs as "unknown" just as this happens. Lately it wont turn off either. 

Turns out that one (1) of my 2 Livarno Home E14 candle RGB+CCT (HG07834B/HG08131B) was consistent. Even when failing at the last step, it did so consistently. No matter which combination of `color_power_on_behavior` and `color_mode` it was in.

|HG07834B/HG08131B |  |  |
|:--|:--|:--|
| light.turn_on brightness 1 | ✅  |  Just not very faint |
|  light.turn_off | ✅ |  |
| light.turn_on brightness 3 | ❌ | Turns itself off. GUI says "ON" |

The others were all over the place and I will put some tables below.

### Livarno Home E27 bulb RGB+CCT
HG07834C - TS0505B

| initial/previous/customized | xy  | xy | xy | color_temp | color_temp | color_temp |
|--------|--------|--------|--------|--------|--------|--------|
|light.turn_on with brightness 1 | Blinks once then off | Blinks once then off  | ✅ | ✅ | ✅ | ✅ |
| light.turn_off | Stays off GUI "OFF" | Stays off GUI "OFF"  | ✅  | Turns itself off.  | ✅  | ✅ |
|  light.turn_on with brightness 3 | ✅ | ✅ | ✅ | Turns on, stays on | Turns off| ✅ |

### Livarno Home E14 candle RGB+CCT
HG07834B/HG08131B - TS0505B

| initial/previous/customized | xy  | xy | xy | color_temp | color_temp | color_temp |
|--------|--------|--------|--------|--------|--------|--------|
|light.turn_on with brightness 1 | Blinks once then off |  ✅ | ✅ | ✅ | blinks and stays off | blinks and stays off |
| light.turn_off | Stays off GUI "OFF" | ✅    | Turns off  | ✅ | Stays off  | Stays off |
|  light.turn_on with brightness 3 | ✅ | Turns off | Turns on, stays on | Turns off | ✅ |Turns off|

### Livarno Lux Ceiling Panel RGB+CCT
399629_2110 - TS0505B

| initial/previous/customized | xy  | xy | xy | color_temp | color_temp | color_temp |
|--------|--------|--------|--------|--------|--------|--------|
|light.turn_on with brightness 1 | Turns off after 3s. Brightness 3 |  ✅ (with brightness 3) | ✅ | ✅ | ✅ | Turns off|
| light.turn_off | Stays off, GUI "OFF" |  Turns off   | ✅  | ✅ | ✅   |  ✅ |
|  light.turn_on with brightness 3 | ✅ | ✅ |✅ | ✅  | ✅ |✅ |

</p>
</details> 

I learnt a few things by this!

- They do not always take the right brightness-value. Sometimes it would be 8 when I sent 3. Sometimes 3 when I sent 1. 
- Rather than flickering mine more often than not just turn off.
- Brightness in a lightcard on the dashboard doesn't always correspond to the brightness-value in z2m. 
- The brightness-value on the "state"-tab in z2m does not update if you change device in the dropdown. 
- If the lamp was still on after light.turn_off, light_turn_on will turn it off.
- When it was already off cause it failed light.turn_on with brightness 1, then it just blinks quickly and stays off.
- Switching to a tab in the z2m UI made one light actually take the command and start.

RubenKelevra, 2025-01-24:
@Adastreia nah that's not it. I've looked at the spectrum of the RGB mode, it's clearly 3 distinct LEDs visible, while the white mode is two different LEDs. So it's RGB+CCT not RGBWW or RGBW. Apart from that the CRI of 98.2 is too high for an RGBWW/RGBW bulb.

You can also look into the light bulb an see the individual LEDs switching, if you switch from the warmest color to RGB White and the coldest color to RGB White.

From what I can tell I think it's simply a bug in the firmware where the Zigbee status and the LED status is somewhat diverging, and thus the transition is wrongly calculated - that's what all those bugs seems to point at.

Adastreia, 2025-01-25:
I think this is where I got this thought: [https://community.home-assistant.io/t/lightify-and-scenes-just-want-warm-white/160501/11?u=cybele](url) It's true for [HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html#lidl-hg06104a) from Lidl at least, but it might be that these are newer versions of bulbs that act different.

> This happens also with the new Lidl smart RGB bulbs: Colored lights on scenes work great. But for actual white tones you want to adjust the the color temperature slider because it gives you many times more light by using the white leds instead of composing the color with the RGB leds)

RubenKelevra, 2025-01-25:
@Adastreia which "thought"?

The spectrum graphs do proof that it's an RGBCT so not sure why we're discussing this further here.

Adastreia, 2025-01-25:
My bad, it was a sidetrack for where my hypothesis came from.

> Apart from that I'm the one who fixed the description of the device in question you've linked, as the information about it being an RGB+CCT were missing before.

No way? Amazing! 🤩

Adastreia, 2025-01-27:
So I wrote to Lidl firstline customer support about the lights. I got a response saying I would be contacted by  T.D.M but instead I got contacted by OWIM service. 

When gathering the information for them I looked at the bulb itself for the model and realised that one of my HG09155B  (E14) lights falsely reports to the Z2M UI that it is a HG08131B.  

And two that in the UI are HG08131C are marked HG09155C on the bulb.

RubenKelevra, 2025-01-28:
@Adastreia the only difference between the HG09155B and the HG08131B are different generations of LEDs inside the bulb. But from the chip and the firmware they are identically. 

So all bugs you see in the HG09155B will be found in the HG08131B firmware as well. 

Btw: I've added the HG08131B and the other variants (like *C and *A) had previous models which used the same hardware/firmware as well. So I suspected that it would show up.

Will add your model later, so it will show up as addition in the list of possible bulbs.

Adastreia, 2025-02-01:
I got an answer from Lidl, not sure if it's of any use:

> Have you installed the firmware update?
> After checking with Tuya, they have released an update version 1.1.3 on Google playstore on October 15th to fix the issue.
> 
> Especially for Samsung Galaxy phone, please inform the users follow before the procedure to reinstall this APP.
> 
>  Step by step:
> - Uninstall Lidl Home APP ->
> - turn the phone off and on again,
> - download the new version (1.1.3) of Lidl Home APP from Google Playstore
> 
> To check the version: My area-> Settings-> About-> Current version

I replied explaining that don't have an android and asked for a url to download the firmware instead. I don't think they will answer but thought it would be worth a shot :)

RubenKelevra, 2025-02-01:
I mean it goes into the right direction: Those devices are AFAIK tuya devices, so it could very well be that there's a firmware update available via the android app.

From their response it sounds like the Android app may be what's fetching the updates and not the Zigbee hub... which means we could extract the URL from the Android app 🤔

Adastreia, 2025-02-11:
Now I also got a pair of lamps in exchange that are working as badly as the others.. was it possible to extract the url from the android app?

wise-rice, 2025-02-11:
What does it have to do with the Android app?
I used the **iOS** Tuya app (with the newer LIDL Gateway, the one that has HomeKit) before to install firmware updates on LIDL lights. The Firmware updates were offered/installed automatically.

But only two types of LIDL devices, ever offered a firmware update for me:
- 2m RGB+CCT light strips (TS0505A by _TZ3000_riwp3k79 "profile_id": "0x0104",      "device_type": "0x010d") Latest Firmware: 1.1.4 (up from 1.1.0 on delivery). The device needed a reset and reconnect after the update to show correct color_temps again.
- Wall Plug (TS011F by _TZ3000_j1v25l17     "profile_id": "0x0104",      "device_type": "0x010a") - this device type even performed succesful OTA Firmware updates through zigbee2mqtt, so there must already be a connection to some LIDL servers. Maybe @Koenkk knows how to add the light strip Firmware, as well?

I also have most other LIDL lamps, but they are still on the same Firmware, that they came with in 2023.
- RGB+CCT E27 Bulbs (TS0505B by _TZ3000_qd7hej8u        "profile_id": "0x0104",      "device_type": "0x010d")
- CCT E27 Bulb (TS0502A by _TZ3000_49qchf10     "profile_id": "0x0104",      "device_type": "0x010c")
- 30W Outdoor Floodlight RGB+CCT (TS0505B by _TZ3210_umi6vbsz      "profile_id": "0x0104",      "device_type": "0x010d") Still on Firmware 1.0.0
- E14 RGB+CCT Spotlights (TS0505B by _TZ3000_quqaeew6    "profile_id": "0x0104",      "device_type": "0x010d",)
- E14 CCT Spotlights (TS0502A by _TZ3000_el5kt5im    "profile_id": "0x0104",      "device_type": "0x010c",)

My lights also often turn on again, directly after being turned off (especially if using a LightLink Dimmer, that does not go through the Coordinator). I suspect the issue to be in my (89 devices large) zigbee network, which might send information twice? Kinda hard to find the culprit.

Adastreia, 2025-02-12:
> From their response it sounds like the Android app may be what's fetching the updates and not the Zigbee hub... which means we could extract the URL from the Android app 🤔

But if you didn't get an update of your lights, chances are they just wanted to make sure I had the latest update.

RubenKelevra, 2025-02-12:
> My lights also often turn on again, directly after being turned off (especially if using a Zigbee Group Dimmer, that does not go through the Coordinator, I think it uses Touchlink ?). I suspect the issue to be in my (89 devices large) zigbee network, which might send information twice? Kinda hard to find the culprit.
> 
> I also use AdaptiveLighting, which had serious issues to work with z2m.

This is likely the bug where the lights does not update their status that they turned off. So adaptive lighting is just sending the next brightness/color temperature adjustment and turns them on again, as home assistant's status still shows them as "on".

Try sending the commands through home assistant, this should fix this issue.

RubenKelevra, 2025-02-12:
> Bonus: ZHA can also turn on the lights instantly with the correct color and brightness from an off state.

Isn't that the same as sending a turn-on command which is intercepted by Adaptive Lighting? 🤔

Or will the lights act differently with ZHA?

ZHA is not an option, as only 80% of my devices work with ZHA, and some which showed as supported had really quirky bugs.

But it's worth exploring what ZHA does differently, so we can modify z2m's zigbee commands send to the light.

wise-rice, 2025-02-12:
> > What does it have to do with the Android app?
> > I used the **iOS** Tuya app (with the newer LIDL Gateway, the one that has HomeKit) before to install firmware updates on LIDL lights. The Firmware updates were offered/installed automatically.
> 
> Yeah but does the Android/iOS app download the firmware and pushes them to the hub, or does the hub download the firmware itself?

I assume the hub downloads and performs the updates.  I think I force closed the app during some updates and they continued. Updates are performed after some no-interaction timeout and the lights needed to be turned off.
LIDL lamps don't have a direct connection do the app, there is no Bluetooth, AFAIK.

 

> This is likely the bug where the lights does not update their status that they turned off. So adaptive lighting is just sending the next brightness/color temperature adjustment and turns them on [...]

I always sent the light-off commands through homeassistant (even when using the group dimmers I programmed an additional light-off command through a HA automation).

The problem was also the current state of brightness and color-temp. Those were never accurate. If I set a new Color Temp or Brightness for a whole group, the lights would be the same in reality, but HA would show different levels, as if some din't update.
Here is an example:

[image]
The first Bulb is exactly the same type and firmware as the last bulb, but it did not update,

When you first connect the LIDL lamps to the coordinator, z2m it will not even establish a reporting for onOff or CurrentLevel or ColorTemperature. The only reporting that is set up (but with a 5 sec minimum interval) is "currentHue", which relates to xy-color (RGB) but not the CCT part of the lamp (the only part I use in everyday life is ColorTemp).
Is this intentional @Koenkk ?

Also bindings: Not all clusters are bound. Only a few.

Is there another way that z2m is getting onOff / Brightness / ColorTemp updates from lamps? Are they polled? At which rate? Would this explain the delays and deviating statusses?

wise-rice, 2025-02-12:
> > Bonus: ZHA can also turn on the lights instantly with the correct color and brightness from an off state.
> 
> Isn't that the same as sending a turn-on command which is intercepted by Adaptive Lighting? 🤔
> 
> Or will the lights act differently with ZHA?
> 
The light will act very much differently.

If you use ZHA and "Enable enhanced light color/temperature transition from an off-state" the light turn on command is correctly intercepted by Adaptive Lighting (or any command which specifies a turn-on color and brightness), and lights will instantly turn on with the desired brightness and ColorTemperatur.  
It replicates the Philips Hue behaviour when turning on a device (to work around Zigbee devices ignoring colour changes when off).  [guspaz](https://www.reddit.com/r/homeassistant/comments/17efki5/what_do_the_zha_settings_actually_do/) describes the method behind it as:
- instant transition to 1% brightness
- instant transition to change colour
- then normal transition to specified brightness. 

The behavior I see is an instant turn on with correct settings (E.g. 1% at 2000K), no turn-on transition, no matter what the turn-off brightness/temp was. This is perfect.

If you use z2m: lights turn on with the last brightness and ColorTemperature settings. No matter if you set the turn-on behaviour to "previous" or "customized".
Adaptive Lighting cannot intercept the light turn on command at this stage. It will need a second or two to react. I'm already blinded by my ten 30W floodlights, which I turned off at 100% and 6500K, last time.

But z2m also seems to have this functionality, but it is hidden and needs to be enabled by setting the color_options.execute_if_off using a command 
Sadly, I don't know how to send this command or automate it (HAOS installation/docker environment without access to root startup scripts) - so I never tested if z2m's "execute_if_off" would work as well as ZHA's implementation. 
It would be awesome if a setting could be added to the z2m GUI.

>> ZHA is not an option, as only 80% of my devices work with ZHA, and some which showed as supported had really quirky bugs.
> 
> But it's worth exploring what ZHA does differently, so we can modify z2m's zigbee commands send to the light.

Agreed, ZHA is inferior when it comes to the number of supported devices, but it is more reliable when it comes to showing the correct status. it is also more user-friendly with some general features. Also e.g. the possibility to disable "Group members assume state of group" to make sure that all lights update their status before the group status is adjusted. Again, z2m can probably do it using configuration.yaml and setting optimistic: false (and maybe off_state: all_members_off). But a new user would have to do a lot of research to find this setting and have correct (but slower?) group reporting...

RubenKelevra, 2025-02-12:
> > Yeah but does the Android/iOS app download the firmware and pushes them to the hub, or does the hub download the firmware itself?
> 
> I assume the hub downloads and performs the updates. I think I force closed the app during some updates and they continued. The lights needed to be turned off. I force-closed the app, because updates are performed after some "no-interaction timeout" of a few minutes. LIDL lamps don't have a direct connection to the app, there is no Bluetooth, AFAIK.

Well, see, that's the problem, it's an assumption. The app could very well also download the firmware and then send them to the hub. Keep in mind, those firmwares are tiny.

And yes, obviously the app does not push the firmware directly to the lamps, but it could instruct the hub with an firmware image to do an update. We don't know.

I assumed that the hub does this on its own, but the message from the support made me curious.

RubenKelevra, 2025-02-12:
> Yes, good point (but it really should also work if you turn off the lights through a hardware switch or a lightlink dimmer). I always sent light-off commands through homeassistant - even when using the lightlink/group dimmers. To (try and) avoid the issue, I programmed an additional light-off command through a HA automation. This leads to other problems, as lightlink is faster than HA/z2m can update. Therefore, sometimes lights turned on and immediately off, again. I needed more complex automations to use the most simple of functions: to turn on/off my lights / z2m light groups.
> 
> The problem was also the current state of brightness and color-temp. Those were never accurate. If I set a new Color Temp or Brightness for a whole group, the lights would be the same in reality, but HA would show different levels, as if some din't update. Here is an example:

This would explain my issues with an automation 🥲

I'm using a color temperature and brightness sensor for closed loop control of the brightness and color temperature of the room, and had issues with the light not correctly "storing" a change in color temperature.

But now it would make sense, that the light just didn't updated it's state (yet) so I read the old color temperature.

I'm currently tinkering a lot with lights, as I would like to add functionality like that closed control loop into adaptive lighting and similar things. Just takes much longer than expected 🙄

RubenKelevra, 2025-02-12:
> I also like the global ZHA setting "Group members assume state of group" for all groups. I disabled it to make sure that all lights update their status before the group status is adjusted. Maybe it is slower? But I accept it for the increased accuracy (all lights should turn off in reality and then report back this state).
> z2m can probably do it using configuration.yaml and setting
> "optimistic: false" for each group (and maybe "off_state: all_members_off").
> But a new user would have to do a lot of research to find this setting in the .yaml to get 100% correct (but slower?) group reporting...

But there's no reason we can't adapt the better behavior to z2m, too. I don't think the behavior here is by choice, but just based on a lack of test devices for each and every device out there.

So it's in the hand of those with the devices to work on the support, at least as a feedback provider, to make them work correctly. :)

wise-rice, 2025-02-12:
> But there's no reason we can't adapt the better behavior to z2m, too. I don't think the behavior here is by choice, but just based on a lack of test devices for each and every device out there.
>
> So [...]

Okay, multiple issues here:

1) Fix the states of LIDL (Tuya) lamps --> make them report their state faster. For example by adjusting the official [z2m zigbee/herdsman converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lidl.ts) 
--> automatically set up reporting for    ) Onoff      )ColorTemperature      )Brightness 
--> automatically set up a minimum reporting intervall of "0" and at minimum change "1" --> The lamp will immediately report a change, if there is a change
--> bind the necessary clusters by default. Right now the z2m [binding documentation ](https://www.zigbee2mqtt.io/guide/usage/binding.html#state-changes)mentions:
`Any manual setup reportings of the clusters genOnOff, genLevelCtrl lightingColorCtrl and closuresWindowCovering will be removed if there are no binds to the device or group a device is in when unbinding. You have to setup these reportings again.`

2) Expose "color_options" (and maybe also ["level_config" without sending an Onoff Cluster](https://github.com/Koenkk/zigbee2mqtt/discussions/13015#discussioncomment-3068729) ) for LIDL lamps, as they support it well. 
Make [this procedure](https://github.com/basnijholt/adaptive-lighting/issues/1022#issuecomment-2218830648) more easy by adding a global option to the UI to "execute_if_off": "true" (like ZHA implemented it)
2.1) Maybe add an option to Adaptive Lighting to keep adjusting the colorTemps of lamps that are "off"? This way, the lights would always turn on with the adapted color (no matter HOW you turn them on - e.g. with a lightlinkDimmer or wall switch)

3) Add a global option to the z2m UI to adjust the "optimistic" behaviour for groups (maybe even all lamps) --> more correct status and automations --> avoid a "ghost" turn-on or off

4) Find server / Firmware files to perform LIDL OTA Firmwares through z2m and/or ZHA --> LIDL / TUYA support / search in the Android app code

wise-rice, 2025-02-12:
Check out what @sjorge is [doing](https://github.com/Koenkk/zigbee2mqtt/discussions/13279#discussioncomment-4726409) 

@Koenkk Can we please find a way to add these settings as defaults and use similar reporting timings for LIDL lamps?

"cluster":"genOnOff","attribute":"onOff","minimum_report_interval":0,"maximum_report_interval":3600,"reportable_change":0

"genLevelCtrl","attribute":"currentLevel","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1

"cluster":"lightingColorCtrl","attribute":"colorTemperature","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1

> Just wait until one of your models triggers and update and you have to do it again -_-
> 
> I just run this after every z2m update:
> 
> ```
> #!/usr/bin/bash
> 
> ## Bulbs
> ## <type=c,ws,cws> <brand=g,h> <name>
> # cws h light/bathroom/mood/bulb
> # cws h light/office/mood/bulb
> # cws h light/livingroom/mood/bulb1
> # cws h light/livingroom/mood/bulb2
> # ws g light/balcony/main/bulb
> # cws h light/bedroom/main/bulb1
> # cws h light/bedroom/main/bulb2
> # cws h light/bedroom/main/bulb3
> # cws h light/bedroom/main/bulb4
> # ws h light/entrance/main/bulb
> # ws h light/kitchen/counter/bulb1
> # ws h light/kitchen/counter/bulb2
> # ws h light/kitchen/counter/bulb3
> # ws h light/kitchen/table/bulb1
> # ws h light/kitchen/table/bulb2
> # ws h light/kitchen/table/bulb3
> # ws h light/kitchen/table/bulb4
> # ws h light/livingroom/main/bulb1
> # ws h light/livingroom/main/bulb2
> # ws h light/livingroom/main/bulb3
> # ws h light/livingroom/main/bulb4
> # cws h light/hallway/main/bulb
> # ws g light/pantry/main/bulb1
> # ws g light/pantry/main/bulb2
> # ws g light/bathroom/main/panel1
> # ws g light/bathroom/main/panel2
> # ws g light/office/main/panel1
> # ws g light/office/main/panel2
> # ws g light/office/main/panel3
> # ws g light/office/main/panel4
> # ws g light/office/desk/bulb
> 
> ## loop and setup
> grep -E '^# (cws|ws|c) ' $0 | \
> while read bulb; do
>         btype=$(awk '{ print $2 }' <<< $bulb)
>         bbrand=$(awk '{ print $3 }' <<< $bulb)
>         bname=$(awk '{ print $4 }' <<< $bulb)
> 
>         echo -ne "[....] Setting up ${bname} as type ${btype} ..."
>         # NOTE: power on and color change when off behavior
>         mosquitto_pub -t zigbee2mqtt/${bname}/set/power_on_behavior -m 'off'
> 
>         # NOTE: state and level reporting
>         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                 -m '{"id":"'${bname}'","cluster":"genOnOff","attribute":"onOff","minimum_report_interval":0,"maximum_report_interval":3600,"reportable_change":0}'
>         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                 -m '{"id":"'${bname}'","cluster":"genLevelCtrl","attribute":"currentLevel","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
> 
>         sleep 1
> 
>         # NOTE: color reporting
>         case $btype in
>                 cws|ws|c)
>                         if [ "$bbrand" != "h" ]; then
>                                 mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                         -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"colorMode","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                         fi
>                 ;;
>         esac
>         case $btype in
>                 cws|ws)
>                         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                 -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"colorTemperature","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                 -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentX","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                 -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentY","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                 ;;
>         esac
>         case $btype in
>                 cws|c)
>                         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                 -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentHue","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                         mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
>                                 -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentSaturation","minimum_report_interval":3,"maximum_report_interval":3600,"reportable_change":1}'
>                 ;;
>         esac
>         sleep 1
> 
>         # NOTE: color options
>         case $btype in
>                 cws|ws|c)
>                         mosquitto_pub -t zigbee2mqtt/${bname}/set/color_options -m '{"execute_if_off": true}'
>                 ;;
>         esac
>         # XXX: seems to not be support by hue, new tradfri stuff
>         #mosquitto_pub -t zigbee2mqtt/${bname}/set/level_options -m '{"execute_if_off": true}'
>         sleep 5
>         echo -e "\r[DONE"
> done
> ```

wise-rice, 2025-02-12:
#13279

RubenKelevra, 2025-02-24:
Apart from the mostly discussed issues, here's a recording of other weird behavior of the lights. This time only my HG08131A (GU10 variant) are affected:

They are controlled with an automation instead of adaptive lighting. But the idea is pretty much the same: Measure the outside brightness and color temperature and do the same indoors and adapt the light of it doesn't match.

I first tried 4 seconds transition followed by 3 seconds of waiting for new data to arrive. But the Lidl lights really don't like having the color temperature and the brightness adjusted at the same time, with a lot of weird transitions in the wrong direction, or down from minimum/maximum brightness instead of the last value etc.

I now moved on to 0 seconds of transition and sending a brightness change followed by a waiting and then a color temperature change (also with transition=0).

However the lights still misbehave. Sometimes turning on at 100% brightness until the next command comes in, turning randomly to cold white instead of the correct color temperature or too warm instead of the right one.

Sometimes even flickering for a fraction of a second, which could very well be dangerous for people prone to photosensitive epilepsy.

Here's a graph showing what the outside brightness and color temperature is, and then blips of the light indoors misbehaving:

[image]

As not all lights are affected, but all lights are controlled in a HA group by my automation I can rule out any issues with my automation.

RubenKelevra, 2025-02-24:
Alright, I've now spend 3 hours setting up ZHA. And no @wise-rice ### **none** of the bugs are resolved by doing this - it was a complete waste of time. So it's clear that the firmware of these devices is broken, as switching between ZHA and Z2M improves nothing.

Apart from this more than half of my light's support in ZHA is half broken and they only show up with an RGB-LED mode, so their brightness is extremly limited and the colors are awful - sometimes even the exactly same model of bulb is sometimes showing up with CCT mode and sometimes without.

I found also a lot of other bugs - in the end I counted 12 individual bugs, while all of these devices work fine with Z2M.

wise-rice, 2025-02-25:
> Alright, I've now spend 3 hours setting up ZHA. And no [@wise-rice](https://github.com/wise-rice) **none** of the bugs are resolved by doing this - it was a complete waste of time. So it's clear [...]

Okay, sad to hear that - and thanks for checking.
Are you referring to LIDL lights? Or other brands? All LIDL lights showed up correctly for me. 

Did you notice that the reporting is different (maybe more reliable?) in ZHA? 
I feel like I need less trickery in my automations with ZHA (e.g. delays to wait for a light to update its status, etc).

Most importantly: Did you try out the "Enable enhanced light color/temperature transition from an off-state" setting?

wise-rice, 2025-02-25:
I also saw problems using any kind of transition. Always using "0" now.
But I am not sending separate brightness and color commands. That seems to work. 
Can you please see if it actually makes a difference to split up the commands ?

RubenKelevra, 2025-02-25:
> I also saw problems using any kind of transition. Always using "0" now.
> But I am not sending separate brightness and color commands. That seems to work.
> Can you please see if it actually makes a difference to split up the commands ?

I think you misread my earlier reply, I'm already using split commands and 0 transitions, the graph above shows how that looks like:

> I now moved on to 0 seconds of transition and sending a brightness change followed by a waiting and then a color temperature change (also with transition=0).
>
> However the lights still misbehave. [...]

RubenKelevra, 2025-02-25:
So from my testing the best solution atm is to:

- turn the lights always on with a brightness setting at or above 4%, to avoid flickering while turning the light on.
- Never submit any transitions, meaning the default transition will be used by the light. Processing  `transition: 0` or any other transition length seems to be completely broken.
-Never lower the lights below `3` in brightness (aka 1%).

I'll test this a couple of days, but it seems fine for me right now with a couple of hours of testing.

bruvv, 2025-02-25:
Thanks for all the testing. How would we use these with adaptive lightning ? Isn’t it something @Koenkk can implement for these lights ?

RubenKelevra, 2025-02-26:
So setting a default transition in z2m for each device of `0` and sending no transition or `transition: 0` seems to fix the flickering on individual commands for brightness and color temperature.

However if there's a default transition other than `0` set the GU10 bulbs I got seem to ignore the `transition: 0` in the `turn_on` command and use their default of around one second of transition, that's why they are flickering still.

I've not yet tried to use combined `light_on` commands for brightness and color temperature changes, and my testing is very limited.

Maybe someone else can test this with their bulbs and see if that fixes their issues as well?

wise-rice, 2025-02-26:
> > Are you referring to LIDL lights? Or other brands? All LIDL lights showed up correctly for me.
> 
> Well, basically all Aldi (Müller Licht) RGB+CCT lights I had died, so I can't really speak for them. The remaining ones have all just CCT. However the [LEDVANCE 4058075208339](https://www.zigbee2mqtt.io/devices/4058075208339.html#ledvance-4058075208339) also showed up with just RGB mode.
> 
> I have two different models of GU10 by Lidl, one model of E14, and one time outdoor spots, one strip and one desktop lamp. All models were affected. It's not for all devices of each model (in case I got more than one), but just a few devices. I also tried to hit the relearn button ZHA offers in the device page, as I first suspected the devices were not fully identified or something like this. But no change.
> 

Okay, we have the same lights, I think.
- I also have two Müller Licht (they need separate brightness and color commands to work with adaptive lighting, I think)
- LIDL lights - they also work if I do *not* use split commands. (I basically have LIDL models they offer: Outdoor spot RGB+CCT, 2m Strip RGB+CCT, GU10 RGB+CCT, GU10 CCT, E14 RGB+CCT, E14 CCT)
- All of them show the correct RGB capabilities in ZHA

> ... was false. The devices just don't support very small changes in color temperature, so sometimes you need to "jump" further until the device changes it's color temperature. This has to do with the Kelvin to mired conversion, as the devices use mired. So if you change one Kelvin, that may not tranlate to a full one mired change, and thus the device "jumps" back to the old value.

aah, I never experienced this issue because I use "tanh" with quite a steep curve, so the temps change in bigger steps - and most importantly: At night time the tanh curve is at a comfortable level around 1 or 3% (the "standard" curve is too bright at night).

[image]

wise-rice, 2025-02-26:
> > Most importantly: Did you try out the "Enable enhanced light color/temperature transition from an off-state" setting?
> 
> I don't think my lights support this?
> 
Since we have the same ones: they do ;-)

It matters when you:
- have the lights turned off
- turn them on with color and brightness (not separate)
--> immediate turn on with correct levels. No Transition.

> I've tried the "startup color temperature" setting, but it had no effect. I think this only take effect if the lights are power cycled to turn them on.
> 
> Edit: Ah I've looked on Reddit. It's just sending one message with turn on at low brightness and the correct color temperature followed by the correct brightness so it dims up from the correct color temperature.
> 
Are you referring to this [comment](https://github.com/Koenkk/zigbee2mqtt/discussions/13279#discussioncomment-4726409) ?
It seems like a theory.
In reality: I see no transition. Immediate turn on at Brightness A with Level B, no matter what the turn off status was.

> Well, that's actually what I'm doing with my automation anyway. I send a brightness 4% to the lights with the correct color temperature and then the correct brightness.
> 
> This avoids the flickering then Lidl lights are turned on, with 4% I see zero flickering.

wise-rice, 2025-02-26:
My LIDL and Müller lights don't flicker. Are we 100% sure, that the firmware is broken? Maybe it is an AdaptiveLighting or zigbee Network issue?

Maybe you have a faulty Zigbee router somewhere in your network?
I had one (eWeLink) that was very slow and sometimes sent commands twice.

A different Issue I always have:
Some (a few) lights will randomly turn back on, after I turned them off (through an Automation or Blueprint).
Weirdly, other lamps (identical type and firmware) stay off. It is only one or two lights in a Zigbee group, which turn back on... Really annoying 

I think it might be related to AdaptiveLighting...

RubenKelevra, 2025-02-26:
> > I've tried the "startup color temperature" setting, but it had no effect. I think this only take effect if the lights are power cycled to turn them on.
> > Edit: Ah I've looked on Reddit. It's just sending one message with turn on at low brightness and the correct color temperature followed by the correct brightness so it dims up from the correct color temperature.
> 
> Are you referring to this [comment](https://github.com/Koenkk/zigbee2mqtt/discussions/13279#discussioncomment-4726409) ? It seems like a theory. In reality: I see an immediate turn on at Brightness A with Level B. No visible transition. No matter what the turn off status was.

Nah, here:

https://www.reddit.com/r/homeassistant/comments/17efki5/comment/k90clgf/

You would not see this, as it's basically two commands send to the bulb in short succession, like 100 or 200 ms behind each other. If I have time later I'll look at the source code. Kinda make sense to add this to z2m as well.

RubenKelevra, 2025-02-26:
> My LIDL and Müller lights don't flicker. Are we 100% sure, that the firmware is broken?

Yes.

> Maybe it is an AdaptiveLighting or zigbee Network issue?

Well, I use adaptive lighting for a couple of rooms, which all have this issue, but one room is just running off of my automations, and this has the same issue.

Some lights do have this issue, others have not - which are all in the same Home Assistant light group, so it's clearly a firmware issue if some lights flicker and others are not.
 
> Maybe you have a faulty Zigbee router somewhere in your network? I had one (eWeLink) that was very slow and sometimes sent commands twice.

Sure, that's a solid theory, but I can nearly completely rule that out, as it's absolutely consistent: On low brightness settings with transition commands the bulbs will flicker with a chance of 1:10 to 1:20, in various ways.

The commands I send are like: okay, the bulb is at 50% brightness, lets turn it to 51% brightness with 4 seconds of transition, and the bulb goes like, okay, "100% brightness for 0.2 seconds, then 51% brightness" or "okay, 6500 K, 100% brightness" or "okay, 2400 K, 10% brightness" or "okay, jump instantly to 100% brightness and then dim down to 51% brightness".

So even in the case the same command reaches the bulb twice, the behavior is still wrong.

> A different Issue I always have: Some (a few) lights will randomly turn back on, after I turned them off (through an Automation or Blueprint). Weirdly, other lamps (identical type and firmware) stay off. It is only one or two lights in a Zigbee group, which turn back on... Really annoying

> 
> I think it might be related to AdaptiveLighting...

This can happen in case you use Adaptive Lighting and it plans to send an update, while you already send the "off" command. In this case both commands reach the bulb, as Adaptive Lighting didn't update fast enough.

That's why I plan to do some major rework on AL, to have AL expose virtual lights, so you turn off virtual lights, and those commands will be send to AL and AL processes all changes internally, so those conflicts can happen anymore.

It also allows for much more interesting applications of AL.

--- 

Another funny bug is, that one single light in all my dozens of lights (manufacturer is Lidl again, but an older generation of GU10) decides randomly to not turn off, if I tell it to, but go to different shades of blue. Which is quite weird if you stand up at night and your hallway is blue for no reason :D

[24 more replies]