Koenkk/zigbee2mqtt #21291: MiBoxer RGB controller effect control not working

DevRedOWL, 2024-02-07:
### What happened?

MiBoxer FUT039Z (RGBW+CCT) device is being detected as TS0505B (seems wrong, because i've found https://www.zigbee2mqtt.io/devices/FUT039Z.html).
Color and brightness change workd properly, but effects not working.

### What did you expect to happen?

Effect control and proper detection

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20210218

### Adapter

sonoff zigbee 3.0 dongle

### Setup

Add on Home Assistant supervised

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14201286/log.txt)

adynis, 2024-02-12:
I confirm !!!
 I have multiple FUT035Z+ and I just thought that the effects are not supported ... But the mentioned link shows it shoudl work.
Additionally for my also "transition" does NOT work. @DevRedOWL -> Are you able to change "transition" and see the expected effect ?

DevRedOWL, 2024-02-12:
@adynis same, transittion makes no effect.

DevRedOWL, 2024-09-03:
Actually, bump

closed 2025-01-08 by github-actions[bot]