Koenkk/zigbee2mqtt #21295: EventBus error 'Receive/deviceMessage': `wait` must not be negative.

darkner258, 2024-02-08:
### What happened?

I re-plugged my coordinator and restarted Z2M. After doing that, one of my devices (IKEA Symfonisk controller, the rotary one), gives me this error in log "EventBus error 'Receive/deviceMessage': `wait` must not be negative." after which the whole network restarts. I did not change anything in the settings and all the other devices work as they did before (even better, since the coordinator is further away)

### What did you expect to happen?

I expected the controller to send MQTT messages

### How to reproduce it (minimal and precise)

Attempt to send any MQTT message from the controller

### Zigbee2MQTT version

1.35.1 commit: 3c962042

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle EZSP v8

### Setup

Z2M plain on RPi

RamonSmit, 2024-02-21:
I am experiencing the same issue. Looking into this, because it makes my home quite unusable.

closed 2024-09-21 by github-actions[bot]

matsahm, 2025-01-18:
I also get the following error message every 50 minutes. How can I fix this or narrow down the problem further?

Error message: z2m: EventBus error 'Receive/deviceMessage': `wait` must not be negative.

RamonSmit, 2025-01-18:
Did you check if you have a `debounce` in your `configuration.yaml`? If so, try to remove it and see if that solves the error message.

matsahm, 2025-01-18:
I already checked that. It's not included in my config.

Westcott1, 2025-02-11:
Yes, removing an inadvertent negative debounce fixed it for me!

matsahm, 2025-03-18:
I detected a negative debounce value in the file 0x00158d000945f5d5. After adjusting it via the web interface, the error disappeared.