Koenkk/zigbee2mqtt #21298: HASS Z2M Add-on frontend/user interface "broken" after update to 1.35.3

denniskemmerling, 2024-02-08:
### What happened?

After the update the frontend/user interface looks broken and not correct. 
[image]

### What did you expect to happen?

The standard user interface of Z2M

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20230507

### Adapter

Smlight SLZB-06

### Setup

Add-on on HASS OS on a Raspberry Pi

joschaschultze, 2024-02-08:
Yes, I had exactly the same problem and went back to 1.35.2.

DuncanRae, 2024-02-08:
This is a big that needs to be fixed, but the workaround is to click on the light/dark button. This seems to fix it.

marsonde, 2024-02-08:
Oh, great, that was simple! 😎 Thanks!

joschaschultze, 2024-02-08:
Great solution :-)

jsa91, 2024-02-08:
> This is a big that needs to be fixed, but the workaround is to click on the light/dark button. This seems to fix it.

Same issue here but thanks for the quick fix!

kayserjan, 2024-02-08:
> This is a bug that needs to be fixed, but the workaround is to click on the light/dark button. This seems to fix it.

where?
[image]

SVH-Powel, 2024-02-08:
[image]

Koenkk, 2024-02-08:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/21268

closed 2024-02-08 by Koenkk