Koenkk/zigbee2mqtt #21301: Button scene switch; model: TS0041 (_TZ3000_mrpevh8p)

hagier, 2024-02-08:
### What happened?

All "single", "double" and "hold" actions are suddenly shown as "1_single", "1_double" and "1_hold" breaking all automations.

### What did you expect to happen?

Actions should be: single/double/hold.

### How to reproduce it (minimal and precise)

Clicking button reports wrong action.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle-P

### Setup

HA Add-on

fufurax, 2024-02-08:
I confirme, same issue with IH-K663

Koenkk, 2024-02-08:
Should be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already

baptisteba, 2024-02-09:
I confirm that it happened since 1 or 2 stable update ago.
My device models concerned : 
SH-SC07
IH-K663

closed 2024-08-12 by hagier