Koenkk/zigbee2mqtt #21303: IKEA and XIAOMI devices not working after 1.35.3-1 uodate (and broken frontend)

kayserjan, 2024-02-08:
### What happened?

After updating to 1.35.3-1 all of my XIAOMI and IKEA devices stopped responding (Conbee II)
Also the frontend is corrupted on iOS and Android devices. Only on Windows all is fine...
Restore to 1.35.2-1 fixed the issue.
I attempted 2nd install and it "prodecued" the same result.

### What did you expect to happen?

Get described fixes.

### How to reproduce it (minimal and precise)

Update Zigbee2MQTT addon 1.35.2-1 to 1.35.3-1

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

HA OS, RPi3, Supervisor

### Debug log

[Info 2024-02-08 092402MQTT publish .txt](https://github.com/Koenkk/zigbee2mqtt/files/14206285/Info.2024-02-08.092402MQTT.publish.txt)

alevike, 2024-02-08:
Exactly the same problem here. 

Setup
HA OS, Intel NUC, Supervisor

Adapter
ConBee2

Creo2005, 2024-02-08:
[image]
:(

Chartreusito, 2024-02-08:
Click on the sun/moon button to enable/disable dark mode, it should at least bring the theme back.

t-liu93, 2024-02-08:
> Click on the sun/moon button to enable/disable dark mode, it should at least bring the theme back.

Thank you @Chartreusito this works.

Koenkk, 2024-02-08:
- Conbee https://github.com/Koenkk/zigbee2mqtt/issues/21162
- Frontend https://github.com/Koenkk/zigbee2mqtt/issues/21268

closed 2024-02-08 by Koenkk

Rempert, 2024-02-08:
Strange operation, I mainly read problems with the conbee II adapter. I use the UZG-01 from Zigstar (IP-based) and have not experienced any problems yet (>120 zigbee devices)