Koenkk/zigbee2mqtt #21304: Again Z2M is not buldable - 1.35.3

emandtf, 2024-02-08:
### What happened?

Every time I would like to update to newest Z2M version there is always something that goes wrong.....

Few months ago the procedure was easy:

1. npm run clean
2. npm run build
3. npm ci

Now the same procedure (on 1.35.3 tag) doesn't works....

Running "_npm run clean_":
> npm ERR! Missing script: "clean"

Running "_npm run build_":
> lib/controller.ts:96:13 - error TS2339: Property 'setLogger' does not exist on type 'typeof import("/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index")'.
> 96         zhc.setLogger(logger);
>              ~~~~~~~~~
> lib/extension/homeassistant.ts:1063:25 - error TS2339: Property 'category' does not exist on type 'Expose'.
  Property 'category' does not exist on type 'Numeric'.
> 1063         if (firstExpose.category === 'config') {
>                            ~~~~~~~~
> lib/extension/homeassistant.ts:1065:32 - error TS2339: Property 'category' does not exist on type 'Expose'.
  Property 'category' does not exist on type 'Numeric'.
> 1065         } else if (firstExpose.category === 'diagnostic') {

it seems most of tags are often left in a development/broken state and it tooks hours to find a buildable commit.....

It's very frustrating....

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

NA

### Adapter

NA

### Setup

running on venv (arm7v arch)

emandtf, 2024-02-08:
Same for 1.35.2.
The last buildable is 1.35.1

mrmaximas, 2024-02-08:
what about?
git checkout HEAD -- package-lock.json
git checkout HEAD -- package.json

closed 2024-08-17 by emandtf