Koenkk/zigbee2mqtt #21306: Cannot pair SE Powertags (ZGP)

estimadarocha, 2024-02-08:
### What happened?

Installed last versions of z2m and fw on adapter and cannot pair powertags. The only messages i can see are:

In pairing mode:
Debug 2024-02-08 11:45:59Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"commandID":227,"frameCounter":209,"srcID":0}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2024-02-08 11:46:00Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"commandID":227,"frameCounter":219,"srcID":0}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2024-02-08 11:46:01Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"commandID":227,"frameCounter":226,"srcID":0}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2024-02-08 11:46:02Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"commandID":227,"frameCounter":236,"srcID":0}' from endpoint 242 with groupID null, ignoring since it is from coordinator

After get paired with SE gateway.
Debug 2024-02-08 11:46:21Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,16,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2024-02-08 11:47:15Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,17,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator

Can i pair this devices directly to the coordinator? or do i need a router in the midle that allow cluster with endpoint 242 and input/output green power?

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

Zigbee (EZSP) 7.4.0.0

### Adapter

Sonoff ZBDongle-E

### Setup

Docker

chris-1243, 2024-02-08:
I would try to use a router with endpoint 242 input/ouput greenPower.
Nowadays, not only Hue are able to translate ZigbeeGreen messages as per instructions from some manufacturers (Sunricher) and by testing from my side. My tries were made with a ptm215z from Enocean (zigbeegreen) and some routers from different brands like Ikea, Sunricher and Nodon

estimadarocha, 2024-02-08:
Any more sugestions about equipaments beside those?

chris-1243, 2024-02-08:
Well, some Hue lights as described in the documention (I do not have a list)
From Philips: [4034031p7](https://www.zigbee2mqtt.io/devices/4034031P7.html#philips-4034031p7), [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738))

From Ubisys, the [power switch S2](https://www.zigbee2mqtt.io/devices/S2.html#ubisys-s2)

From Ikea, those two lights and this driver: [LED2103G5](https://www.zigbee2mqtt.io/devices/LED2103G5.html#ikea-led2103g5), [LED2101G4](https://www.zigbee2mqtt.io/devices/LED2101G4.html#ikea-led2101g4), [ICPSHC24-10EU-IL-1, Version 2](https://www.zigbee2mqtt.io/devices/ICPSHC24-10EU-IL-1.html#ikea-icpshc24-10eu-il-1).

I use those devices and did some tries with them. I have been able to pair a ZigbeeGreen PTM215Z from Enocean without issues.
It is quite difficult to have good support for zigbeegreen devices. At least in Z2M it is fully implemented with some minors difficulties. As far as I know, ZHA is not able to manage such devices...

estimadarocha, 2024-02-10:
@chris-1243 did bought a Ikea led buld that have both input/output on endpoint 242 but still no luck.

But i think i am one point closer because with bulb and unpairing and pairing the powertag with EcoStruxure Panel Server (the hw i want to avoid) sometimes zigbee2mqtt can discover it, but even when discovered the only value i can read is LQI.

Note that my powertags are the A9MEM.... version.

This is probably related with https://github.com/Koenkk/zigbee2mqtt/issues/7975

In that issue it's related by @jlama that created this https://github.com/jlama/powertagd that the:

"These devices communicate with ZigBee Green Power and require precise control of the timings to work reliably.

A Sonoff ZigBee Dongle-E device (not the -P model!) with custom firmware is used as a gateway. The firmware is based on the NCP-UART firmware provided by Silicon Labs, modified to automatically send a special ACK reply which is specific to PowerTags"

is there anyway to replicate this ACK packages directly on zigbee2mqtt @Koenkk?

thanks

chris-1243, 2024-02-10:
I am sorry, I would not be able to help much more. It is out of my knowledge. 

In #7975 , it seems some users have been able to run such ZG device via specific routers. It may be a solution

Kiki3872, 2024-02-10:
I've been trying to do the same thing for a long time, I hesitate to flash the sonoff E dongle because then I don't know what to do

estimadarocha, 2024-02-10:
Like i said i am using a EcoStruxure Panel Server (SE) to pair/unpair and force the discovery... next move is to test the Wiser Hub 2... 

the router from aeotek doesn't have the endpoint 242 input/output (it was my first approach) i think the way the powertag comunicates is completely different from r9 to a9. R9 is for residencial use and A9 is for power distribution cabinets not residencial.

eriktack, 2024-03-14:
Here is how I paired RESI9 powertags with a HA Green. Pay extra attention to the pairing part, thats where I failed at first.

Preparation:

1. I got a Sonoff Zigbee 3.0 USB Dongle Plus. This MUST be the slightly larger P version, NOT the E version. The E version wont even pair the tags.
2. I installed zigbee2mqtt and the mosquitto broker with the default settings.
3. I added a IKEA range extender and placed it close to where I have the tags, this is the version wiht a usb port you can detach from its power plug. You can probably use any device that support zigbee green power and has _Endpoint 242 > Output clusters > greenPower_ under the Clusters tab of the device. The IKEA one worked for me and is fairly cheap and available.

Pairing:

1. Its very important to not allow anything to pair by default in zigbee2mqtt, once you connect the tags to power they enter pairing mode immediately and if they pair through the wrong device, they wont pair correctly.
2. In the devices view of zigbee2mqtt, use the permit join dropdown and select the IKEA range extender and then click it to temporarily allow devices to join through that only.
3. After a few moments your tag should join. If it doesnt join immediately, make sure its blinking orange and also try repeating step 2.
4. Wait a couple of minutes and make sure the tag reports more then just signal strength under the “exposes” tab.
If the tag only report signal strength, chances are it joined through the coordinator or another router.

Issues:

The 3 phase tags dont report any energy or power on the separate a, b, and c phases, only the total. The individual phases are all null. This was a bit of a bummer for me as I had hoped to split up the stove(2 phases) and oven (1 phase) in separate meters. I suppose I could use 3 single meters instead, but only the total is fine for now.
Interestingly enough, the single phase meters do report power and energy on the a port in addition to the total.

Its also a bit of a pain to remove them form the network if you make a mistake in the pairing, they dont completely unpair, but if you turn them off for a while, and also retart HA (and pray to moloch), they eventually reset and enter pairing mode again.

eriktack, 2024-03-14:
> But i think i am one point closer because with bulb and unpairing and pairing the powertag with EcoStruxure Panel Server (the hw i want to avoid) sometimes zigbee2mqtt can discover it, but even when discovered the only value i can read is LQI.
try allowing pairing only through the bulb in the pairing dropdown in the zigbee2mqtt devices view. Disable the always allow join setting in zigbee2mqtt before powering the tag.

mti-sk, 2024-06-14:
Hi,
Look at #23031 I've same problem.

estimadarocha, 2024-07-03:
I did upgrade my coordinator (ZBDongle-E) to the last firmware 7.4.3 and moved to ember.

with this change i got now this:

zh:controller:greenpower: Error: Parameter 'tempMaster' is missing

@mti-sk @chris-1243 @Koenkk @jllama

Koenkk, 2024-07-03:
@estimadarocha could you provide some more debug logging of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

estimadarocha, 2024-07-05:
@Koenkk 
[log.log](https://github.com/user-attachments/files/16109716/log.log)

Koenkk, 2024-07-08:
@estimadarocha can you try to:

```
docker exec -it zigbee2mqtt sh # (where `zigbee2mqtt` is the container name)
apk add nano 
```

replace two times `tempMaster: frame.payload.gppNwkAddr,` with `tempMaster: frame.payload.hasOwnProperty('gppNwkAddr') ? frame.payload.gppNwkAddr : 0,` , save and exit the container

Restart the container and check if it works now

estimadarocha, 2024-07-08:
> @estimadarocha can you try to:
> 
> ```
> docker exec -it zigbee2mqtt sh # (where `zigbee2mqtt` is the container name)
> apk add nano 
> ```
> 
> replace two times `tempMaster: frame.payload.gppNwkAddr,` with `tempMaster: frame.payload.hasOwnProperty('gppNwkAddr') ? frame.payload.gppNwkAddr : 0,` , save and exit the container
> 
> Restart the container and check if it works now (using `docker restart`)

please can you please tell me where do i replace those entries?

estimadarocha, 2024-07-08:
@Koenkk new logs...

[log 1.log](https://github.com/user-attachments/files/16135386/log.1.log)

now i got those:

[2024-07-09 00:34:21] error: 	zh:controller:greenpower: Error: ~x~> [ZCL BROADCAST] Failed to send with status=BUSY.
[2024-07-09 00:39:31] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "35036".

and i still need to pair and unpair from the own schneider gateway to get it detected, and even when it get's detected it doesn't have any exposed values

Koenkk, 2024-07-10:
So I'm not entirely sure what we should send here. But to confirm, when pairing it to a router (so not directly to the coordinator) it pairs fine?

eriktack, 2024-07-10:
It seems in my case it must be paired with a device that support green power on endpoint 242. In my case an ikea range extender (model E1746 firmware 2.3.086)

Also, I could never get them to pair at all using a sonoff dongle E as coordinator. But with everything else the same and using a sonoff dongle P, they paired with the ikea range extender if I permitted joining with that device only.

Sorry for the less-then-specific answers, but they only seem to like certain coordinator hardware, and also only certain green power enabled devices.

/Erik

estimadarocha, 2024-07-10:
> So I'm not entirely sure what we should send here. But to confirm, when pairing it to a router (so not directly to the coordinator) it pairs fine?

@Koenkk in my case I can’t neither on coordinator or any of the ikea bulbs with 242 endpoint. What to do next?

mti-sk, 2024-07-11:
> Here is how I paired RESI9 powertags with a HA Green. Pay extra attention to the pairing part, thats where I failed at first.
>
> Preparation:
>
>     1. I got a Sonoff Zigbee 3.0 USB Dongle Plus. This [...]

Hello,
Advice, using sonoff dongle version P instead of version E for ZGP pairing is unfortunately pointless, Sonoff dongle version P can no longer be bought/obtainable. We have to use what is available. You also cannot implement a new installation with ZGP, you will not get another Sonoff dongle P. The ZB2MQTT project should be sustainable, conclusions usable with any components and at any time.

mti-sk, 2024-07-11:
> So I'm not entirely sure what we should send here. But to confirm, when pairing it to a router (so not directly to the coordinator) it pairs fine?

Hi,
At this time, there is no possibility to pair ZGP devices either through the coordinator or through any ZB GreenPower Proxy (GPP).
If ZPG pairing worked for anyone, it was using previous versions of ZB2MQQT and also Sonoff Dongle ver. P. But today (the current version of ZB2MQTT, unavailability of Dongle ver. P ) it cannot be repeated (this is my assumption).
I also confirm log messages like:
1. Debug: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,16,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator -- **why ignored ????**
2.error: zh:controller:greenpower: Error: Parameter 'tempMaster' is missing -- **probably an error fw Sonoff ver.E ???**

chris-1243, 2024-07-11:
@mti-sk 

TI CC2652P2 are still available on the market (RF Star RF-RG-52PAS for example...). This chip is used in the Dongle-P. You may find easily TI CC2652P7 which is an updated chip and using the same driver `zstack` in Z2M.

closed 2025-01-08 by github-actions[bot]