Koenkk/zigbee2mqtt #21310: [New device support]: TS0601

aleksandarjon, 2024-02-08:
### Link

https://www.aliexpress.com/item/1005006366812532.html?spm=a2g0o.refund_spm.0.0.5f8872e4uL0d3e

### Database entry

const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZE200_wbhaespm',     description: 'Automatically generated definition',     extend: [],     meta: {}, };  module.exports = definition;

### Comments

Could you please add this device to the list?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_wbhaespm',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

aleksandarjon, 2024-02-08:
> what about database.db entry for this device?

Not sure how to provide you that? :)

aleksandarjon, 2024-02-10:
> what about database.db entry for this device?

Please explain how to deliver you the mentioned info

mrmaximas, 2024-02-10:
> Please explain how to deliver you the mentioned info

in the zigbee2mqtt/ folder please find /data/database.db

aleksandarjon, 2024-02-12:
> > Please explain how to deliver you the mentioned info
> 
> depend of your os in the zigbee2mqtt folder please find /data/database.db

Hope this will help:

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c0c030","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}

mrmaximas, 2024-02-12:
@aleksandarjon it's coordinator entry instead of _TZE200_wbhaespm

aleksandarjon, 2024-02-12:
> @aleksandarjon it's coordinator entry instead of _TZE200_wbhaespm

Sorry... hope its right...

{"id":54,"type":"Router","ieeeAddr":"0xa4c1389d8f23d05c","nwkAddr":35681,"manufId":4417,"manufName":"_TZE200_wbhaespm","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�W-i�W-i�W-i�W-i","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_wbhaespm","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1707396204397}

aleksandarjon, 2024-03-07:
@mrmaximas could you please confirm I shared the correct data? :)

closed 2024-10-04 by github-actions[bot]