Koenkk/zigbee2mqtt #21311: TS011F _TZ3000_ky0fq4ho Din smart relay (with power monitoring)

Descagir, 2024-02-08:
### What happened?

I have an issue with the "power" expose. A value of 0 is shown whereas it is working, as you can see for the current value.
[image]
I just migrate from ZHA to zigbee2mqtt and everything was working fine with ZHA. So i am pretty sure that it's not coming from my smart relay.

### What did you expect to happen?

I expected a power value.

### How to reproduce it (minimal and precise)

I just have to switch the relay (on/off).

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

    20210708

### Adapter

Sonoff Zigbee 3.0 CC2652P

### Setup

Home Assistant on Raspberry PI 4B+

Descagir, 2024-02-13:
Could it be possible that the value (for active power) I try to monitor is too weak (under 4/8 Watt) and I have a bad config in the parameter field ? 
Here is what I have for the active power.
Any help is welcome.
[image]

closed 2024-09-11 by github-actions[bot]

mikeziri, 2026-02-15:
In my case, using ZHA, some of my TS011F plugs also work but the Power sensor with <= 8w devices, show 0w.
The strange thing is the descrepancy. Some seem to have more precision 🤷‍♂️