Koenkk/zigbee2mqtt #21316: Conbee III - Not Working - Latest Firmware and Dev

thk-socal, 2024-02-08:
### What happened?

I am working on setting up the Conbee III. I have tried all three versions of firmware released. I have a test network with the latest-dev tagged image from Docker. I have a single Philips Hue light to test with. When I use the SkyConnect or Sonoff adapters with EZSP, there are no issues at all. If I try the Conbee III, the device joins and after joining it does not get anymore messages from the adapter. 

### What did you expect to happen?

For the Conbee III to work like the other adapters

### How to reproduce it (minimal and precise)

New Z2M install.
Connect Conbee III and configure the network.
Join Philips Hue light.
Try to have the light do anything.

### Zigbee2MQTT version

1.35.3-dev commit: 55653c9

### Adapter firmware version

0x26450900 and 0x264e0900 and 0x264f0900

### Adapter

Conbee III

### Setup

Docker

### Debug log

debug 2024-02-08 12:29:02: Publishing 'set' 'color' to '0x001788010d18eaf6'
debug 2024-02-08 12:29:05: Received MQTT message on 'zigbee2mqtt/0x001788010d18eaf6/set' with data '{"state":"OFF"}'
debug 2024-02-08 12:29:05: Publishing 'set' 'state' to '0x001788010d18eaf6'
error 2024-02-08 12:29:09: Publish 'set' 'color' to '0x001788010d18eaf6' failed: 'Error: Command 0x001788010d18eaf6/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":8880,"colory":2615}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (16))'
debug 2024-02-08 12:29:09: Error: Command 0x001788010d18eaf6/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":8880,"colory":2615}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (16))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:146:13)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1204:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:266:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
error 2024-02-08 12:29:12: Publish 'set' 'color' to '0x001788010d18eaf6' failed: 'Error: Command 0x001788010d18eaf6/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":8880,"colory":2615}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (17))'
debug 2024-02-08 12:29:12: Error: Command 0x001788010d18eaf6/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":8880,"colory":2615}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (17))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:146:13)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1204:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:266:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
error 2024-02-08 12:29:15: Publish 'set' 'state' to '0x001788010d18eaf6' failed: 'Error: Command 0x001788010d18eaf6/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (18))'
debug 2024-02-08 12:29:15: Error: Command 0x001788010d18eaf6/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (18))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1116:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:266:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
debug 2024-02-08 12:30:39: Saving state to file /app/data/state.json
debug 2024-02-08 12:34:48: Received Zigbee message from '0x001788010d18eaf6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16785412,"imageType":287,"manufacturerCode":4107}' from endpoint 11 with groupID null
debug 2024-02-08 12:34:48: Device '0x001788010d18eaf6' requested OTA
debug 2024-02-08 12:34:48: OTA: Checking if an update is available for '0x001788010d18eaf6' (1741930V7)
debug 2024-02-08 12:34:48: OTA: Is new image available for '0x001788010d18eaf6' (1741930V7), current '{"fieldControl":0,"manufacturerCode":4107,"imageType":287,"fileVersion":16785412}'
debug 2024-02-08 12:34:48: ZigbeeOTA: Getting image metadata for '1741930V7'
debug 2024-02-08 12:34:48: ZigbeeOTA: Downloaded main index
debug 2024-02-08 12:34:48: OTA: Is new image available for '0x001788010d18eaf6' (1741930V7), latest meta '{"fileVersion":16785412,"fileSize":423942,"url":"https://otau.meethue.com/storage/ZGB_100B_011F/38f5ed2b-119a-4d6c-9852-2e6470fecf0e/100B-011F-01002004-ConfLightBLE-ModuLumV3-EFR32MG21.zigbee","sha512":"68b70b7f152b246dc34086982c2961eb5bc2a4f4c30d3276619e8db9458549177b3ea391cccaa825821268c0f4be8f86d3a1dc66d55876aa3119755ea8aabde0"}'
debug 2024-02-08 12:34:48: OTA: Update available for '0x001788010d18eaf6' (1741930V7): NO
info  2024-02-08 12:34:48: MQTT publish: topic 'zigbee2mqtt/0x001788010d18eaf6', payload '{"color_mode":"xy","linkquality":187,"power_on_behavior":null,"state":"ON","update":{"installed_version":16785412,"latest_version":16785412,"state":"idle"},"update_available":null}'
debug 2024-02-08 12:34:48: Responded to OTA request of '0x001788010d18eaf6' with 'NO_IMAGE_AVAILABLE'

thk-socal, 2024-02-08:
I disabled Home Assistant settings and restarted Z2M and it started working perfectly after that. Not sure why flipping that setting fixed it. Appears to be something to do with state.

thk-socal, 2024-02-08:
It then works for a few minutes and breaks again. When it breaks the light actually flashes from on to off and back on. Almost like it lost connection in someway. I will keep this small instance up for any testing you want.

MartijnVdS, 2024-02-10:
Could this be the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/21162 ?

thk-socal, 2024-02-11:
> Could this be the same issue as #21162 ?

No, this issue has existed for many versions of Z2M and the new Conbee III. There are multiple issues with Conbee III that are being worked.

thk-socal, 2024-02-11:
Just installed the latest Conbee III firmware 0x26500900 and everything "appears" to be working like it should now. Will close for now as it appears fixed and will reopen if other issues arise.

closed 2024-02-11 by thk-socal