Koenkk/zigbee2mqtt #21317: After most recent update (within HASS) plenty of error messages are in log [Zigbee2MQTT:error 2024-02-08 22:14:30: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}]

tomtom13, 2024-02-08:
### What happened?

Hi,
After updating to newest version 1.35.3, I get plenty of those errors in the log (and also those show up as overlay popups)
Zigbee2MQTT:error 2024-02-08 22:14:30: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Nothing in the system changed between updates.
I've tried rebooting HASS,
rebooting proxmox node - sine HASS is in VM in proxmox (usb passthrough)

### What did you expect to happen?

Not to get an avalanche of errors:
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:50Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:50Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:51Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:51Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:52Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:56Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:57Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:20:00Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:20:01Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}

### How to reproduce it (minimal and precise)

well, I'm using Zonoff USB dongle, passthrough to HASS. Z2M is within HASS. Rather large network (157 devices)

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20210708

### Adapter

zStack3x0 - Sonoff usb zigbee dongle

### Setup

HASS addon, within VM on proxmox

### Debug log

Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:49Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:50Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:50Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:51Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:51Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:52Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:53Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:56Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:57Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:58Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:19:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:20:00Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-08 22:20:01Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}

Koenkk, 2024-02-10:
You are probably using an in compatible external converter

tomtom13, 2024-02-11:
Can you please elaborate ? 
USB dongle is on a supported list (actually advised list). It's directly connected to the VM.

Koenkk, 2024-02-11:
I misinterpreted the logging, can you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

tomtom13, 2024-02-11:
FYI, now issues seems to happen every 10-20 seconds (two entries in log)

Error 2024-02-11 22:54:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-11 22:54:59Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-11 22:55:16Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-11 22:55:17Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-11 22:55:28Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}
Error 2024-02-11 22:55:28Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '1')}

I shal enable the debug loggin in a sec and get back to you.

tomtom13, 2024-02-11:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14234399/log.txt)
here you go.

Koenkk, 2024-02-12:
Can you provide your data/database.db?

Note to self: problematic device `56694`: `{"attrId":1024,"dataType":42,"attrData":502}`

tomtom13, 2024-02-12:
from within the z2m container OR hass ?

Koenkk, 2024-02-13:
z2m container

tomtom13, 2024-02-14:
Sorry for being nagy - but maybe "what happened" ?
Also I've had a question in the past of how to get into dev / debug on z2m, and nobody gave any pointers (I'm a emb s eng, and I was considering making some small dev for the zigbee - would be nice to have some debug platform)

closed 2024-09-13 by github-actions[bot]