Koenkk/zigbee2mqtt #21319: OTA update problem with ZNXDD01LM (Ceiling light L1-350)

psike01, 2024-02-09:
### What happened?

After click button "Check for new updates" recive error message "Device didn't respond to OTA request".
In the "state" of this lamp in z2m:

> 
> "update": {
>   "installed_version": -1,
>   "latest_version": -1,
>   "state": null
> }
> "update_available": null
> 

### What did you expect to happen?

Firmware build date of my lamp is 20220331, after checking update i expect recive info about new version (today is 20230905 and present in https://github.com/Koenkk/zigbee-OTA/pull/354), and start update to new version.
Try to power off this lamp/ re-join to zigbee coordinator - doesn't solve problem

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20201010

### Adapter

USB stick MODKAM_V3 (zStack30x)

### Setup

Add-on on Home Assistant

psike01, 2024-03-06:
Same in Zigbee2MQTT version 
1.36.0

andrazek, 2024-09-29:
I’m also unable to update the ZNXDD01LM device. The "Firmware build date" shows 20220331, but the "Firmware version" field is blank. When I press the "Check for new updates" button, after a few seconds, I receive the following message: "Device didn't respond to OTA request."

Is there a specific procedure for updating Aqara products? I’m currently using the latest firmware version for the SONOFF Zigbee 3.0 USB Dongle Plus-P, as well as the latest version of Zigbee2MQTT.

closed 2025-03-10 by github-actions[bot]