Koenkk/zigbee2mqtt #21320: Front UI broken since 1.35.3-1

jmcollin78, 2024-02-09:
### What happened?

Since the installation of 1.35.3-1, the front UI is broken. Css is not found:
[image]

Of course I try to empty the browser cache, restart the add-on and alos restart HA without success.

Am I alone with this problem ?

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

-

### Adapter

ezsp

### Setup

add-on

Usandu, 2024-02-09:
You are not alone m8, same problem here.

jmcollin78, 2024-02-09:
As a workaround, clicking on the sun icon solve the problem:

[image]

Wesleyl89, 2024-02-09:
Same here but clicking on the sun icon indeed fixes it like @jmcollin78 said

neovarsis, 2024-02-09:
Same here, as above, sun icon does fix it.

araik1998, 2024-02-09:
То же самое, что и выше

NReiher, 2024-02-09:
Same issue. Sun resolved it. Happened on two different computers. Resolving it in one (clicking the sun) does not fix it on the other.

giscal, 2024-02-09:
same issue for me to - no front end

energywave, 2024-02-10:
Same here. Clicking the sun fix it.
To add something: I'm using Edge and just updated zigbee2mqtt to 1.35.3, then, after update, it was like in the screenshot of this issue (italian language, if that matters).
So I've installed a clean Chrome and there was no problem.
I checked the browser dev tools and found this error (in Edge, not in chrome):
```
Could not find specified theme
(anonimo) @ index-a782d339.js:1058
```

lgander, 2024-02-10:
Same here. Sun fixes it.

Koenkk, 2024-02-10:
Dupe of https://github.com/Koenkk/zigbee2mqtt/issues/21268

closed 2024-02-10 by Koenkk