Koenkk/zigbee2mqtt #21322: zigbee2mqtt 1.35.3 GZCGQ01LM and LYWSD03MMC (with devbis 1.1.0 firmware) do not appear on the dashboard.

lbarjak, 2024-02-09:
### What happened?

GZCGQ01LM and LYWSD03MMC (with devbis 1.1.0 firmware) are visible in the devices but do not appear on the dashboard.

### What did you expect to happen?

GZCGQ01LM and LYWSD03MMC appear on the dashboard.

### How to reproduce it (minimal and precise)

After upgrading to 1.35.3, this happens all the time.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle-E

### Setup

x86-64 standalone with git clone

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14223115/log.txt)

jonnycastaway, 2024-02-09:
Same?
[https://github.com/Koenkk/zigbee2mqtt/issues/21274](https://github.com/Koenkk/zigbee2mqtt/issues/21274)

Koenkk, 2024-02-10:
Please create issues related to the frontend in this repository: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2024-02-10 by Koenkk