Koenkk/zigbee2mqtt #21323: after Update to 1.35.2-1 OTA update problem, Updates for products they have no Update

Oggymator, 2024-02-09:
### What happened?

Since the new version it show me updates for products that have no update.
New Firmware Version is -1, looks not correct.

### Zigbee2MQTT version

1.35.2-1 

### Adapter firmware version

-

### Adapter

-

### Setup

Home Assistant

bdokka, 2024-02-09:
[image]
[image]
Same issue, screenshots

Oggymator, 2024-02-13:
`Debug 2024-02-13 21:50:16Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Arbeitszimmer Spot 2","transaction":"lsgv8-1"}'
Info 2024-02-13 21:50:16Checking if update available for 'Arbeitszimmer Spot 2'
Debug 2024-02-13 21:50:16Received Zigbee message from 'Arbeitszimmer Spot 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":385953793,"imageType":5206,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Info 2024-02-13 21:50:16No update available for 'Arbeitszimmer Spot 2'
Info 2024-02-13 21:50:16MQTT publish: topic 'zigbee2mqtt/Arbeitszimmer Spot 2', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":40,"state":"OFF","update":{"installed_version":385953793,"latest_version":385953793,"state":"idle"},"update_available":null}'
Info 2024-02-13 21:50:16MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Arbeitszimmer Spot 2","updateAvailable":false},"status":"ok","transaction":"lsgv8-1"}'
Debug 2024-02-13 21:50:18Received Zigbee message from 'Arbeitszimmer Spot 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":385953793,"imageType":5206,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-13 21:50:18Device 'Arbeitszimmer Spot 2' requested OTA
Debug 2024-02-13 21:50:18Responded to OTA request of 'Arbeitszimmer Spot 2' with 'NO_IMAGE_AVAILABLE'`

Koenkk, 2024-02-14:
`"installed_version":385953793,"latest_version":385953793` -> looks the `-1` is not there anymore, so seems to be fixed

Oggymator, 2024-02-15:
> `"installed_version":385953793,"latest_version":385953793` -> looks the `-1` is not there anymore, so seems to be fixed

ok, I have to check all lights of updates again and now its ok.
Thought its checked by the system itself, like the first time.

Problem is solved!

closed 2024-02-15 by Oggymator