Koenkk/zigbee2mqtt #21324: Bosch BTH-RA Radiator thermostat II - broken Window Open state in MQTT

sebi1412, 2024-02-09:
### What happened?

As of version 1.35.2, the MQTT Device in Homeassistant for the Bosch BTH-RA Radiator thermostat II no longer publishes the "Window Open" property and thus, homeassistand cannot access it.

This was fully working until 1.35.1
A rollback of a 1.35.1 backup fixes the issue.

I believe, this has something to do with the latest commits in zigbee-herdsman-converters/src/devices/bosch.ts possible with #6996 or #7029

[image]

### What did you expect to happen?

I expected the availability of the "window open" as a switch to use it in automations.

### How to reproduce it (minimal and precise)

Update from 1.35.1 to 1.35.2 and check the MQQT Devices

### Zigbee2MQTT version

1.35.2 / herdsman converters 18.28.3

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Bosch BTH-RA Radiator thermostat II 

### Setup

Homeassistant on Raspberry Pi 4

danielpetrovic, 2024-02-09:
Nope, it does not. It has to do with other changes and this is a duplicate of: https://github.com/Koenkk/zigbee2mqtt/issues/21208

danielpetrovic, 2024-02-11:
Fixed with next release. If you run Edge version now the fix is there.

DSRocker, 2024-02-17:
I have the same issue with V1.35.3-1

sebi1412, 2024-03-02:
Closed with 1.36.0

closed 2024-05-05 by sebi1412