Koenkk/zigbee2mqtt #21326: [New device support]:

seryozhabelton, 2024-02-09:
### Link

https://www.dresden-elektronik.de/produkt/scene-switch.html

### Database entry

IEEE Address: 0x00212effff01482b

### Comments

I closely followed the "support new devices" page and could pair the Scene Switch so that it shows up in Z2M and outputs messages to the log. However, no matter what I type into the .js file as converter definition, the messages in the log always stay the same. I have the feeling that Z2M does not even attempt to decode the messages. From the messages I get in the log it looks like it should be a cakewalk to write a converter for this switch, based on `fz.genOnOff` and the like in the `fromzigbee` field, but I am at a complete loss and have been ripping my hair our over this for a few days now. Any help or suggestion where I might have gone wrong would be appreciated.

The switch has one dimmer button ("elongated button"), and 4 scene recall buttons (buttons 1-4) in addition. Here is the response I get in the log for using the switches

Elongated button, short press top
``
Debug 2024-02-09 20:27:42Received Zigbee message from 'DresdenSwitch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:42No converter available for 'Scene Switch ' with cluster 'genOnOff' and type 'commandOn' and data '{}'
``

Elongated button, short press bottom
``
Debug 2024-02-09 20:27:43Received Zigbee message from 'DresdenSwitch', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:43No converter available for 'Scene Switch ' with cluster 'genOnOff' and type 'commandOff' and data '{}'
``

Elongated button, start long press top ...
``
Debug 2024-02-09 20:27:46Received Zigbee message from 'DresdenSwitch', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":40}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:46No converter available for 'Scene Switch ' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":0,"rate":40}'
``

... and after releasing the button
``
Debug` 2024-02-09 20:27:46Received Zigbee message from 'DresdenSwitch', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:46No converter available for 'Scene Switch ' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'
``

Elongated button, start long press bottom ...
``
Debug 2024-02-09 20:27:49Received Zigbee message from 'DresdenSwitch', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":40}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:49No converter available for 'Scene Switch ' with cluster 'genLevelCtrl' and type 'commandMove' and data '{"movemode":1,"rate":40}'
``

... and after releasing the button
``
Debug 2024-02-09 20:27:50Received Zigbee message from 'DresdenSwitch', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:50No converter available for 'Scene Switch ' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'
``

Button 1: 
``
Debug 2024-02-09 20:27:51Received Zigbee message from 'DresdenSwitch', type 'commandRecall', cluster 'genScenes', data '{"groupid":10999,"sceneid":1}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:51No converter available for 'Scene Switch ' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":10999,"sceneid":1}'
``

Button 2
``
Debug 2024-02-09 20:27:52Received Zigbee message from 'DresdenSwitch', type 'commandRecall', cluster 'genScenes', data '{"groupid":10999,"sceneid":2}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:52No converter available for 'Scene Switch ' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":10999,"sceneid":2}'
``

Button 3
``
Debug 2024-02-09 20:27:53Received Zigbee message from 'DresdenSwitch', type 'commandRecall', cluster 'genScenes', data '{"groupid":10999,"sceneid":3}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:53No converter available for 'Scene Switch ' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":10999,"sceneid":3}'
``

Button 4
``
Info 2024-02-09 20:27:53MQTT publish: topic 'zigbee2mqtt/DresdenSwitch', payload '{"last_seen":"2024-02-09T19:27:55.169Z","linkquality":0}'
Debug 2024-02-09 20:27:54Received Zigbee message from 'DresdenSwitch', type 'commandRecall', cluster 'genScenes', data '{"groupid":10999,"sceneid":4}' from endpoint 1 with groupID 10999
Debug 2024-02-09 20:27:54No converter available for 'Scene Switch ' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":10999,"sceneid":4}'
``

Each of these messages is followed by a linkquality statement, as the linkquality was recognized as automatically exposed. Looks like this: 
``
Info 2024-02-09 20:27:52MQTT publish: topic 'zigbee2mqtt/DresdenSwitch', payload '{"last_seen":"2024-02-09T19:27:54.287Z","linkquality":36}'
``

### External definition

```shell
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Scene Switch   '],
    model: 'Scene Switch   ',
    vendor: 'dresden elektronik',
    description: 'Automatically generated definition',
    extend: [identify()],
    meta: {},
};

module.exports = definition;
```

closed 2024-09-10 by github-actions[bot]

ftonat, 2024-12-10:
Hi, have you ever found a solution? I am facing same problems. Or any other switch you are using instead?
Frank

seryozhabelton, 2024-12-10:
Hi there, 
I never found a solution - it was an extra switch, so I simply gave up on it. Most of my switches run on Homematic, so it was not absolutely necessary to get the Dresden-Electronik one to work. Annoying, though, as this is not a cheapy little thing....