Koenkk/zigbee2mqtt #21327: TS0201 is detected wrong - detected as TS0201 with display

khayamgondal, 2024-02-09:
### What happened?

TS0201 is detected as a model with display and nothing is reported in z2m dashboard. I am using the latest z2m version. 
[image]
[image]

### Zigbee2MQTT version

 1.35.3 commit: fe0742a

### Adapter firmware version

20230507

### Adapter

SONOFF 3.0 plus

### Setup

Intel NUC with docker

### Debug log

Error 2024-02-09 14:32:40Failed to configure '0xa4c138061b33d470', attempt 3 (Error: Read 0xa4c138061b33d470/1 hvacUserInterfaceCfg([16], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Hoween, 2024-08-20:
Hi all. Today I bought an [THS317-ET-TY](https://www.zigbee2mqtt.io/devices/THS317-ET.html) and having same issue, it's incorrectly recognized as [TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html). According to https://github.com/Koenkk/zigbee2mqtt/issues/19804, THS317 should be correctly supported.

Z2M 1.39.1-1, log from pairing:
`2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/battery/config', payload 'null'
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffeac74b3', payload ''
info 2024-08-20 20:04:53z2m: Successfully removed device '0x0ceff6fffeac74b3' (block: false, force: true)
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x0ceff6fffeac74b3"},"status":"ok","transaction":"uvq23-1"}'
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/temperature/config', payload 'null'
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/humidity/config', payload 'null'
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/voltage/config', payload 'null'
info 2024-08-20 20:04:53z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/linkquality/config', payload 'null'
info 2024-08-20 20:04:57z2m: Zigbee: allowing new devices to join via 0xe0798dfffeb90ed8.
info 2024-08-20 20:04:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"0xe0798dfffeb90ed8","time":254,"value":true},"status":"ok","transaction":"uvq23-2"}'
info 2024-08-20 20:05:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x943469fffe716aee', payload '{"child_lock":"UNLOCK","current":0.03,"energy":30.1,"indicator_mode":"off/on","linkquality":176,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":69,"latest_version":69,"state":"idle"},"update_available":null,"voltage":244}'
info 2024-08-20 20:05:06zh:controller: Interview for '0x0ceff6fffeac74b3' started
info 2024-08-20 20:05:06z2m: Device '0x0ceff6fffeac74b3' joined
info 2024-08-20 20:05:06z2m: Starting interview of '0x0ceff6fffeac74b3'
info 2024-08-20 20:05:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffeac74b3","ieee_address":"0x0ceff6fffeac74b3"},"type":"device_joined"}'
info 2024-08-20 20:05:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffeac74b3","ieee_address":"0x0ceff6fffeac74b3","status":"started"},"type":"device_interview"}'
info 2024-08-20 20:05:08zh:controller: Succesfully interviewed '0x0ceff6fffeac74b3'
info 2024-08-20 20:05:08z2m: Successfully interviewed '0x0ceff6fffeac74b3', device has successfully been paired
info 2024-08-20 20:05:08z2m: Device '0x0ceff6fffeac74b3' is supported, identified as: Tuya Temperature & humidity sensor with display (TS0201)
info 2024-08-20 20:05:08z2m: Configuring '0x0ceff6fffeac74b3'
info 2024-08-20 20:05:08z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffeac74b3/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0ceff6fffeac74b3"],"manufacturer":"Tuya","model":"Temperature & humidity sensor with display (TS0201)","name":"0x0ceff6fffeac74b3","via_device":"zigbee2mqtt_bridge_0xe0798dfffe83a219"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x0ceff6fffeac74b3_battery","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x0ceff6fffeac74b3","unique_id":"0x0ceff6fffeac74b3_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2024-08-20 20:05:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor with display","exposes":[{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","label":"Temperature","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","label":"Humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0201","options":[{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","label":"Temperature calibration","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device. This option can only decrease the precision, not increase it.","lab [... cut ...] by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x0ceff6fffeac74b3_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x0ceff6fffeac74b3","unique_id":"0x0ceff6fffeac74b3_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2024-08-20 20:05:09z2m: Successfully configured '0x0ceff6fffeac74b3'`

AlexDelm, 2024-09-03:
Hi,
same issue here, THS317-ET-TY incorrectly recognized as [TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html).
[image]

closed 2025-01-08 by github-actions[bot]