Koenkk/zigbee2mqtt #21329: Zigbee2mqtt won't start in HA OS VM

eswenson1, 2024-02-10:
### What happened?

When I try to start zigbee2mqtt in Home Assistant (HA OS VM running under Ubuntu 22.04 LTS), it fails with the following messages in the log:

```
[17:03:55] INFO: Preparing to start...
[17:03:55] INFO: Socat not enabled
[17:03:57] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-02-09 17:03:59: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-09.17-03-59' filename: log.txt
Zigbee2MQTT:info  2024-02-09 17:03:59: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-09 17:03:59: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-02-09 17:04:19: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-09 17:04:19: Failed to start zigbee
Zigbee2MQTT:error 2024-02-09 17:04:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-09 17:04:19: Exiting...
Zigbee2MQTT:error 2024-02-09 17:04:19: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```
I have verified that I have write access to /dev/ttyUSB0.  In fact, using the HA Zigbee Home Automation add-on, it can connect to the ZiGate USB-TTL 3.21 USB Dongle (a ThirdReality Zibgee USB Dongle) fine, and even get it into pairing mode.  In that mode, however, it fails to find my ThirdReality Wireless Motion Sensor, so upon recommendations of various people, I'm trying zigbee2mqtt. But as you can see, above, it never starts successfully.  I've even turned on zigbee-herdsman debug logging and see nothing useful in the logs.

I do have the mosquitto add-on configured.

What do you recommend?

### What did you expect to happen?

I expected zigbee2mqtt to come up.  

### How to reproduce it (minimal and precise)

Simply bring up HA OS (latest version) in a QEMU VM on Ubuntu 22.04.  Install the mosquitto add-on and then the zigbee2mqtt add-on.  Ensure that the VM can see and write to the /dev/ttyUSB0 device (the Zigbee USB dongle) and try to start the zigbee2mqtt add-on.  It will fail with the above messages (at least on my setup).

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

Can't get to frontend.

### Adapter

ThirdReality Zigbee USB Dongle

### Setup

Add-on in Home Assistant. HA OS running in QEMU VM.

### Debug log

```
[17:19:03] INFO: Preparing to start...
[17:19:04] INFO: Socat not enabled
[17:19:04] INFO: Zigbee Herdsman debug logging enabled
[17:19:05] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2024-02-09 17:19:07: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-02-09 17:19:07: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-09.17-19-07' filename: log.txt
Zigbee2MQTT:debug 2024-02-09 17:19:07: Removing old log directory '/config/zigbee2mqtt/log/2024-02-09.16-59-58'
Zigbee2MQTT:info  2024-02-09 17:19:07: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-09 17:19:07: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:debug 2024-02-09 17:19:07: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[211,232,136,60,94,15,219,154],"networkKey":"HIDDEN","panID":42162},"serialPort":{"path":"/dev/ttyUSB0"}}'
2024-02-10T01:19:07.799Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2024-02-10T01:19:07.801Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[132,64,49,19,144,154,102,183,88,79,20,19,229,5,74,18],"panID":42162,"extendedPanID":[211,232,136,60,94,15,219,154],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2024-02-10T01:19:07.801Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-02-10T01:19:07.826Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2024-02-10T01:19:07.828Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2024-02-10T01:19:07.829Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2024-02-10T01:19:08.082Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2024-02-10T01:19:08.082Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2024-02-10T01:19:09.086Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2024-02-10T01:19:09.087Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2024-02-10T01:19:09.339Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2024-02-10T01:19:09.811Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2024-02-10T01:19:09.813Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2024-02-10T01:19:15.819Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2024-02-10T01:19:15.821Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2024-02-10T01:19:21.823Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2024-02-10T01:19:21.823Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2024-02-09 17:19:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-09 17:19:27: Failed to start zigbee
Zigbee2MQTT:error 2024-02-09 17:19:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-09 17:19:27: Exiting...
Zigbee2MQTT:error 2024-02-09 17:19:27: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

chris-1243, 2024-02-10:
Could you provide your configuration.yaml ?

eswenson1, 2024-02-10:
```
# Loads default set of integrations. Do not remove.
default_config:

# Load frontend themes from the themes folder
frontend:
  themes: !include_dir_merge_named themes

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
```

eswenson1, 2024-02-10:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: debug
zigbee_herdsman_debug: true
```

chris-1243, 2024-02-10:
According to your log, Z2M tries to use z-stack. As you are using a zigate coordinator, you should modify your configuration.yaml as shown below.
```
serial:
  port: /dev/ttyUSB0
  adapter: zigate
``` 

You may check the documentation via this [link](https://www.zigbee2mqtt.io/guide/adapters/#experimental)

eswenson1, 2024-02-10:
Thanks. That appears to allow it to progress further. The log now says:
```
[09:50:49] INFO: Preparing to start...
[09:50:49] INFO: Socat not enabled
[09:50:49] INFO: Zigbee Herdsman debug logging enabled
[09:50:50] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2024-02-10 09:50:53: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-02-10 09:50:53: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-10.09-50-53' filename: log.txt
Zigbee2MQTT:debug 2024-02-10 09:50:53: Removing old log directory '/config/zigbee2mqtt/log/2024-02-09.17-01-28'
Zigbee2MQTT:info  2024-02-10 09:50:53: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-10 09:50:53: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:debug 2024-02-10 09:50:53: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[211,232,136,60,94,15,219,154],"networkKey":"HIDDEN","panID":42162},"serialPort":{"adapter":"zigate","path":"/dev/ttyUSB0"}}'
2024-02-10T17:50:53.501Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2024-02-10T17:50:53.503Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[132,64,49,19,144,154,102,183,88,79,20,19,229,5,74,18],"panID":42162,"extendedPanID":[211,232,136,60,94,15,219,154],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zigate"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2024-02-10T17:50:53.542Z zigbee-herdsman:adapter:zigate:driver:log Serialport opened
2024-02-10T17:50:53.543Z zigbee-herdsman:adapter:zigate:adapter:log Connected to ZiGate adapter successfully.
2024-02-10T17:50:53.548Z zigbee-herdsman:adapter:zigate:driver:log Send command >>>> Reset 0x0011 <<<< 
Payload: {}
2024-02-10T17:50:53.554Z zigbee-herdsman:adapter:zigate:driver:log ZiGateFrame { msgCodeBytes: <Buffer 00 11>, msgLengthBytes: <Buffer 00 00>, checksumBytes: <Buffer 11>, msgPayloadBytes: <Buffer >, rssiBytes: <Buffer >, msgLengthOffset: 0 }
2024-02-10T17:50:53.556Z zigbee-herdsman:adapter:zigate:driver:log <-- send command  <Buffer 01 02 10 11 02 10 02 10 11 03>
2024-02-10T17:50:53.556Z zigbee-herdsman:adapter:zigate:driver:log DisableResponse: false
```
But when I return to the Info tab, the START button is still available, and I don't see any button to allow me to connect to the web ui for zigbee2mqtt.  It seems it actually didn't start.

eswenson1, 2024-02-10:
Actually, waiting a little longer, I get these additional messages in the log:
```
Zigbee2MQTT:error 2024-02-10 09:52:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-10 09:52:59: Failed to start zigbee
Zigbee2MQTT:error 2024-02-10 09:52:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-10 09:52:59: Exiting...
Zigbee2MQTT:error 2024-02-10 09:53:00: Error: failed to connect to zigate adapter [object Object] after 5000ms
    at ZiGateAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/zigate/adapter/zigateAdapter.ts:101:19)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

chris-1243, 2024-02-10:
I read you had successful tries with ZHA. Is the addon off completely? It may occur ZHA could still be connected to your coordinator.

eswenson1, 2024-02-10:
I had not disabled ZHA.  Disabling it leads to these logs when starting zigbee2mqt:
```
2024-02-10T19:34:56.398Z zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 09>, checksumBytes: <Buffer 99>, msgPayloadBytes: <Buffer 00 00 00 10 00 00 00 00>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 }
2024-02-10T19:34:56.399Z zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< 
2024-02-10T19:34:56.400Z zigbee-herdsman:adapter:zigate:driver:ziGateObject:error Last bytes of data were not parsed 00 00 00 10 00 00 00 00  
2024-02-10T19:34:56.400Z zigbee-herdsman:adapter:zigate:driver:log { status: 0, sequence: 0, packetType: 16 }
2024-02-10T19:34:56.404Z zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 10>, msgLengthBytes: <Buffer 00 05>, checksumBytes: <Buffer b4>, msgPayloadBytes: <Buffer 00 03 03 21>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 }
2024-02-10T19:34:56.404Z zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> VersionList 0x8010 <<<< 
2024-02-10T19:34:56.405Z zigbee-herdsman:adapter:zigate:driver:log { major: 0, minor: 3, revision: 8451 }
Zigbee2MQTT:info  2024-02-10 11:34:56: Coordinator firmware version: '{"meta":{"maintrel":"2103","majorrel":"0","minorrel":"3","product":0,"revision":"2103","transportrev":0},"type":"zigate"}'
2024-02-10T19:34:56.408Z zigbee-herdsman:adapter:zigate:adapter:log getNetworkParameters
2024-02-10T19:34:56.409Z zigbee-herdsman:adapter:zigate:driver:log Send command >>>> GetNetworkState 0x0009 <<<< 
Payload: {}
2024-02-10T19:34:56.410Z zigbee-herdsman:adapter:zigate:driver:log ZiGateFrame { msgCodeBytes: <Buffer 00 09>, msgLengthBytes: <Buffer 00 00>, checksumBytes: <Buffer 09>, msgPayloadBytes: <Buffer >, rssiBytes: <Buffer >, msgLengthOffset: 0 }
2024-02-10T19:34:56.410Z zigbee-herdsman:adapter:zigate:driver:log <-- send command  <Buffer 01 02 10 02 19 02 10 02 10 02 19 03>
2024-02-10T19:34:56.410Z zigbee-herdsman:adapter:zigate:driver:log DisableResponse: false
2024-02-10T19:34:56.425Z zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 09>, checksumBytes: <Buffer 80>, msgPayloadBytes: <Buffer 00 00 00 09 00 00 00 00>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 }
2024-02-10T19:34:56.426Z zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< 
2024-02-10T19:34:56.426Z zigbee-herdsman:adapter:zigate:driver:ziGateObject:error Last bytes of data were not parsed 00 00 00 09 00 00 00 00  
2024-02-10T19:34:56.426Z zigbee-herdsman:adapter:zigate:driver:log { status: 0, sequence: 0, packetType: 9 }
2024-02-10T19:34:56.519Z zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 09>, msgLengthBytes: <Buffer 00 16>, checksumBytes: <Buffer 09>, msgPayloadBytes: <Buffer 00 00 b8 3d fb 60 83 22 00 00 23 04 dd dd dd dd dd dd dd dd 0f>, rssiBytes: <Buffer 01>, msgLengthOffset: -1 }
2024-02-10T19:34:56.519Z zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> NetworkState 0x8009 <<<< 
2024-02-10T19:34:56.521Z zigbee-herdsman:adapter:zigate:driver:log { shortAddress: 0, extendedAddress: '0xb83dfb6083220000', PANID: 8964, ExtPANID: '0xdddddddddddddddd', Channel: 15 }
Zigbee2MQTT:debug 2024-02-10 11:34:56: Zigbee network parameters: {"channel":15,"extendedPanID":"0xdddddddddddddddd","panID":8964}
Zigbee2MQTT:info  2024-02-10 11:34:56: Currently 0 devices are joined:
Zigbee2MQTT:info  2024-02-10 11:34:56: Zigbee: disabling joining new devices.
2024-02-10T19:34:56.534Z zigbee-herdsman:controller:log Disable joining
2024-02-10T19:34:56.539Z zigbee-herdsman:adapter:zigate:adapter:error source endpoint 242, not supported
2024-02-10T19:34:56.541Z zigbee-herdsman:adapter:zigate:driver:log Send command >>>> PermitJoin 0x0049 <<<< 
Payload: { targetShortAddress: 65532, interval: 0, TCsignificance: 0 }
2024-02-10T19:34:56.542Z zigbee-herdsman:adapter:zigate:driver:log ZiGateFrame { msgCodeBytes: <Buffer 00 49>, msgLengthBytes: <Buffer 00 03>, checksumBytes: <Buffer 49>, msgPayloadBytes: <Buffer ff fc 00>, rssiBytes: <Buffer >, msgLengthOffset: 0 }
2024-02-10T19:34:56.543Z zigbee-herdsman:adapter:zigate:driver:log <-- send command  <Buffer 01 02 10 49 02 10 02 13 49 ff fc 02 10 03>
2024-02-10T19:34:56.544Z zigbee-herdsman:adapter:zigate:driver:log DisableResponse: false
2024-02-10T19:34:56.561Z zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 09>, checksumBytes: <Buffer c0>, msgPayloadBytes: <Buffer 00 00 00 49 00 00 00 00>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 }
2024-02-10T19:34:56.562Z zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< 
2024-02-10T19:34:56.563Z zigbee-herdsman:adapter:zigate:driver:ziGateObject:error Last bytes of data were not parsed 00 00 00 49 00 00 00 00  
2024-02-10T19:34:56.563Z zigbee-herdsman:adapter:zigate:driver:log { status: 0, sequence: 0, packetType: 73 }
Zigbee2MQTT:info  2024-02-10 11:34:56: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:debug 2024-02-10 11:34:56: Using MQTT login with username: addons
Zigbee2MQTT:info  2024-02-10 11:34:56: Connected to MQTT server
Zigbee2MQTT:info  2024-02-10 11:34:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[... cut ...] :"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0xb83dfb6083220000_permit_join_zigbee2mqtt","value_template":"{{ value_json.permit_join | lower }}"}'
```
But switching to Info tab still shows START button active, and no web ui button.

Why is it checking the version so many times?  Why is it unable to parse some bytes at the end of the response?  Can you tell from the log what it is unhappy about?

eswenson1, 2024-02-10:
Hmm.... it seems it is started (despite having a START button and no web ui button).  I had added zibgee2mqtt to the sidebar, and clicking on it does bring up a UI.  My ThirdReality Zigbee Motion Sensor, of course, was not showing up. So I clicked the button "Permit Join" button, and the blue light on my ThirdReality Zigbee USB Dongle started flashing.  I placed my ThirdReality Zigbee Motion Sensor near my USB Dongle, and put it into pairing mode.  It started flashing blue and the USB Dongle did a blue/green led dance, and then eventually, my Motion Sensor went back to not flashing (only flashing red when it sensed motion).  

The zigbee2mqtt log shows this:
```
debug 2024-02-10 11:43:41Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"kninv-1","value":true}'
info 2024-02-10 11:43:41Zigbee: allowing new devices to join.
info 2024-02-10 11:43:41MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"kninv-1"}'
debug 2024-02-10 11:44:54Saving state to file /config/zigbee2mqtt/state.json
debug 2024-02-10 11:49:54Saving state to file /config/zigbee2mqtt/state.json
```
I don't see any devices found in the devices list.  What now?

chris-1243, 2024-02-10:
Would you just confirm ZHA is disabled. You can not have ZHA and Z2M trying to use the same coordinator. It is either ZHA or Z2M.
The herdsmann debug log is out of my scope of knowledges. As Zigate adapter are still experimental in Z2M, I would suggest some codes have to be improved.
Some end devices (like motion sensors, remotes, etc) are know to be hard to pair. You will have to retry several times if not successful the first time

threadstone83, 2024-02-10:
Since today I have the same problem. I can not start Z2M. I have nothing changed in the last days, only updated to 1.35.3-1

eswenson1, 2024-02-10:
> Would you just confirm ZHA is disabled. You can not have ZHA and Z2M trying to use the same coordinator. It is either ZHA or Z2M.

Yes. I disabled ZHA. That is what allowed things to progress to the point they have. 

It would appear zigbee2mqt is installed and running now, and can talk to the Zigbee USB dongle. I just can’t (yet) pair my motion sensor.

> Some end devices (like motion sensors, remotes, etc) are known to be hard to pair. You will have to retry several times if not successful the first time

ok. I’ll try it a few more times. I also should attach usb dongle via an extension cable to get it away from other radios.

closed 2024-09-08 by github-actions[bot]