Koenkk/zigbee2mqtt #21330: [New device support]:zigbeeModel: ['LXN56-TS27LX1.2'],    vendor: '3A Smart Home DE' Double GPO Black touch plate

kamaldeepdhiman, 2024-02-10:
### Link

https://www.amazon.com.au/Approved-ZigBee-Outlet-Automation-Control/dp/B09CPYY4BN

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x00158d0006af743a","nwkAddr":25656,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN56-TS27LX1.2","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,3,6,5,4096],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255,"onTime":0,"globalSceneCtrl":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x287681fffef01b38","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,4,3,6,5],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x287681fffef01b38","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019.11.11","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1707554128445}

### Comments

I tried to follow new device support page but no success. please add this double GPO to zigbee2mqtt support. This device can be add to Philips hue bridge if you have one.

### External definition

```shell
const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LXN56-TS27LX1.2'],
    model: 'LXN56-TS27LX1.2',
    vendor: '3A Smart Home DE',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpoints":{"1":1,"2":2}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Koenkk, 2024-02-18:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/559a9856e15f67223cf10dfea98b3ba0): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

kamaldeepdhiman, 2024-02-18:
hi there, should I add the below lines in the ext_converters.js file.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXN56-TS27LX1.2'],
    model: 'LXN56-TS27LX1.2',
    vendor: 'Nue / 3A',
    description: 'Smart light relay - 2 gang',
    extend: [onOff({endpoints: {left: 1, right: 2}})],
    whiteLabel: [{vendor: 'Zemismart', model: 'ZW-EU-02'}],
};

module.exports = definition;

I try both methods with above lines and without above lines in the ext_converters.js file but no success.

Koenkk, 2024-02-19:
What doesn't work?

kamaldeepdhiman, 2024-02-19:
> What doesn't work?
After put in new external converters file( you suggest ) not exposed to on and off switch.
It's same like initially I had issue. It pair with zigbee2mqtt but said not supported.

Koenkk, 2024-02-20:
Can you provide the debug log when starting z2m with this ext converter?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

kamaldeepdhiman, 2024-02-21:
hi there, mate I am sending you my zigbee2mqtt log file. I setup according to you what you suggest in previous reply. after log debug mode setting my zigbee2mqtt running on portioner fail to start and I do not have access through frontend. I get log file through portioner.
[_zigbee2mqtt_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/14356900/_zigbee2mqtt_logs.txt)

Koenkk, 2024-02-22:
Your configuration.yaml syntax is wrong, paste it in https://www.yamllint.com/ to find the error.

kamaldeepdhiman, 2024-02-23:
hi there, I try above link you sent me and paste all configuration.yaml file in there then click on Go. it says
Valid YAML!

kamaldeepdhiman, 2024-03-02:
Hi there, any help regarding previous new device support I was asking for. Please

Koenkk, 2024-03-02:
I've updated https://gist.github.com/Koenkk/559a9856e15f67223cf10dfea98b3ba0, can you try again?

kamaldeepdhiman, 2024-03-03:
hi there, mate I update my ext_converters.js file according to link you provided above but I do not get success to control the GPO. It's still paired but says not supported. sending you latest log file attached with it.
[_zigbee2mqtt_logs-4.txt](https://github.com/Koenkk/zigbee2mqtt/files/14472574/_zigbee2mqtt_logs-4.txt)

Koenkk, 2024-03-04:
Are you sure the contents of your external converter is correct?

kamaldeepdhiman, 2024-03-05:
hi there, I use the same file you sent me. See the ext_converters.js file attached.
[ext_converter.txt](https://github.com/Koenkk/zigbee2mqtt/files/14492693/ext_converter.txt)
I change extension to .txt for sending purpose only. original file in .js extension I am unable to attach.

Koenkk, 2024-03-05:
I've updated https://gist.github.com/Koenkk/559a9856e15f67223cf10dfea98b3ba0, can you try again with z2m 1.36.0?

kamaldeepdhiman, 2024-03-06:
hi there, mate I try above file you sent me early today but no success. same issue

Koenkk, 2024-03-14:
Updated https://gist.github.com/Koenkk/559a9856e15f67223cf10dfea98b3ba0 , could you try again?

jamesahendry, 2024-06-02:
I have submitted a working external converter for this device under https://github.com/Koenkk/zigbee2mqtt/issues/22887.  Please use the External Definition under the new issue I have logged.   Tested in 1.38.0 by removing my external converters to confirm it is not currently natively supported.

kamaldeepdhiman, 2024-06-08:
i use external convertor of link below
https://github.com/Koenkk/zigbee2mqtt/issues/22887 and ext_converter.js file below as well
my double GPO working fine. no issues. Thanks
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXN56-TS27LX1.2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'MG-AUSW02', // Vendor model number, look on the device for a model number
    vendor: 'Nue / 3A', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart Double GPO', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later // fz.battery, fz.temperature, fz.humidity
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right') ],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {left: 1, right: 2};
    },
};

module.exports = definition;

closed 2024-06-08 by kamaldeepdhiman