Koenkk/zigbee2mqtt #21331: [Feature request]: Website needs a proper introduction

schauveau, 2024-02-10:
### Is your feature request related to a problem? Please describe

The website seems very nice with a lot of documentation but it can be quite confusing for newbies because it does not really explain what  Zigbee2MQTT is really about. 

Let's consider my experience for example. I am a software engineer.  I know pretty much nothing about IoT but I am investigating for my future system that will probably be using Zigbee devices and Home Assistant. At one point I see a comment where someone mentions that he is very happy to with his Zigbee2MQTT+HomeAssistant configuration so I end up on your website.  

The main page tells me that this is a "Zigbee to MQTT bridge". Some of the questions that come to my mind are

- What is MQTT? There no link but a quick Google search tells me that is another protocol for IoT. Why would I need that?
- How does it fit with Home Assistant and the Zigbee adapter (e.g. SONOFF or other)? 
- Why should I bother adding an additional MQTT layer since Home Assistant can already talk to Zigbee directly?

So I click the **Getting Started** button and I end up in the **Prerequisite** section that tells me that I need a Zigbee adapter but also a MQTT broker. Remember that I still do not really know what Zigbee2MQTT is really about. 

The next sections are **Installation**, **Connect a device**, **Supported Hardware**, ... but nothing that can really help me understand the real purpose of Zigbee2MQTT.

The **Frontend** section tells me that Zigbee2MQTT has a built-in web-based interface. Why? Is that redundant with a HomeAssistant? Things are becoming quite confusing.

The FAQ contains troubleshooting question that are not helpful either. 

So after about 20 minute browsing through the doc, I still do not really understand what this is about and I gave up on Zigbee2MQTT.

The only reason, I am back is because I eventually found a YT video where someone is explaining his setup and the advantages of using Zigbee2MQTT. 

### Describe the solution you'd like

So to summarize, what is missing in the documentation is a short introduction about
- The purpose of Zigbee2MQTT with links to Zigbee and MQTT resources.
- How Zigbee2MQTT fits in in a typical system. 
- Why I should consider Zigbee2MQTT and what are the pros and cons

Later, I found the **Architecture** diagram in the github Readme. This is a good start but it requires some additional explanations to be really helpful.  

### Describe alternatives you've considered

none

### Additional context

none

Hypfer, 2024-02-14:
The docs are just markdown which you can edit and propose changes to on GitHub.

You can find them here: https://github.com/Koenkk/zigbee2mqtt.io

Koenkk, 2024-02-15:
@schauveau it would be really nice if you could make a PR or proposal what changes you want to see. I think I'm too deep into this to write a simple introduction for new users.

schauveau, 2024-02-16:
The problem for me is the opposite. I am a new user or, should I say, I may become a new user since I never used Zigbee2MQTT nor do I own any Zigbee device. For me, all that is still very abstract.  My current understanding of Zigbee2MQTT is probably very limited and incorrect.

closed 2024-09-14 by github-actions[bot]