Koenkk/zigbee2mqtt #21336: [Wrong device]: Philips 9290030675

kenni, 2024-02-10:
### Link

https://www.zigbee2mqtt.io/devices/9290030675.html

### Model

9290030675

### Description

Hue motion sensor

### Vendor

Philips

### Picture (link)

https://drive.google.com/file/d/1rzGk0zqj8fHMS31Czq--gDbl38dDlFbC/view?usp=sharing

### Database entry

{"id":75,"type":"EndDevice","ieeeAddr":"0x001788010bcd3487","nwkAddr":30714,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"SML003","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,1024,1026],"outClusterList":[25,0,3,6],"clusters":{"genBasic":{"attributes":{"modelId":"SML003","manufacturerName":"Signify Netherlands B.V.","powerSource":3,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6"}},"msOccupancySensing":{"attributes":{"48":2,"64":2,"65":2,"66":0,"67":2,"68":1,"occupancy":0,"pirOToUDelay":0}},"genPowerCfg":{"attributes":{"61441":3175,"61442":3145,"61443":3178,"batteryPercentageRemaining":200}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2005}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a53a32","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a53a32","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a53a32","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a53a32","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":1030,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":1024,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null}],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1707578356497}

### Notes

Current picture is broken as everything around the very center of the device has been marked as transparent (see issue on https://www.zigbee2mqtt.io/devices/9290030675.html). This new picture fixes it and is also the correct version with "Philips Hue"-text on top of the device (similar device 9290012607 only has "Philips"-text on top).

Koenkk, 2024-02-10:
Im already fixing all broken pictures, will take some time

kenni, 2024-02-10:
Sounds good - hopefully the attached picture can help you save some time. Let me know if it should be in a different format or size, I just made it 512x512 in PNG with transparent background as that was the format used by its sibling (9290012607) in the Z2M database.

closed 2024-09-08 by github-actions[bot]