Koenkk/zigbee2mqtt #21339: [New device support]: SMLIGHT SLZB-06M (as router)

nicola-spreafico, 2024-02-10:
### Link

https://smlight.tech/product/slzb-06m/

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0xe0798dfffe8849ff","nwkAddr":45488,"manufId":4098,"manufName":"SMLIGHT","powerSource":"Unknown","modelId":"SLZB-06M","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25,1030],"clusters":{"genBasic":{"attributes":{"modelId":"SLZB-06M","manufacturerName":"SMLIGHT","appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"","zclVersion":8}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x943469fffe71ac61","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"","swBuildId":"0.1","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707578639166}

### Comments

Although this device is bought primarily as a coordinator, you can also flash firmware from a router, which I did.

The device is recognized as a router and works properly within the network, however it is listed as "not supported".

Being only a router I don't think there is something to actually "support", but I guess that if it can be officially recognized, we can remove the "not supported" label.

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SLZB-06M'],
    model: 'SLZB-06M',
    vendor: 'SMLIGHT',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-03-24:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-24 by Koenkk

marianob85, 2024-03-24:
Looks good. One question remain. Is it possible to show power status ? Right now is marked as question mark.

Koenkk, 2024-03-25:
done

DittmannCz, 2024-07-24:
Hello, how can I mod it to see it as suppoprted?
Sorry, I have no idea...
Thank You