Koenkk/zigbee2mqtt #21343: T&H Sensor TuYa ZTH08-E: Battery state is always none

rhmunich, 2024-02-10:
### What happened?

I do have multiple devices of the **Temperature & Humidity sensor TuYa ZTH08-E** integrated.
[https://www.zigbee2mqtt.io/devices/ZTH08-E.html](https://www.zigbee2mqtt.io/devices/ZTH08-E.html)

For each the entity `sensor.t_h_sensor_battery_state` is available in Home Assistant but is always with the state `none`. 

### What did you expect to happen?

The values for the `battery_state` should be one of the following: `low`, `medium`,` high`

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add-on on Home Assistant OS on Intel NUC

rhmunich, 2024-02-15:
[image]
[image]

grericht, 2024-05-06:
is there something new? i have 4 of these devices. 3 send battery state. 1 always "none". dlete and relink was not useful

rhmunich, 2024-05-06:
No, no improvement at my end. Battery state is still "none".

thomaswoeckinger, 2024-05-31:
Also on my site two are sending battery state two not

thomaswoeckinger, 2024-06-02:
> Also on my site two are sending battery state two not.
> 
> Delete and repair does not solve the problem

I finally solved it by removing the batteries wait 10 seconds and Insert it again
It seems that the state is ony transmitted during change which is also forced when putting them Our and in again

rhmunich, 2024-06-02:
> I finally solved it by removing the batteries wait 10 seconds and Insert it again It seems that the state is ony transmitted during change which is also forced when putting them out and in again

Great. This also helped at my end. Thanks Thomas!

closed 2024-06-02 by rhmunich