Koenkk/zigbee2mqtt #21346: z2m crashes when trying to add new device

rampestamper, 2024-02-10:
### What happened?

As soon as i try to add a new device to z2m it crashes. its detecting the new bulb(s) and is initializing. after that is crashes. then when it is offline and i remove the unfinished devices to try again i get about 20-30 messages that "Device (ID) left the network" after that "adapter disconnected, stopping"

somtimes it just restarts and then it is ok again (until i try to pair a device again) otherwise it gives the following error:
[23:28:54] INFO: Preparing to start...
[23:28:55] INFO: Socat not enabled
[23:28:59] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-02-10 23:29:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-10 23:29:27: Failed to start zigbee
Zigbee2MQTT:error 2024-02-10 23:29:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-10 23:29:27: Exiting...
Zigbee2MQTT:error 2024-02-10 23:29:27: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,8]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:646:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:456:13)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:167:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
 

### What did you expect to happen?

Device will be added to z2m

### How to reproduce it (minimal and precise)

see What happened

### Zigbee2MQTT version

1.35.3 commit: unknown

### Adapter firmware version

 6.10.7.0 build 435

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Home assistant OS

### Debug log

[23:28:54] INFO: Preparing to start...
[23:28:55] INFO: Socat not enabled
[23:28:59] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-02-10 23:29:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-10 23:29:27: Failed to start zigbee
Zigbee2MQTT:error 2024-02-10 23:29:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-10 23:29:27: Exiting...
Zigbee2MQTT:error 2024-02-10 23:29:27: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,8]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:646:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:456:13)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:167:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

klu16, 2024-02-11:
I assume that "Sonoff Dongle Plus V2" is the same as Dongle-E? If so, I would start by bumping the controller firmware version to 7.3.1.0.  For me it fixed some of the issues that I had (not all), but also crashing while pairing some end devices.

rampestamper, 2024-02-11:
> I assume that "Sonoff Dongle Plus V2" is the same as Dongle-E? If so, I would start by bumping the controller firmware version to 7.3.1.0.  From my personal experience 6.10.x is not stable for Dongle-E. For me the update fixed some of the issues that I had (not all), but also crashing while pairing some end devices. 

So that was a eventful morning. I updated the firmware of indeed my Dongle-E to 7.3.1.0 first alll of my devices where gone. But my network was crashing even more then before. I restored a backup from last night. That gave me all my devices back. But my network keeps crashing. Even when just starting up zigbee2mqtt it crashes. Need to try a couple of times to even get it online. 

Anybody a solution?

klu16, 2024-02-11:
It keeps crashing constantly with v1.35.3. Last stable for me is 1.35.1 or older.

rampestamper, 2024-02-11:
> It keeps crashing constantly with v1.35.3. Last stable for me is 1.35.1 or older.

ok do you know how i can downgrade back to v1.35.1?

klu16, 2024-02-11:
As far as I know, officially only from the backup.
Go to Settings > System > Backups and check if you have a Z2M backup there...

rampestamper, 2024-02-11:
So thankfully i still had the backup for the z2m update. so i reverted back and it started back up. now back to the problem with crashing when i add a device. that problem still exists. but i think the log is a bit different now?

[12:45:17] INFO: Preparing to start...
[12:45:18] INFO: Socat not enabled
[12:45:21] INFO: Starting Zigbee2MQTT...
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Zigbee2MQTT:error 2024-02-11 12:45:50: Publish 'set' 'brightness' to 'Off_Lgt_2' failed: 'Error: Command 0x00158d0004b27736/1 genLevelCtrl.moveToLevelWithOnOff({"level":203,"transtime":20}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Zigbee2MQTT:error 2024-02-11 12:45:55: Publish 'set' 'color_temp' to 'Off_Lgt_2' failed: 'Error: Command 0x00158d0004b27736/1 lightingColorCtrl.moveToColorTemp({"colortemp":200,"transtime":20}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Error: DatabaseEntry with ID '46' already exists
    at Database.insert (/app/node_modules/zigbee-herdsman/src/controller/database.ts:40:19)
    at Device.undelete (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:438:25)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:555:20)

rampestamper, 2024-02-11:
> As far as I know, officially only from the backup. Go to Settings > System > Backups and check if you have a Z2M backup there...

So i am a bit further now. i added 2 of the 3 lights. when i try to add the thirth light i get a lot of warnings that say device (ID) left the network. after that z2m crashes. but the strange thing is that that device is not showing in my device list?

rampestamper, 2024-02-11:
So i dont know what i did. but it is working now. To make it hopefully a bit more stable i just to be sure ordered the Sonoff dongle-P to migrate to. see if that makes any difference

closed 2024-02-11 by rampestamper