Koenkk/zigbee2mqtt #21348: TuYa button switch sends "1_single" instead of "single" event

PontifexPrimus, 2024-02-11:
### What happened?

After updating to 1.35.3-1 one of my switches stopped working. Checking the log showed that the events that were previously recorded as 

Zigbee2MQTT:info  2024-02-11 09:31:47: MQTT publish: topic 'zigbee2mqtt/ZigBeeButton1/action', payload 'single'

were now logged as

Zigbee2MQTT:info  2024-02-11 09:31:47: MQTT publish: topic 'zigbee2mqtt/ZigBeeButton1/action', payload '1_single'

### What did you expect to happen?

I had expected that the update would not change the payload and so make my automations not trigger.

### How to reproduce it (minimal and precise)

It happened during the update to the latest version (1.35.3-1) without any manual intervention from me.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

zStack3x0

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on home assistant running on a mini PC

matkes, 2024-02-11:
I have a similar issue with Smart Buttons TuYa IH-K663 and TuYa SH-SC07.
Two new actions are now available: "1_single" and "1_double". Previous "single" and "double" are no longer working. In order for existing (or new) automations to work, "1_single" and "1_double" must be now used. Zigbee2mqtt documentation mentions the following values for actions: single, double, hold.
[image]

PontifexPrimus, 2024-02-11:
Additional information: zigbee model for the switch with the above listed behavior is TS0041, manufacturer _TZ3000_8rppvwda.

seblu, 2024-02-11:
Probably duplicate of  #21301

Koenkk, 2024-02-12:
Should already be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

NeoFax75, 2024-04-07:
No, it is not totally fixed. Single and double work, but hold sends a null.

Koenkk, 2024-04-08:
could you provide the debug log of this with the latest dev branch?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

closed 2024-11-06 by github-actions[bot]