Koenkk/zigbee2mqtt #21349: Latest zigbee2mqtt (1.35.3) fails to communicate with devices, 1.35.1 works

mmithril, 2024-02-11:
### What happened?

My server disk went down. I reinstalled the operating system & recreated the home automation docker composition so latest images (i.e. zigbee2mqtt 1.35.3) were pulled. The network started but nothing worked with coordinator firmware `CC1352P2_CC2652P_launchpad_coordinator_20230507.hex` that was OK previously.

I read some thread that the fw is unstable so I installed `CC1352P2_CC2652P_launchpad_coordinator_20221226.hex` but nothing changed.

I messed with it for about 3 hours, being able to get some readings from Aquara sensors, but I was not able to control switches or lights.

Then, during the night, no sensor sent any data.

In the morning I noticed that 1.35.3 is quite new and rolled back to the version I have found in the older logs (1.35.1) Now the network works well.

### What did you expect to happen?

No issues with the new version - network starts, devices communicate.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20230507

### Adapter

https://www.aliexpress.com/item/1005004160661011.html

### Setup

Intel Celeron home server with HASS + zigbee2mqtt installed via docker-compose

mmithril, 2024-02-11:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14231859/log.txt)

Hessenpower01, 2024-02-11:
Similar problem here, after updating to 1.35.3 some Aqaea Opple and Huw switches were not sending anything anymore. After downgrading to 1.35.1 al works well again.

Adapter is Sonoff 3.0 USB

Adriankor, 2024-02-11:
Same here with IKEA Tradfri and Conner 2 stick. 

Downgrading solving the bug.

metaljay, 2024-02-11:
Just to add, mine also has the same problem, devices don’t communicate since updating

lyczko, 2024-02-12:
I hve the same with TS0601 TRV, don't communicate since updateing from 1.35.1 to 1.35.2 or 1.35.3

Filero, 2024-02-12:
Same problem since the update with : 
- QS-Zigbee-C03
- WHD02
- ti.router
- TS0011_switch_module

Klikini, 2024-02-12:
Same here with a ConBee II. Updating from 1.35.1 to 1.35.2 or 1.35.3 breaks communication with all my Aqara motion sensors, TRÅDFRI motion sensors, Sonoff contact sensors, and Stelpro thermostats. Possibly more devices, but that's all I noticed before downgrading again.

lyczko, 2024-02-12:
> I hve the same with TS0601 TRV, don't communicate since updateing from 1.35.1 to 1.35.2 or 1.35.3

Also conbee II, i don't check every single device, but trv is on the first position

frank-1971, 2024-02-13:
For me 1.35.1 made the whole Zigbee network very unstable. 1.35.2 works fine. So I did not tried 1.35.3 yet. And from what I am reading, I might wait for the next one...
I am on RaspberryPi, with Sonoff ZBDongle-E

jwGeorge, 2024-02-27:
I found this thread after opening my own issue. #21592 
On 1.35.3 any communication from HA to devices is extremely slow. Device to HA is fine. Using Sonoff 3.0 Dongle-P.  Devices are NUE/3A switches and LED Controllers.

Updating to `CC1352P2_CC2652P_launchpad_coordinator_20230507.hex ` made it worse and lose communication with devices after a few hours. 

Back to `CC1352P2_CC2652P_launchpad_coordinator_20221226.hex` and v1.35.1 - very fast now. Haven't tried 1.35.2

jwGeorge, 2024-03-04:
**Update**: Installed 1.36.0 and seems to be working well since this morning. Messages between HA and devices are quick and no timeouts. Will monitor for the rest of the day and report back.

Klikini, 2024-03-04:
Same here, this seems to be fixed!

Adriankor, 2024-03-04:
Version 1.36.0 Fixed as well

mmithril, 2024-03-17:
I do not know if it is the same problem or a new one. The network is very unstable.

Now have version `1.36.0` with `CC1352P2_CC2652P_launchpad_coordinator_20230507.hex`. It works for couple of hours then fails again. It seems that disconnecting the dongle makes it work for a longer time than just restarting zigbee2mqtt.

```zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 05:08:25: Publish 'set' 'state' to 'Ikea-socket2' failed: 'Error: Command 0x5c0272fffe882182/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":5329,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":224,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,102,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
zigbee2mqtt_1    | Zigbee2MQTT:debug 2024-03-17 05:08:25: Error: Command 0x5c0272fffe882182/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":5329,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":224,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,102,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))
zigbee2mqtt_1    |     at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:320:27
zigbee2mqtt_1    |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
zigbee2mqtt_1    |     at ZStackAdapter.dataRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:965:9)
zigbee2mqtt_1    |     at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:387:35)
zigbee2mqtt_1    |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
zigbee2mqtt_1    |     at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
zigbee2mqtt_1    |     at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
zigbee2mqtt_1    |     at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17)
zigbee2mqtt_1    |     at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
zigbee2mqtt_1    |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
```

Please see the full log attached.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14628245/log.txt)

Koenkk, 2024-03-17:
@mmithril that's a coordinator crash, 20230507 has some issues, try the [new test firmware](https://github.com/Koenkk/Z-Stack-firmware/discussions/496)

mmithril, 2024-03-17:
I am not able to start the coordinator using the test firmware

```zigbee2mqtt_1    | Using '/app/data' as data directory
zigbee2mqtt_1    | Zigbee2MQTT:debug 2024-03-17 22:06:25: Loaded state from file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2024-03-17 22:06:25: Logging to console and directory: '/app/data/log/2024-03-17.22-06-25' filename: log.txt
zigbee2mqtt_1    | Zigbee2MQTT:debug 2024-03-17 22:06:25: Removing old log directory '/app/data/log/2024-03-17.22-02-06'
zigbee2mqtt_1    | Zigbee2MQTT:info  2024-03-17 22:06:25: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
zigbee2mqtt_1    | Zigbee2MQTT:info  2024-03-17 22:06:25: Starting zigbee-herdsman (0.35.1)
zigbee2mqtt_1    | Zigbee2MQTT:debug 2024-03-17 22:06:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[26],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":49908},"serialPort":{"path":"/dev/ttyACM0"}}'
zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 22:06:36: Error while starting zigbee-herdsman
zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 22:06:36: Failed to start zigbee
zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 22:06:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 22:06:36: Exiting...
zigbee2mqtt_1    | Zigbee2MQTT:error 2024-03-17 22:06:36: Error: SRSP - AF - register after 6000ms
zigbee2mqtt_1    |     at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
zigbee2mqtt_1    |     at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:312:45
zigbee2mqtt_1    |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:26)
zigbee2mqtt_1    |     at Znp.request (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:300:27)
zigbee2mqtt_1    |     at ZnpAdapterManager.registerEndpoints (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:442:32)
zigbee2mqtt_1    |     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:100:9)
zigbee2mqtt_1    |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
zigbee2mqtt_1    |     at Zigbee.start (/app/lib/zigbee.ts:62:27)
zigbee2mqtt_1    |     at Controller.start (/app/lib/controller.ts:109:27)
zigbee2mqtt_1    |     at start (/app/index.js:107:5)
```

closed 2024-10-14 by github-actions[bot]