Koenkk/zigbee2mqtt #21351: New Alarm switches for Bosch BSD-2 not ideally represented

N3rdix, 2024-02-11:
### What happened?

With https://github.com/Koenkk/zigbee-herdsman-converters/pull/6862 the ability to trigger the alarm states was introduced (thanks so much to @vjonas, super happy) and this works basically.
Although the state can be switched on/off it is not repesented correctly via MQTT and for example shows as example unavailable in Home Assistant consequently.

Can someone help on improving the converter to get the state of the alarm switches to resolve this issue? Currently it is not being reported:
```
{
    "ac_status": false,
    "battery": 100,
    "battery_defect": false,
    "battery_low": false,
    "last_seen": "2024-02-11T10:37:42+01:00",
    "linkquality": 128,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "tamper": false,
    "test": false,
    "trouble": false
}
```

One minor question: can this be made a real "switch" instead of (however it is called) entity with 2 buttons for toggling?

[image]
[image]

### What did you expect to happen?

The entities are not unavailable but on or off

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Add-On

N3rdix, 2024-02-13:
Seems I made some progress with an external converter, once I have it in a testable state I will create a PR.
I am sticking to the values sniffed from the original feature request https://github.com/Koenkk/zigbee2mqtt/issues/17229#issuecomment-1824402822.

[image]

vjonas, 2024-03-01:
> Seems I made some progress with an external converter, once I have it in a testable state I will create a PR. I am sticking to the values sniffed from the original feature request [#17229 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/17229#issuecomment-1824402822).
> 
> ![image](https://private-user-images.githubusercontent.com/70015952/304488323-31d92aa4-11a8-4149-a4e3-24acb360b01c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkzMDUxNTAsIm5iZiI6MTcwOTMwNDg1MCwicGF0aCI6Ii83MDAxNTk1Mi8zMDQ0ODgzMjMtMzFkOTJhYTQtMTFhOC00MTQ5LWE0ZTMtMjRhY2IzNjBiMDFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzAxVDE0NTQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2ZjY1ZjY0ZTQ3ZTk0YjhjMjVlYWYwYzUzYzgyZjZmNGZlODM5MzcwN2MwMmQzOTBmOGE5Njg5MWE5MTY3NDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.etfJ8SeGRsLz_CezHVM246OqPzlABKKR6TcbIyQEEmM)

Great!
How is it going?
We should probably try to use the zonestatus in the convertGet method to get the correct values for the toggles.
I believe there are at least 4 states to consider:
- smoke on
- intruder on
- smoke & intruder on
- off 
I'm not sure wether we checked the zone status for distinct states.

N3rdix, 2024-03-01:
That's also what I implemented so far in the local converter, but did not find the time to finish it.
Hope to do so next week, happy for your feedback then

N3rdix, 2024-03-03:
That's the current converter I use, maybe someone can test before I create a PR: https://gist.github.com/N3rdix/7b27b208a61c08b1c7282b3f47d32dab

I added a mapping for the case when both switches are on (which does not make much sense), just for completeness.

Pferdebockwurst, 2024-06-15:
I think this can be closed.

N3rdix, 2024-06-15:
One thing could be improved. After a restart of Z2M the switches are unavailable until you trigger an explicit state change.

Workaround for me is currently manually clicking the "Off" button for the affected entities.

Not sure where this comes...

Pferdebockwurst, 2024-06-15:
Oh ok. Maybe someone has an idea.

Btw. can you recommend them? I think about buying a couple of them, price would be around 48 Euro.

N3rdix, 2024-06-15:
Buy them for this insane price, they are worth it. Fully recommended from my side, no problems so far and a lot of features for this price

N3rdix, 2024-06-15:
> @N3rdix Do you have `retain` enabled? 🤔

No it's not active for the BSDs, but good hint. Have you turned it on for all devices in general or selectively for specific ones? So far I never needed it

N3rdix, 2024-06-15:
Indeed this works, just tested and thanks for the recommendation @burmistrzak !

I'll close this Issue now.

closed 2024-06-15 by N3rdix

vjonas, 2024-06-24:
Hey guys, 1 more question about the " new sensitivity " state of the smoke detectors. Have you got any idea what the default value would be?
In the garage I find it gives too many false positives and it's quite annoying.

N3rdix, 2024-06-25:
> @vjonas 👋 I believe the default state is `low`. 

That's correct, default is `low`. Just confirmed by pairing a new device.

vjonas, 2024-06-25:
Okay thank you guys.

N3rdix, 2024-06-27:
> Indeed this works, just tested and thanks for the recommendation @burmistrzak !
> 

Need to revert this statement, after a restart the switches are again unavailable although retain has been enabled for those devices. Any idea @burmistrzak ?

```
{
    "ac_status": false,
    "alarm_burglar": false,
    "alarm_silenced": false,
    "alarm_smoke": false,
    "alarm_state": "1",
    "battery": 100,
    "battery_defect": false,
    "battery_low": false,
    "last_seen": "2024-06-27T12:33:11+02:00",
    "linkquality": 220,
    "restore_reports": true,
    "sensitivity": "medium",
    "smoke": false,
    "supervision_reports": true,
    "tamper": false,
    "test": false,
    "trouble": false
}
```

[image]

N3rdix, 2024-06-27:
Observation:
I changed the config of the switch in the MQTT broker manually (for testing) and this solves the issue, but I am not sure where this is coming from. Could this be an issue with the refactoring when "ON"/"OFF" were replaced in the converter (https://github.com/Koenkk/zigbee-herdsman-converters/pull/7399)?

Example topic I changed (redacted) the `payload_on `and `payload_off `values from "ON"/"OFF" to true/false:

```
{
  ...
  "command_topic": "zigbee2mqtt/0xb43a31fffe8717ac/set/alarm_burglar",
  ...  
  "entity_category": "config",
  "name": "Alarm burglar",
  "object_id": "0xb43a31fffe8717ac_alarm_burglar",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.38.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "payload_off": false,
  "payload_on": true,
  "state_topic": "zigbee2mqtt/0xb43a31fffe8717ac",
  "unique_id": "0xb43a31fffe8717ac_switch_alarm_burglar_zigbee2mqtt",
  "value_template": "{{ value_json.alarm_burglar }}"
}
```