Koenkk/zigbee2mqtt #21353: Changes introduced in release 18.4.0 breaks/switch Avatto wall thermostat Presets Mode for 'manual' and 'auto'  (manufacturer: '_TZE200_viy9ihs7')

uxdesignerhector, 2024-02-11:
### What happened?

Changes introduced in [realease 18.14.0](https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v18.14.0)  :
> Detect _TZE204_lzriup1j as TuYa ZWT198/ZWT100-BH (https://github.com/Koenkk/zigbee-herdsman-converters/issues/6912) ([34de76f](https://github.com/Koenkk/zigbee-herdsman-converters/commit/34de76fe3d1ffb2aa6dbcd712ceb5178f38712f9))

Breaks/switch preset modes `auto`, `manual` for Avatto wall thermostat in HomeAssistant front-end, showing  'manual' (in home assistant) when 'auto' is selected at the device and viceversa.

Device info:
- Description: `Avatto wall thermostat`
- Zigbee Manufacturer: `_TZE200_viy9ihs7 `
- Model[ ZWT198/ZWT100-BH](https://www.zigbee2mqtt.io/devices/ZWT198_ZWT100-BH.html#tuya-zwt198%252Fzwt100-bh).

It happened after updating the Home Assistant Add-on to `1.35.3-1`, before it was working properly.

I've been looking at the code in this commit ([34de76f](https://github.com/Koenkk/zigbee-herdsman-converters/commit/34de76fe3d1ffb2aa6dbcd712ceb5178f38712f9)), it seems that `auto` and `manual` were changed from position at line 4352

[image]

And as per issue https://github.com/Koenkk/zigbee2mqtt/issues/20072 it seems there  that `ZTW100` and `ZWT198` presets modes are different so here could be the cause for this error.

              Also just to confuse us on the ZWT100 the Preset values for Auto and Manual are different to the ZWT198.  The settings above are correct for the ZWT100

_Originally posted by @MrEcosse in https://github.com/Koenkk/zigbee2mqtt/issues/20072#issuecomment-1842706522_

### What did you expect to happen?

To have the preset mode working properly as before in Home Assistant. Maybe we could autodetect model/version and setup presets modes accordingly.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20211217

### Adapter

CC2652R1

### Setup

Add-on

### Debug log

[image]

uxdesignerhector, 2024-02-11:
Actions that I tried before making this issue:

- Reconfiguring device via UI in Zigbee2MQTT
- Removing device from Zigbee network and rejoining it

None of them worked

MrEcosse, 2024-02-12:
The ZWT198 and ZWT100 are different thermostats, with different abilities and should not be treated as the same model.

The ZWT198 is fingerprint _TZE200_viy9ihs7  In the datapoints section the preset should be enumerated as follows:
  [4, 'preset', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'manual': tuya.enum(1), 'temporary_manual': tuya.enum(2)})],

The ZWT100 is fingerprint _TZE204_lzriup1j.  In the datapoints section the preset should be enumerated as follows:
 [4, 'preset', tuya.valueConverterBasic.lookup({'manual': tuya.enum(0), 'auto': tuya.enum(1), 'temporary_manual': tuya.enum(2)})],

The ZWT100 also has the ability to select the sensor (floor, air or both).  I will post both files that are working with my system below for testing and hopefully adoption.

MrEcosse, 2024-02-12:
ZWT100

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_viy9ihs7']),
        model: 'ZWT198',
        vendor: 'TuYa',
        description: 'Avatto Battery Wall Thermostat',
        onEvent: tuya.onEvent({timeStart: '1970'}),
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'key_sound', 'inverse']}],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Full factory reset, use with caution!'),
	    e.binary('key_sound', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Switches sound when interacting with thermostat'),
            e.child_lock(),
            e.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['auto', 'manual', 'temporary_manual'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Antifreeze function'),
            e.max_temperature_limit()
                .withUnit('C')
                .withValueMin(15)
                .withValueMax(90)
                .withValueStep(0.5)
                .withPreset('default', 60, 'Default value')
                .withDescription('Maximum upper temperature'),
            e.numeric('deadzone_temperature', ea.STATE_SET)
                .withUnit('C')
                .withValueMax(10)
                .withValueMin(0.5)
                .withValueStep(0.5)
                .withPreset('default', 1, 'Default value')
                .withDescription('The delta between local_temperature (5<t<35)and current_heating_setpoint to trigger Heat'),
            e.enum('backlight_mode', ea.STATE_SET, ['off', 'low', 'medium', 'high'])
                .withDescription('Intensity of the backlight'),
	    e.binary('inverse', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Primary and secondary reverse selection.  ON is Auxillary reversal'),
            e.enum('working_day', ea.STATE_SET, ['disabled', '6-1', '5-2', '7'])
                .withDescription('Workday setting'),
            e.text('schedule_weekday', ea.STATE_SET).withDescription('Workdays (6 times `hh:mm/cc.cC`)'),
            e.text('schedule_holiday', ea.STATE_SET).withDescription('Holidays (2 times `hh:mm/cc.cC)`'),
            // ============== exposes for found, but not functional datapoints:
            /*
            e.min_temperature_limit() // dp 16
                .withValueMin(5)
                .withValueMax(15)
                .withValueStep(0.5)
                .withPreset('default', 10, 'Default value')
                .withDescription('dp16 is listed in Tuya, but no communication from device'),
            */
        ],
        meta: {
            tuyaDatapoints: [
[... 15 lines cut ...]
                [109, 'schedule_weekday', tuya.valueConverter.ZWT198_schedule],
                [109, 'schedule_holiday', tuya.valueConverter.ZWT198_schedule],
                [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
		[111, 'inverse', tuya.valueConverter.onOff], 
                // ============== found but not functional datapoints:
                // [16, 'min_temperature_limit', tuya.valueConverter.divideBy10],  // datapoint listed in Tuya, but no communication from device

                // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
                // For each period:
                // 1st byte: hour
                // 2nd byte: minute
                // 3rd, 4th bytes: temperature multiplied by 10
                // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
                // scheduling can't be configured at all on the device.
                // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
                // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
                // Then the schedule is:
                // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
                // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
            ],
        },
    };

module.exports = definition;
```

MrEcosse, 2024-02-12:
ZWT198

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),	
    runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lzriup1j']),
        model: 'ZWT100-16A',
        vendor: 'TuYa',
        description: 'Avatto wall thermostat for electric heating',
        onEvent: tuya.onEvent({timeStart: '1970'}),
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'key_sound', 'sensor_choice', 'sensor_limit', 'inverse']}],
        whiteLabel: [{vendor: 'AVATTO', model: 'ZWT-100-16A'}],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Full factory reset, use with caution!'),
            e.binary('key_sound', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Switches sound when interacting with thermostat'),
            e.child_lock(),
            e.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['auto', 'manual', 'temporary_manual'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Antifreeze function'),
            e.max_temperature_limit()
                .withUnit('Â°C')
                .withValueMin(15)
                .withValueMax(90)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Maximum upper temperature'),
             e.numeric('sensor_limit', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMin(35)
                .withValueMax(60)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Floor sensor temperature limit'),
            e.numeric('deadzone_temperature', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMax(10)
                .withValueMin(0.5)
                .withValueStep(0.5)
                .withPreset('default', 1, 'Default value')
                .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
            e.binary('inverse', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Primary and secondary reverse selection.  ON is Auxillary reversal'),
            e.enum('sensor_choice', ea.STATE_SET, ['inside', 'outside', 'all'])
[... 28 lines cut ...]
                [109, 'schedule_weekday', tuya.valueConverter.ZWT198_schedule],
                [109, 'schedule_holiday', tuya.valueConverter.ZWT198_schedule],
                [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
                [111, 'inverse', tuya.valueConverter.onOff], 

		
                // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
                // For each period:
                // 1st byte: hour
                // 2nd byte: minute
                // 3rd, 4th bytes: temperature multiplied by 10
                // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
                // scheduling can't be configured at all on the device.
                // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
                // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
                // Then the schedule is:
                // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
                // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
            ],
        },
    };

module.exports = definition;

```

Ronayke, 2024-02-12:
I have the same experience as @uxdesignerhector with a ZWT198. The code of @MrEcosse solves the issue!

Koenkk, 2024-02-13:
@MrEcosse thanks, integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-13 by Koenkk

ckegels, 2024-02-26:
> ZWT198
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

You put 198 but in the code it is 100? Is that correct???

MrEcosse, 2024-02-26:
There are two different versions posted earlier in the thread one for the 198 and the other for the 100

ckegels, 2024-02-26:
> There are two different versions posted earlier in the thread one for the 198 and the other for the 100

just wanted to say you wrote the wrong name above the code (switch them it seems like) or i am misunderstanding something. however i tried this code and also tried the edge version of z2m and didnt help.
thanks for trying to help me out tho.

dakush, 2025-01-09:
Same here with Sonoff dongle E after updating to the latest z2m version, "manual" and "auto" modes are wrong (they worked fine before the update)

dawogid, 2025-01-26:
Sadly I have the same issue after updating zigbee2mqtt to latest...

dakush, 2025-02-10:
Fixed for me with the latest version (2.1.1 commit: 1accb8b66b95b99658c50e4c0209edcb946a5fe2)