Koenkk/zigbee2mqtt #21354: IKEA Parasoll does not join the network and does not configure

smartmatic, 2024-02-11:
### What happened?

I am not able to add one IKEA Parasoll. The Interview starts but after some seconds the interview is canceled and ends with a the „device left the network“ message.

### What did you expect to happen?

Add IKEA Parasoll successfully to zigbee2mqtt

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Dongle-E

### Setup

Home Assistant Add-On

### Debug log

```

2024-02-11 17:15:45MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61/availability', payload '{"state":"online"}'
Info 2024-02-11 17:15:45MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:15:45+01:00"}'
Info 2024-02-11 17:15:45Device '0x3c2ef5fffe438d61' joined
Info 2024-02-11 17:15:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c2ef5fffe438d61","ieee_address":"0x3c2ef5fffe438d61"},"type":"device_joined"}'
Info 2024-02-11 17:15:46Starting interview of '0x3c2ef5fffe438d61'
Info 2024-02-11 17:15:47MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c2ef5fffe438d61","ieee_address":"0x3c2ef5fffe438d61","status":"started"},"type":"device_interview"}'
Info 2024-02-11 17:15:47MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:15:47+01:00"}'
Info 2024-02-11 17:15:47MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:15:47+01:00"}'
Debug 2024-02-11 17:16:00Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"modelId":"PARASOLL Door/Window Sensor"}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:00MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:00+01:00","linkquality":224}'
Debug 2024-02-11 17:16:01Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:01MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:01+01:00","linkquality":224}'
Debug 2024-02-11 17:16:02Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:02MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:02+01:00","linkquality":228}'
Debug 2024-02-11 17:16:03Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:03MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:03+01:00","linkquality":224}'
Debug 2024-02-11 17:16:04Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:04MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:04+01:00","linkquality":224}'
Debug 2024-02-11 17:16:07Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"stackVersion":113}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:07MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:05+01:00","linkquality":224}'
Debug 2024-02-11 17:16:07Received Zigbee message from '0x3c2ef5fffe438d61', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Info 2024-02-11 17:16:07MQTT publish: topic 'zigbee2mqtt/0x3c2ef5fffe438d61', payload '{"last_seen":"2024-02-11T17:16:06+01:00","linkquality":224}'
Warning 2024-02-11 17:16:07Device '0x3c2ef5fffe438d61' left the network
Info 2024-02-11 17:16:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c2ef5fffe438d61","ieee_address":"0x3c2ef5fffe438d61"},"type":"device_leave“}‘

```

dave21w, 2024-02-14:
Can't see this in the supported list on blakadder's site so probably why it doesn't work, hopefully it will be added soon

smartmatic, 2024-02-14:
The device is described here as supported device:
https://www.zigbee2mqtt.io/devices/E2013.html

dave21w, 2024-02-15:
Apologies, I missed that but unusually it isn't on blakadders site yet (unless I missed that too!). I can help on saturday evening as after seeing your post I am going to collect some. I like the fact they run on a single AAA rechargeable battery. Once I have mine I will see what happens when I try to pair it.

smartmatic, 2024-02-15:
I am curious if you are facing the same issues or how mine can be solved?

dave21w, 2024-02-17:
I think you misunderstood, I didn't actually have any at the time but have since been out and purchased some, mine paired just fine the first time I tried so I do not know why you are having issues. What version of z2m are you running,  I am on 1.35.1 and get the data as per the attached image.
[image]

smartmatic, 2024-02-17:
The Z2M version i am using is mentioned in my initial post.

dave21w, 2024-02-19:
Sadly i cannot answer your question however i have now paired another ikea door sensor with the latest version of z2m and that aslo works without issue, maybe during the pairing process you could try opening and closing the sensor a few times just in case it is going into sleep mode before the process completes which is preventing z2m from getting all the information it needs from the sensor.

smartmatic, 2024-02-19:
Thank you for your feedback but i am wondering who can answer my questions especially in regards to the provided log and why my sensor behave like shown in the log?
I does not help or solve my problem to keep the sensor active as the sensor left always the network after some seconds as shown in the log attached in my initial post!

dave21w, 2024-02-20:
Not that this should make any difference but what coordinator are you using, mine is the CC2652R stick from Electrolama
https://electrolama.com/projects/zig-a-zig-ah/
Do you have another type you could try on a test system or something.

smartmatic, 2024-02-20:
> Not that this should make any difference but what coordinator are you using, mine is the CC2652R stick from Electrolama https://electrolama.com/projects/zig-a-zig-ah/ Do you have another type you could try on a test system or something.

You can find this information in my initial post.

smartmatic, 2024-02-24:
> Not that this should make any difference but what coordinator are you using, mine is the CC2652R stick from Electrolama https://electrolama.com/projects/zig-a-zig-ah/ Do you have another type you could try on a test system or something.

I have setup a new test system and you are right. The Sensor has paired immediately in my first attempt.

For that reason i am really wondering what is causing the issue in my main prod systems.

Only thing i observed is that when the sensor starts the interview my Nous AZ1 are kind of crashing and losing the connection for some seconds. I have 10 in total in my mesh of 60 zigbee devices. 

Should i try to pair the sensor directly to the coordinator?

dave21w, 2024-02-24:
I never really thought about pairing with anything other than the coordinator, my test system is at work but I can try and pair another one with my prod system via the mesh instead of directly with the coordinator and see what happens and let you know.

smartmatic, 2024-02-24:
Recently I have read more and more often that users pair their new Zigbee devices from the MESH at their destination via a nearby Zigbee router, which is supposed to increase stability. Is that wrong?

After a successful test, I reset the sensor to pair it in the Prod system. Then some strange happend.
Shortly after the sensor was reset (I held down the pairing button for more than 10 seconds), the sensor automatically paired directly with an IKEA bulb and was available in Zigbee2mqtt. This means the sensor was connected directly to the light source. When I opened the sensor the light came on. When I closed the sensor the light went off.

I then splitt the direct connection between the light and the sensor and reconfigured the sensor using the reconfigure button in zigbee2mqtt and can use it as expected.

dave21w, 2024-02-24:
No I don't think that's wrong, my coordinator is on top of my cabinets in my kitchen (rear of the house) but the extension we have uses K Glass which means it has a metalised coating to reflect heat and this used to really cause me issue with zigbee stuff in my garage (at the bottom of my garden) until I put an Ikea plug in my greenhouse which is half way down my garden. I didn't delete and repair any devices they just sorted themselves out and connected by multiple routes. I will try pairing the sensor via the mesh though as now I'm curious to see what happens !

closed 2024-09-22 by github-actions[bot]