Koenkk/zigbee2mqtt #21361: [Wrong device]: Lonsonho ZB-RGBCW

rufabf, 2024-02-11:
### Link

https://es.aliexpress.com/item/1005005073267687.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60497396209&aff_fcid=72ff9d597bcb47f7b15cfe11fe277bfe-1707685085563-02883-_EHPYaWN&tt=MG&aff_fsk=_EHPYaWN&aff_platform=default&sk=_EHPYaWN&aff_trace_key=72ff9d597bcb47f7b15cfe11fe277bfe-1707685085563-02883-_EHPYaWN&shareId=60497396209&businessType=ProductDetail&platform=AE&terminal_id=f249ef705ca142579d5606c9a0a2bbcf&afSmartRedirect=y

### Model

YSRSAI YSR-MINI-01_rgbcct

### Description

Led strip controler

### Vendor

Tuya

### Picture (link)

https://ae03.alicdn.com/kf/Scbd96754074a434b8f09a58e9869410f8.jpg

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0xa4c1385626311785","nwkAddr":59433,"manufId":4107,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,2821],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":24939,"currentY":24701,"currentSaturation":0,"colorTemperature":500}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":0,"dateCode":"20230629","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707683033636}

0berkampf, 2024-03-06:
Hi,

Same issue for me, this device YSRSAI YSR-MINI-01_rgbcct is recognized as Lonsonho ZB-RGBCW and the ON/OFF action of the button is not reported.

Thx for your help.

ingoratsdorf, 2024-03-28:
Looks like my issue is the same:
[](https://github.com/Koenkk/zigbee2mqtt/issues/21912)

ingoratsdorf, 2024-03-28:
Same Lonsonho that creates issues here:
[https://github.com/Koenkk/zigbee2mqtt/issues/19465](https://github.com/Koenkk/zigbee2mqtt/issues/19465)

ingoratsdorf, 2024-03-28:
From what I can read on the forum and by reading through custom converters, it looks like we / Z2M needs a better profile matching algorithm. Coming from a few months of Homeassistant ZHA integration, they do matches via clusters, not just model and manufacturer. Something to think about. There, a custom converter is overwriting the built-in one which is not the case in Z2M. I could not find a method to "overwrite" a builtin converter.
There's too many false positives.

dotWee, 2024-05-11:
Had the same issue; fixed it by writing a short converter that copies the Lonsonho fingerprint and uses the YSRSAI light extension definition.

Put the code below as `fix-ysrsai-confusion.js` in your zigbee2mqtt `/data` directory and enable it by adding `fix-ysrsai-confusion.js` as custom converter:

```yaml
external_converters:
  - fix-ledstrip-vendor-detection.js
```

Then remove and join the LED strip controller to your network. Then it should be detected and working as expected.

```js
/*
 * Fixes the "YSRSAI YSR-MINI-01_wwcw" LED Strip wrongfully detected as a "Lonsonho ZB-RGBCW" bulb.
 *
 * Attributes zigbeeModel & fingerprint taken from the Lonsonho converter definition:
 *   https://github.com/Koenkk/zigbee-herdsman-converters/blob/c385d3fa7766ac5ab33bcc7124a1691663b6b8b6/src/devices/lonsonho.ts#L151
 *
 * Attributes model, vendor, description & light definition taken from the YSRSAI converter definition:
 *   https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/ysrsai.ts#L4
 */
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['ZB-RGBCW'],
    fingerprint: [{modelID: 'ZB-CL01', manufacturerName: 'eWeLight'}, {modelID: 'ZB-CL01', manufacturerName: 'eWeLink'},
        {modelID: 'ZB-CL02', manufacturerName: 'eWeLight'}, {modelID: 'ZB-CL01', manufacturerName: 'eWeLi\u0001\u0000\u0010'},
        {modelID: 'Z102LG03-1', manufacturerName: 'eWeLink'}],
    model: 'YSR-MINI-01_wwcw',
    vendor: 'YSRSAI',
    description: 'Zigbee LED controller (RGB+CCT)',
    extend: [tuya.modernExtend.tuyaLight({colorTemp: {range: [160, 370]}, color: true})]
};

module.exports = definition;
```

rufabf, 2024-05-11:
> Had the same issue; fixed it by writing a short converter that copies the Lonsonho fingerprint and uses the YSRSAI light extension definition.
>
> Put the code below as `fix-ysrsai-confusion.js` in [...]

What folder did you put the file? i have no zigbee2mqtt/data folder

closed 2024-12-08 by github-actions[bot]

coasttech, 2025-03-06:
did not work for me :(

dotWee, 2025-03-15:
@coasttech maybe try one of those definitions I posted here: https://github.com/Koenkk/zigbee2mqtt/issues/21912#issuecomment-2106317017