Koenkk/zigbee2mqtt #21366: [EnOcean PTM 216Z] Cannot read properties of undefined (reading 'raw')

sebastienserre, 2024-02-12:
### What happened?

This is followup of #17104 which has been closed too early.
The issue still exists.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff USB

### Setup

ADD on Home Assistanbt

danielpetrovic, 2024-02-13:
No problems here, but I am using the Niko FriendsOfHue which are based on the [PTM 215Z ](https://www.zigbee2mqtt.io/devices/PTM_215Z.html).

Keep in mind that for Zigbee Green Power to work, that your button needs to be in the region of the Hue lamp that you want to use as 'translator' for the Zigbee Green Power messages. Simply having the Hue lamps inside your Zigbee network is not enough. You don't need to control that specific light with it, it is just the translator.

When pairing it is advisable to select a specific Hue light to 'pair' it with instead of through 'all' . This way you are only sending unicast Zigbee signals and this will enhance your performance greatly, especially on bigger Zigbee networks.

We have only Hue lights, 47 all across the house and outside, so I can place the Friends of Hue buttons anywhere and have currently 12 of them working with 0 issues. We are using them to control our Somfy RTS and IO motors through HA.

sebastienserre, 2024-02-13:
But in case, I do not want to command a Hue lamp but a Sonoff micromodule, I do not understand how to link them ....

danielpetrovic, 2024-02-13:
> But in case, I do not want to command a Hue lamp but a Sonoff micromodule, I do not understand how to link them ....

Read again what I said. The buttons are only linked for translation of the Zigbee Green Power signal to normal Zigbee signal. The Zigbee Green Power devices do not support binding.

So you only link it so the Hue light can convert the signal to normal signal. And then in your Automation software like HA you can choose what to do when a button is pressed.

As I said I am controlling Somfy curtains and shutters which do not use Zigbee at all.

wayetii, 2024-02-22:
Personally, I have the same problem even when using a Hue lamp. I bought a Hue bulb and paired through it, but I still get the same error message.

wayetii, 2024-02-22:
Yes of course.

I use ZBDongle-E and PTM 216Z

danielpetrovic, 2024-02-22:
What about others with issues? Maybe also the ZBDongle-E?

ZBDongle-E is still experimental support, not sure if Zigbee Green is implemented yet?

I’m using ZBDongle-P as it uses the preferred chip and is on the recommended list for Z2M.

chris-1243, 2024-02-22:
> What about others with issues? Maybe also the ZBDongle-E?
> 
> 
> 
> ZBDongle-E is still experimental support, not sure if Zigbee Green is implemented yet?
> 
> 
> 
> I’m using ZBDongle-P as it uses the preferred chip and is on the recommended list for Z2M.

I have both Dongle-P (prod network) and Dongle-E (test network). 

On the P version, both PTM215Z / PTM216Z are working. The drawback with PTM216Z is the few Hue bulbs able to translate the ZGP frames (I have 4 differents bulbs and only one type is able to do the pairing).
PTM215Z is easy to pair even with other brands like some Ikea bulbs, Sunricher modules...

On the E version, it is a bit more complicated. The PTM215Z is working as expected. The pairing via a router is sometimes complicated as I experienced.
PTM216Z is throwing errors even when paired via a specific router... I have not been able to use it.

chris-1243, 2024-02-27:
For those interersted, I would suggest to read this [thread](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) carefully. It is a new driver (ember) for EZSP adapter like Sonoff Dongle-E.

Enocean PTM216Z has been tested and it is fully working.

wayetii, 2024-03-10:
> For those interersted, I would suggest to read this [thread](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) carefully. It is a new driver (ember) for EZSP adapter like Sonoff Dongle-E.
> 
> Enocean PTM216Z has been tested and it is fully working.

Tested. This fixes the problem.

Nerivec, 2024-05-15:
Closing as this should be fixed in `ember`.

closed 2024-05-15 by Nerivec