Koenkk/zigbee2mqtt #21371: TuYa BSD29_1: Updates to measurement polling interval require a restart to apply

nARN, 2024-02-12:
### What happened?

Following up on #21018. Similarly to the @felipebraz I found it confusing that changing measurement poll interval does not affect the frequency of updates until Z2M is restarted.

It seems to only be necessary when polling is initially enabled. When polling interval is set to -1 the next change applies without restarting.

### What did you expect to happen?

Either polling interval should apply right away, or, if the restart requirement is intentional, it should be documented (ideally in the UI itself, in addition to the device doc)

### How to reproduce it (minimal and precise)

- On the "Settings (specific)" tab of a TuYa BSD29_1, ensure that measurement poll interval is _not_ -1 (the default of 60 seconds is ok)
- Set measurement poll interval to a new value
- Observe update frequency on the corresponding topic. It will be the previous update frequency
- Restart zigbee2mqtt
- Observe update frequency on the corresponding topic. It will be the new frequency.

To confirm that updates apply without restarting:
- set poll interval to -1
- restart zigbee2mqtt (to apply the change)
- set poll interval to a new positive value
- Observe that updates are now coming through without a restart

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Setup

RPi, RasPiOS bullseye arm64, systemd

redpixy, 2024-02-15:
Yes, I also realised this, the settings doesn't get updated until z2m is restarted.

klelifo, 2024-02-18:
Same here (using HomeAssistant).
Changed the setting for the polling interval and it did not become active until I restarted zigbee2mqtt.
You can easily verify that the setting is applied (or not) by filtering for the relevant messages in the "Log" tab of Homeassistant.

verybadsoldier, 2024-05-20:
I can confirm z2m needs a restart after changing the polling setting.

ZionFox, 2024-07-15:
Can confirm as of 1.39.0 it still requires a restart of Z2M for the changes to take effect.

If this isn't intended to be implemented any time soon, perhaps it's wise to inform the users in the description to `measurement_poll_interval` to instruct them to restart Z2M for changes to take effect?

rolandas-valantinas, 2024-10-07:
Still problem on 1.40.0!
Spent hours trying to figure out why my TS011F is not updating with correct power figures, thought it was a faulty plug. When stumbled upon this issue, restarted Z2M and plug now polling at correct intervals.

ericcastro, 2024-10-22:
confirm this problem is still present as of october 2024

ZionFox, 2024-12-30:
Commenting to remove stale label.

rolandas-valantinas, 2025-03-09:
Still an issue

JoergAdler, 2025-04-05:
still also here an issue with Tuya TS011F

JoergAdler, 2025-06-05:
Still an issue

closed 2025-08-13 by github-actions[bot]