Koenkk/zigbee2mqtt #21374: BHT-006GCZB - no zigbee connection

Ny-Mic, 2024-02-12:
### What happened?

I have tried to connect a new Moes BHT-006GCZB to Zigbee2mqtt. 
First permit join on zigbee2mqtt. Then as described in the manual I have reset the BHT. (when the thermostat is off, press an hold the "arrow down" - button until the screen flashes quickly an displays the "Signal"-icon.)
Result: The right columns of icons  flashes, but no ""Signal"-icon is displayed
I have tried the on 2 different Raspberry PI ( Model 4 an 3b+) and 2 new Moes BHT-006GCZB. Every time with same result.
.

### What did you expect to happen?

Reset and connect Moes BHT-006GCZB to zigbee2mqtt

### How to reproduce it (minimal and precise)

Every time when I tried to reset and connect the  BHT-006 to zigbee2mqtt

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20211115

### Adapter

TI CC2531 and SONOFF Zigbee 3.0 USB Dongle

### Setup

Plain on RPI 3B+ and docker image on RPI4

### Debug log

info  2024-02-12 21:07:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-12.21-07-32' filename: log.txt
info  2024-02-12 21:07:32: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a)
info  2024-02-12 21:07:32: Starting zigbee-herdsman (0.33.8)
info  2024-02-12 21:07:33: zigbee-herdsman started (resumed)
info  2024-02-12 21:07:33: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
info  2024-02-12 21:07:33: Currently 0 devices are joined:
info  2024-02-12 21:07:33: Zigbee: disabling joining new devices.
info  2024-02-12 21:07:34: Connecting to MQTT server at mqtt://192.168.178.69
info  2024-02-12 21:07:34: Connected to MQTT server
info  2024-02-12 21:07:34: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2024-02-12 21:07:34: Started frontend on port 8080
info  2024-02-12 21:07:34: Zigbee2MQTT started!
info  2024-02-12 21:10:28: Zigbee: allowing new devices to join.
info  2024-02-12 21:10:29: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"t2byj-1"}'
info  2024-02-12 21:13:50: Zigbee: disabling joining new devices.
info  2024-02-12 21:13:50: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"t2byj-2"}'
info  2024-02-12 21:13:53: Zigbee: allowing new devices to join.
info  2024-02-12 21:13:53: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"t2byj-3"}'
info  2024-02-12 21:38:01: Zigbee: allowing new devices to join.
info  2024-02-12 21:38:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"6jlot-1"}'

Ny-Mic, 2024-02-24:
Hello @fabrunner-HA ,
sorry I haven't find a solution
I have bought a second one at the Official MOES Shop, but this device has also the same behavior.
Now I'm looking for an other thermostat with zigbee.

casaprox2023, 2024-03-28:
""Amigos con la pantalla Apagada""" pulasar el modo eco que es para que la pantalla se apague y luego boton hacia abajo 10sg

closed 2024-10-25 by github-actions[bot]