Koenkk/zigbee2mqtt #21377: [New device support]: (Insta) NEXENTRO Dimming Actuator 57006000

bar-xx3, 2024-02-12:
### Link

https://www.insta.de/produkte/zigbee_dimmaktor_mini

### Database entry

0x5c0272fffe62a716

### Comments

I haven't tried adding the dimming actuator yet.
can anyone help
[image]

### External definition

```shell
const {onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['NEXENTRO Dimming Actuator'],
    model: 'NEXENTRO Dimming Actuator',
    vendor: 'Insta GmbH',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), identify()],
    meta: {},
};

module.exports = definition;
```

hherm, 2024-03-08:
I'm on the way. I paired my NEXENTRO dimm yesterday, but it showed up as blinds actuator. So it's useless.
But I can't forecast how long it will take.

bar-xx3, 2024-03-08:
Hello, thank you for your message, that's no problem. I can still wait. I
would like to thank you very much for your work and wish you a nice
weekend. Best regards

rua136, 2024-08-26:
Hello, just wanted to check-in on the Z2M integration of this NEXENTRO dimmer.

I wish there was an alternative dimmer that is already Z2M-integrated, but no other in-wall dimmer can dim both RC (trailing edge) and RL (leading edge) like this one, which is needed for certain LEDs. All other in-wall dimmers are RC (trailing edge) only - a real shame. So I feel stuck to get the one from NEXENTRO.

Thanks a lot,
Andreas

bar-xx3, 2024-08-26:
> Hello, just wanted to check-in on the Z2M integration of this NEXENTRO dimmer.
>
> I wish there was an alternative dimmer that is already Z2M-integrated, but no other in-wall dimmer can dim both RC [...]

Hello Andreas,

the Nexentro dimmer still works! I can operate and control it. It just says "not supported" in the support status.

You can buy it with confidence.

Greetings Steven

rua136, 2024-08-27:
Steven, I did not expect that. Thanks!
I just ordered one of the Nexentro dimmers. Will let you know about the outcome.

Cheers, Andreas

hherm, 2024-08-27:
> [Verfolgungsbild][https://github.com/notifications/beacon/AJLDMN2XYTOOYQ5KRE74YWDZTSEOZA5CNFSM6AAAAABDFO37MOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUJ3GCJM.gif]

Hi Steven, hi Andreas!
I'm already experimenting with a configuration file.
I have the log output, I can switch on and off,
but I stucked at the endpoint wrapping.
I'll try to post what I have this evening.

Greetings
Henry

rua136, 2024-08-29:
Hi guys,
I just installed the dimmer.
Via Bluetooth through the Nexentro Android app, it works like a charm. Perfectly dims my LEDs.
When pairing it via Zigbee in Home Assistant, z2m recognizes it as Insta 57008000 blind actor - just like you wrote earlier, Henry.
[image]

No working control over switching/dimming yet.

Steven, can you explain how you got it to work?

Henry, I wish I could be of any help, but I have not fumbled around with endpoint configuration yet.

Cheers,
Andreas

rua136, 2024-08-29:
Just found there is another 'new device support' thread dealing with this dimmer:
[](https://github.com/Koenkk/zigbee2mqtt/issues/22072)

No working solution there yet though as well.

hherm, 2024-08-30:
Just a quick post what I have at the moment.
I wrote a "NEXENTRO_dimmer.js" as an extention of the configuration.yaml:
```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['NEXENTRO Dimming Actuator'],
    model: '57006000', // Update this with the real model of the device (written on the device itself or product page)
    vendor: 'Insta GmbH', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'Dimming actor with brightness control and an input for a wall switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [],
    fromZigbee: [fz.on_off,fz.command_on, fz.command_off, fz.command_move, fz.command_stop], // We will add this later
    toZigbee: [tz.on_off,tz.light_brightness_move], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.action(['toggle', 'brightness_move_up', 'brightness_move_down', 'brightness_stop'])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    ota: ota.zigbeeOTA,
};
module.exports = definition;
```

But the on/off and brightness_move does not show up in the "Exposes" tab. So I think what's in the "exposes"-line in the config can't be right. Therefore the Problem I face is to find the right functions or to map them in the right way.
I started trying to pick up some hints from other dimmers, but no success for now.

From the logs:
Sequence: On - Off - On - Dimm-up - Off

```
debug 2024-08-30 08:29:07Received Zigbee message from '0x5c0272fffebd41fb', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 4 with groupID 30518
info 2024-08-30 08:29:07MQTT publish: topic 'zigbee2mqtt/0x5c0272fffebd41fb', payload '{"action":"on","action_group":30518,"linkquality":90,"state":"OFF","update":{"installed_version":5212,"latest_version":5212,"state":"idle"}}'
debug 2024-08-30 08:29:10Received Zigbee message from '0x5c0272fffebd41fb', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 4 with groupID 30518
info 2024-08-30 08:29:10MQTT publish: topic 'zigbee2mqtt/0x5c0272fffebd41fb', payload '{"action":"off","action_group":30518,"linkquality":90,"state":"OFF","update":{"installed_version":5212,"latest_version":5212,"state":"idle"}}'
debug 2024-08-30 08:29:15Received Zigbee message from '0x5c0272fffebd41fb', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":51}' from endpoint 4 with groupID 30518
info 2024-08-30 08:29:15MQTT publish: topic 'zigbee2mqtt/0x5c0272fffebd41fb', payload '{"action":"brightness_move_up","action_group":30518,"action_rate":51,"linkquality":90,"state":"OFF","update":{"installed_version":5212,"latest_version":5212,"state":"idle"}}'
debug 2024-08-30 08:29:17Received Zigbee message from '0x5c0272fffebd41fb', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 4 with groupID 30518
info 2024-08-30 08:29:17MQTT publish: topic 'zigbee2mqtt/0x5c0272fffebd41fb', payload '{"action":"brightness_stop","action_group":30518,"linkquality":90,"state":"OFF","update":{"installed_version":5212,"latest_version":5212,"state":"idle"}}'
```

bar-xx3, 2024-09-01:
> Just a quick post what I have at the moment.
> I wrote a "NEXENTRO_dimmer.js" as an extention of the configuration.yaml:

Andreas, I'm just a beginner too and I added the Nexentro to my existing network via the APP (pairing mode Z2M) and it worked right from the start.

Here is the output from the "Developer Console" external_definition:

const {deviceEndpoints, identify, light, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['NEXENTRO Dimming Actuator'],
    model: 'NEXENTRO Dimming Actuator',
    vendor: 'Insta GmbH',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"2":2,"4":4}}), identify(), light(), commandsOnOff({"endpointNames":["4"]}), commandsLevelCtrl({"endpointNames":["4"]}), commandsColorCtrl({"endpointNames":["4"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

rua136, 2024-09-12:
Hi guys,
just fiddled around with your input.
Using bar-xx3's external_definition as a js-file in the zigbee2mqtt folder actually makes it work in a basic way.
Light can be turned on/off, and also the brightness slider works.
It is not as perfect though as through the Bluetooth Android app, which is still functioning  in parallel: The z2m connection seems to ignore the minimum brightness which I set in the app. As a result, only starting from a z2m brightness value of 6 the lights actually start lighting up.

Still, it seems to be a great starting point for optimization.
If you make any progress, please feel free to share.
I may try to make my first steps in optimizing a new device as well, and will report back.

Cheers, Andreas

closed 2025-01-08 by github-actions[bot]