Koenkk/zigbee2mqtt #21389: [Feature request]: Customize HA discovery and/or disable discovery on a per-device basis

nateww, 2024-02-13:
### Is your feature request related to a problem? Please describe

I have a Zigbee thermostat (https://www.zigbee2mqtt.io/devices/Zen-01-W.html) setup to controller my in-floor heating system.  As such, when I configure the thermostat, it has no cooling mode OR fan-mode.  Unfortunately, the default discovery sends out discovery for both the cooling mode and fan-mode, which causes multiple errors, and then one annoyance.  Because the fan , and one error. First, the device can not be created properly, because the thermostat does not report the fan-mode, causing the following to appear in the home-assistant logs.
```
 [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fan_mode' when rendering '{{ value_json.fan_mode }}'
[homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_cooling_setpoint' when rendering '{{ value_json.occupied_cooling_setpoint }}'
```

The fix is to update the configuration to remove these from the discovery config line.

Similarly, the default discovery template also sets both the high/low setpoint for the thermostat, which has no value in my setup (and the thermostat doesn't actually support this functionality as configured in my environment).

Since I had to modify the HA configuration anyway, it was trivial to remove the extra functionality and create a custom HA configuration that only contains the functionality my thermostat supports.

### Describe the solution you'd like

Unfortunately, everytime Zigbee2MQTT restarts, it over-writes my customized discovery line.  One solution is to disable HA discovery.  However, I have dozens of other devices (and am constantly adding/removing devices on a fairly regular basis as I build out my new house), so I'd rather not disable discovery completely.

Two solutions come to mind:

1. Allow a configuration item (in the devices perhaps) that would allow me to disable discovery for a single device.  Then, I could create a custom discovery topic JUST for the misbehaving device.  This would allow me to leave HA discovery on for the dozens of other devices.
2. Allow me to customize the discovery information on a per-device basis.  Then, I could remove the fan properties from the discovery (ie; suppress that functionality from the discovery payload) and thus have a long-term solution that doesn't require me to create custom MQTT messages at startup.

These are two potential solutions I've considered, but perhaps others have ideas I haven't considered.

### Describe alternatives you've considered

Disabling discovery completely, and then manually creating discovery topics for all of my devices.  This is both error-prone, and removes one of the single-biggest advantage to Zigbee2MQTT, which is auto-configuration setup of Zigbee devices.

### Additional context

I have researched the documentation as well as looked through the historical record, and I couldn't find anything of note that indicates it was possible to accomplish what I was hoping for.

https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#overriding-discovery-properties

I attempted to decode the above information, and it wasn't obvious (to me) how to customize the discovery properties to modify and/or suppress existing discovery properties, or a way to disable discovery for a particular device.

nateww, 2024-02-13:
This issue seems related, but no responses were given to it, and it was closed over 3 years ago.
- https://github.com/Koenkk/zigbee2mqtt/issues/4941

This also seems relevant, but recent documentation implies that the homeassistant configuration effects the normal (non-discovery payload, not the discovery payload).
- https://github.com/Koenkk/zigbee2mqtt/pull/1030

https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html

```
homeassistant
Allows overriding the values of the Home Assistant discovery payload. See example below.
```

I didn't see any examples on this page related to discovery.

YogoGit, 2024-08-15:
Not stale (just ignored)

YogoGit, 2024-12-30:
Still think it would be a good thing, so it's still not stale (just ignored).

Barsonax, 2025-01-15:
I have some lights that I grouped in zigbee groups. I don't want to expose the individual lights to HA as I just use the group and/or zigbee scenes to control them so being able to disable some of them from showing up in homeassistant would be useful to both clean up clutter but also reduce the amount of needed updates.

YogoGit, 2025-03-17:
Not stale, just ignored.

KazWolfe, 2025-05-07:
Yet another +1 here. I just got an Inovelli Blue switch, which spammed the heck out of my entities list. I have no desire or need for all the parameters these switches expose to be settings, and it's extremely painful to have to manually add all the entries to the device configuration.

It'd be nice to be able to define a short list of the entities I _do_ want exposed to Home Assistant, with the rest accessible only via the Zigbee2MQTT UI.