Koenkk/zigbee2mqtt #21392: Philips Bulb 9290018216 was found as Philips Bulb 8718699673147

cracyfloyd, 2024-02-14:
### What happened?

The Philips Bulb 9290018216 was found as Philips Bulb 8718699673147
Wrong recognition.

### What did you expect to happen?

The result is now that i have the wrong firmware flashed. Can you help to fix.

### How to reproduce it (minimal and precise)

every time i learn the device new. new recognition of bulb.

### Zigbee2MQTT version

1.35.3-dev commit: 2aaf878

### Adapter firmware version

0x26500900

### Adapter

Conbee 3

### Setup

Docker

propi62, 2024-04-07:
i have the same problem with 4 bulps. all 4 bulps dont turn off completely.
the picture in the frontend is also wrong.

closed 2024-11-04 by github-actions[bot]

Miesje, 2024-12-14:
This issue still exists.
Will this be altered or not.

Thanks in advance