Koenkk/zigbee2mqtt #21396: IKEA LED1923R5 Transition Time off by one

cmprmsd, 2024-02-14:
### What happened?

I bought LED1923R5 (color) and LED2106R3 (white spectrum) GU10 bulbs.
Transition time on LED2106R3 works fine. Setting 0 enables me to do "blink" switching manually.
However, LED1923R5 (color) seems to add **one second** to my transition time that I set in the **Settings (specific)** tab.

Setting 0 will have 1 second transition time.
Setting 1 will have 2 seconds transition time.
Setting -1 will result in the following error message:
```
Publish 'set' 'state' to 'KGBueroLEDFarbe' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 0x287681fffe537155/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":-10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received -10)'
```

### What did you expect to happen?

The transition time should work the same way for the color LEDs as it does for the white spectrum LEDs.

### How to reproduce it (minimal and precise)

Set the transition to 0 and see if on/off/toggle blinks instead of a transition effect.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB - ZBDongle-P TI CC2652P + CP2102(N)

### Setup

Add-on to HomeAssistant in a Proxmox VM

cmprmsd, 2024-08-13:
Just for reference. This is an ongoing bug I tried to inform ikea about and they replied a fix is in work. But that was quite a while ago. 

https://www.reddit.com/r/tradfri/comments/1cp0b25/transition_times_on_led1923r5_and_led1924g9_are/

I'll update and close this issue if there is any news.

cmprmsd, 2025-01-01:
Sadly ikea does not answer any more to the thread on Reddit. I have no idea how to fix this without firmware changes. 

Shall this issue be closed @Koenkk? I think it's "their" issue.

Koenkk, 2025-01-02:
Yes lets close it

closed 2025-01-02 by Koenkk